espressif/arduino-esp32 #963: LoLin32 Pro - SD Card Issue

cterna, 2018-01-03:
Hi all, I have just purchased a LoLin32 Pro for a home project where I need to log some data to an SD card. I am trying to initialise a Micro SD card through the onboard Micro SD slot using the SD library functions but cannot get the card to work. On a previous project, I have had the SD card work on the LoLin32 through a Wemos Micro SD shield with no problems so the SD card is fine, but it won't initialise when used in the SD card slot for the LoLin32 Pro. 

The set up of the board is quite literally just the micro sd card, inserted into the micro sd card slot and then the board connected to the USB cable. 

Here is the portion of my code that is responsible for initialising the micro SD card. 

`#include <SD.h>                  
#include <SPI.h>                  

#define CS 13

File dataFile;

void setup() {
  Serial.begin(9600); 

  setUpSD();
  
}

void loop() {
  // put your main code here, to run repeatedly:

}

void setUpSD() { 

  Serial.print("\n\n"); 

  //Tell user to insert SD card
  Serial.println("Please insert SD card"); 

  while(!SD.begin()) {
    
  }

  Serial.println("SD card present"); 

  if(SD.exists("/dataLog.txt") != true) {
    dataFile = SD.open("/dataLog.txt", FILE_WRITE); 
    dataFile.print("\n"); 
    dataFile.close();
    Serial.println("File 'dataLog.txt' created"); 
  } else {
    Serial.println("File 'dataLog.txt' already exists"); 
  }
  
}

When I upload it all I receive on the serial monitor is 'Please insert SD card' and then it gets stuck in the while loop. 

I even tried using the examples sketch from the esp32 library for the SD card for esp32 but that doesn't work either, it just tells me that the 'card mount has failed', which is the response it gives if SD.begin() returns false. 

Here is the schematic for the LoLin32 Pro as well. 

[sch_lolin32_pro_v1.0.0.pdf](https://github.com/espressif/arduino-esp32/files/1599518/sch_lolin32_pro_v1.0.0.pdf)

Any help as to why SD.begin() won't work will be greatly appreciated. 
Thanks!

tueddy, 2018-01-15:
Hi,

did you try the SD_MMC example in SD(ESP32)?
Pin connection of Lolin32 Pro SD card is the SDMMC hardware bus on the ESP32 port.
I got SD card working with "SD_MMC.h" instead of "SD.h".
Make sure you have a fast SD micro card because the card is being initialized in high speed mode.,

Hope this helps
Dirk

OneForSorrow, 2018-02-01:
Hello, I am currently trying to use the SD card with the loli32 pro. I noticed something. If I change in the code the speed from 9600 to 115200 I get the "please insert the sd card" But if I leave it at 9600 it doesn't display this message.

cterna, 2018-02-02:
Hi tueddy, 

Thank you very much for your reply. 

I didn't get the opportunity to try the example sketch as I noticed the boards were overheating at the USB port when plugged into the computer. I first noticed this when I went back to try your suggestion on a new LoLin32 Pro (I had already bought 5 of them). I opened it from its packaging and I held it whilst I plugged it in, within seconds I had to drop it as it got so hot. 

This is very strange as I have run multiple WeMos D1 Mini boards, various WeMos LoLin32 boards, Arduino Uno boards, Adafruit Adaloggers and Adafruit HUZZAH32 boards from this computer with the same USB cable and never experienced any issues. I narrowed it down to a hardware issue within the board. 

I then attempted to contact WeMos by joining their forums but never received a response to my registration and am now stuck with 5 overheating WeMos LoLin32 Pro's. 

But thank you for your help. I looked through the code and what you said made a lot of sense. 

Regards, 
cterna

OneForSorrow, 2018-02-05:
I also had the same problem. And when I tried the SD_MMC it worked fine.

tueddy, 2018-02-05:
Hi cterna,
i have the same problem with LoLin32 Pro now, suddenly they get very hot when plugged in to USB. I assume they have a hardware design problem- it seems have nothing to do with this SD card issue. I cannot test here anylonger, the two LoLin32 Pro burns like hot..
All i can say the SD_MMC worked before.
Dirk

OneForSorrow, 2018-02-05:
Hi Dirk,

I had the same problem once with the  lolin32 pro. My code was about bluetooth communications. I am not sure it has to do with the SD part. Maybe there is something fishy going on like there is something bugging or causes it to overload. I am not the best guy about these things.

Gaai, 2023-04-22:
Just in case other people are looking for this and land here. The current Lolin32 Pro v2 board has it's onboard sd card CS/SS pin connected to IO pin 4. Sd card sketches work fine, using SD.begin(4). The CS label for pin 5 might be a bit confusing there.