espressif/arduino-esp32 #965: attachInterrupt() callback methods with C++ class for ESP32?

huming2207, 2018-01-05:
Hi all,

I was very confused about this issue for a few hours. I've searched this issue but for AVR micro controllers, the callback method must be static, or some other tricky workarounds are needed. So I would like to ask that is there any better solutions for implementing non-static method as callback?

Regards,
Jackson

ferchinas, 2018-01-14:
Hi, I am looking for the same information. @ huming2207 Did you find something?
For the ESP8266 I have successfully used this form https://github.com/esp8266/Arduino/issues/1925
But I can not find a way for ESP32.

atanisoft, 2018-01-14:
I couldn't find a way to do this without static. I opted for a struct with methods instead and templates on the methods for instance specific params. It works pretty well for my usage.

ferchinas, 2018-01-15:
@atanisoft I do not know how to do what you mention, can you pass me some link that is explained with that?
Thank you very much

atanisoft, 2018-01-15:
@ferchinas Here is how I am doing it today... 

This function:
https://github.com/atanisoft/DCCppESP32/blob/master/src/SignalGenerator.cpp#L169

gets used for two ISRs by using the template parameter to pass in an index which points back to a struct which holds parameters for the ISR routine.  The call to attach the ISR is here: https://github.com/atanisoft/DCCppESP32/blob/master/src/SignalGenerator.cpp#L225

The timers are used to generate a square wave with specific durations based on the data being fed into the dcc signal generator.

closed 2019-08-01 by huming2207

sauvant, 2019-11-08:
I would like to reopen this issue.