espressif/arduino-esp32 #968: A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

stormbytes, 2018-01-05:
Have a new (ai-thinker.com) ESP-32 dev board. Board powers up and blinks blue LED. All kinds of output in terminal. I've installed NEW drivers, installed arduino-esp32 core, selected the /dev/cu.SLAB_USBtoUART port and still no upload. 

This is issue #333 which was previously closed

esptool.py v2.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

When I try to upload the sketch it does appear that 'something' is going on there as the blinking LED stops blinking then lags and blinks, etc

I'm using a USB cable to connect to a Mac running the 1.8.5 IDE

message2prateek, 2018-01-12:
I have a WeMos LoLin32 board and I get the same error.

message2prateek, 2018-01-12:
It is now working for me, not sure what exactly solved the issues but I did the following

a) install drivers manually
b) manually installed Python 2.7 and selected the option to add Python to the Path environment variable in the installation process. 
c) changed the USB cable.
d) restarted my PC. ( I am on a Windows 10 machine)

I installed these drivers 
https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers
http://www.ftdichip.com/Drivers/D2XX.htm (Under " D2XX Drivers" heading there is a table with different OS'es. The last column is called "comments", in there is present the link to the "setup executable" 

More info https://dl.espressif.com/doc/esp-idf/latest/get-started/establish-serial-connection.html

Once I restarted my PC and connected the ESP32 board, I could "Silicon Labs CP210x USB to UART Bridge" appear under "Device Manager" -> "Ports".

Hope this helps.

a0908896067, 2018-03-12:
Hi for those who still have this issue, you can replace USB cable and test again. I had same issue like you before and just fixed this issue last night and link below have detail test report. Now my NodeMCU-32S Lua ESP32 development module (from AiThinker, no need to press BOOT or RST key before press Arduino upload function) can use Aduino IDE on both Windows and iMac.

https://www.blogger.com/blogger.g?blogID=7530656930048997875#editor/target=post;postID=1747782269863974052;onPublishedMenu=allposts;onClosedMenu=allposts;postNum=0;src=postname

rabinpar, 2018-04-02:
can i get help for this error: "A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header!"
i am using esp32 kit in windows 10  and using cable to connect! 
and also i am using Arduino IDE 1.8.5

a0908896067, 2018-04-02:
Hi Rabinpar,

In my case, the reason I got the error as you mentioned above is because of low quality high impedance USB cable. After use another USB cable, no issues so far.  So you can try other USB cables, maybe it works

rabinpar, 2018-04-03:
Thanks for your reply but i tried with other USB cable also but same
problem arise! Is this problem arise due to any defects in ESP32?

Salty-Doug, 2018-07-17:
I have the same problem with Arduino IDE V1.8.6 hourly build, esptools.py V2.3.1 on a Mac OsX 10.13.5 (latest) and a HiLetGo ESP32 Dev Board. I have tried two manufacturers of USB cables. by pushing the reset and boot buttons I can get the upload to complete most times.

Dawge, 2018-07-17:
Hello,
I had the same problem with my ESP32 WROOM. I used a USB hub and Arduino IDE coudn't upload.
I tried to upload without the hub and it worked properly.
Hope it helped.

Salty-Doug, 2018-07-18:
Thanks Dawge. I am not using a USB hub. I saw in another thread that adding a larger 3.3v power supply helped so I added a 3.3v PS to the breadboard but I still have the problem.

Leandro-G-Ribeiro, 2018-08-04:
ONLY PRESS BUTTON "BOOT" WHEN SHOW conecting...

kp1851, 2018-09-11:
Yes Leandro-G-Ribeiro "ONLY PRESS BUTTON "BOOT" WHEN SHOW connecting…" is the solution.

eziosoma, 2018-11-29:
This solution is not accettable when you have deveoped a device with ESP32 kit in a Box...
You have to "repeat" externally to the box the boot button.
I have two esp32doit-devkit-v1 and one uploads without pressing boot button, the other one needs that the boot button is pressed. Why?

pjvander, 2019-01-07:
I ran into the same issue and couldn't resolve for the life of me. Turns out I have been shipped (twice, now) an ESP8266 in an ESP32 packaging. Configuring for the ESP8266 solved my issue, naturally. Throwing this out there in case this same thing happened to others.

closed 2019-08-15 by stale[bot]

mehtaSmitK, 2019-09-22:
Just make sure whether following conditions meet,
1. Check if GPIO 0(**IO0**)) is connected with Ground(**gnd**),
2. Press & Hold  **rst**  button  when you are uploading the code,
3. In my case, I had connected 5V output instead of 3 .3V output.