espressif/arduino-esp32 #969: get.py does not download compiler etc.

McMornan, 2018-01-06:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 with integrated oled 1306 (CP210x serial uart)
Core Installation/update date:			?11/jul/2017?
IDE name:							1.8.5 on Armbian on an Orange Pi 2 plus E
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
I install the esp32 support files, by following the instructions for Linux on Arm (Ubuntu 16.04 LTS)  to the letter:

everything works fine, and get.py completes successfully. However the compiler and other tools are not downloaded. The only file which gets downloaded is mkspiffs. Consequently when I try to compile anything (e.g. the simple default sketch below) the Arduino IDE tells me that:

fork/exec /home/vnc/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++: no such file or directory

any ideas ?

thanks,
Lars

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2018-01-06:
I don't think there has been a lot of demand for the toolchain on ARM. http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup-scratch.html

McMornan, 2018-01-06:
thanks, I will try to create it from scratch - the reason why I want to do that is, that I have a number of  ESP32 boards with 1306 oled - unfortunately they use an CP210x uart which does not seem to be properly supported under win10 - maybe its time to setup a dedicated linux environment on x86 if that orange thing does not work ...

lbernstone, 2018-01-06:
A cp2102 should also work fine with a linux virtual machine (virtualbox.org)

BlrUsr, 2018-01-24:
I have same problem on armbian (ubuntu) , anybody  decide this problem?

yoursunny, 2018-08-18:
Duplicates #789.

closed 2019-08-15 by stale[bot]