espressif/arduino-esp32 #970: SNTP not working with name server

thubaggins, 2018-01-06:
### Hardware: ESP12-E
Board:							ESP32 Dev Module
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					921600

Doesn't work
```
#include "time.h"
void printLocalTime()
{
    struct tm timeinfo;
    if(!getLocalTime(&timeinfo)){
        Serial.println("Failed to obtain time");
        return;
    }
    Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
}

configTime(3600, 3600, "pool.ntp.org");
```

Worked
```
#include "time.h"
void printLocalTime()
{
    struct tm timeinfo;
    if(!getLocalTime(&timeinfo)){
        Serial.println("Failed to obtain time");
        return;
    }
    Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
}

configTime(3600, 3600, "69.10.161.7");
```

I think problem is dns issues?

lbernstone, 2018-01-06:
You can try this snippet to see if you have resolution problems:
`  WiFiGenericClass wfgc;
  IPAddress resolve;
  if (wfgc.hostByName("pool.ntp.org",resolve)) Serial.println(resolve);
`

thubaggins, 2018-01-09:
Sorry for the delay.  I have tried your code but it doesn't showing anything.

CelliesProjects, 2018-01-12:
@thubaggins Does this sketch work for you?
````c
#include "time.h"
#include "WiFi.h"

void printLocalTime()
{
    struct tm timeinfo;
    if(!getLocalTime(&timeinfo)){
        Serial.println("Failed to obtain time");
        return;
    }
    Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
}

void setup() {
  Serial.begin(115200);
  Serial.println( "Starting WiFi..." );
  WiFi.begin("YOURSSID","YOURPSK");

  while ( !WiFi.isConnected() ) {
    vTaskDelay( 50 / portTICK_PERIOD_MS );
  }
  
  Serial.println( "Syncing time..." );
  configTime( 3600, 3600, "nl.pool.ntp.org" );

  printLocalTime();
}

void loop() {
  // put your main code here, to run repeatedly:

}
````

closed 2018-01-14 by thubaggins