espressif/arduino-esp32 #972: Support WiPy 2.0 ?

sandmen, 2018-01-07:
Is it possible to flash the WiPy from pycom ?
Which board should I select ? Esp32-dev-modul ?

### Hardware:
Board:							WyPi pycom
IDE name:							Arduino IDE 1.8.5

beegee-tokyo, 2018-01-08:
No guarantee, but it should be possible. 
I checked WiPy pinout and Esp32-dev-modul would fit, however some pins are not available on the WiPy board. Flash size is the same as well.

sandmen, 2018-01-08:
Thanks for double checking.
I will try it.
To be honest, thought to put the wipy board into bootloader mode (pin23 to GND), and then using the
Arduino 1.8.5 and arduino-esp32 to "flash" bootloader.
Then, I can use Arduino-IDE for "programming" ... First project with esp32... :-(

Thanks

closed 2018-04-19 by everslick