espressif/arduino-esp32 #973: get.py produces SSL certificate verification error

DrewHoyt, 2018-01-07:
### Hardware: N/A
Board: esp32 dev board
Core Installation/update date:		7/jan/2018
IDE name:					Arduino IDE
Flash Frequency:	N/A
Upload Speed:		N/A

### Description:
attempting to get tools set up for use with the Arduino IDE. When I run get.py I receive the following error:

`IOError: [Errno socket error] [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:661)`

Looks to me like there is there is something up with the SSL certificate?

detailed traceback:

`[dhoyt@localhost tools]$ python get.py
System: Linux, Info: Linux-4.14.11-300.fc27.x86_64-x86_64-with-fedora-27-Twenty_Seven
Platform: x86_64-pc-linux-gnu
Downloading xtensa-esp32-elf-linux64-1.22.0-75-gbaf03c2-5.2.0.tar.gz
Traceback (most recent call last):
  File "get.py", line 139, in <module>
    get_tool(tool)
  File "get.py", line 94, in get_tool
    urlretrieve(url, local_path, report_progress)
  File "/usr/lib64/python2.7/urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "/usr/lib64/python2.7/urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "/usr/lib64/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/usr/lib64/python2.7/urllib.py", line 443, in open_https
    h.endheaders(data)
  File "/usr/lib64/python2.7/httplib.py", line 1038, in endheaders
    self._send_output(message_body)
  File "/usr/lib64/python2.7/httplib.py", line 882, in _send_output
    self.send(msg)
  File "/usr/lib64/python2.7/httplib.py", line 844, in send
    self.connect()
  File "/usr/lib64/python2.7/httplib.py", line 1263, in connect
    server_hostname=server_hostname)
  File "/usr/lib64/python2.7/ssl.py", line 363, in wrap_socket
    _context=self)
  File "/usr/lib64/python2.7/ssl.py", line 611, in __init__
    self.do_handshake()
  File "/usr/lib64/python2.7/ssl.py", line 840, in do_handshake
    self._sslobj.do_handshake()
IOError: [Errno socket error] [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:661)
`

DrewHoyt, 2018-01-17:
Re-ran this week and SSL certificate issue seems to have worked itself out. No changes in my set up, but get.py seems to be functioning properly now.

closed 2018-01-17 by DrewHoyt

WayneKeenan, 2018-04-17:
still occurs as of today.

urllib.error.URLError: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:749)>

WayneKeenan, 2018-04-17:
A workaround on mac/linux to JFDI:

Edit line 89 (or thereabouts to display there `url` instead of the `archive_name`):
```python
print('Downloading ' + url)
```

Then, for each download that fails run `get -P `/dist <the url>`

e.g.:

```bash

wget -P ./dist https://github.com/igrr/mkspiffs/releases/download/0.2.2/mkspiffs-0.2.2-arduino-esp32-osx.tar.gz
```

Then re-run `get.py` till it's happy.

StefanHri, 2019-09-18:
I have the same problem by installing the esp tools. My set up:
1)esp-edf (commit 9f145ff165099f32b92b7ee022a1debc5f0a583d) from Sep. 12. 2019
2)Running install.sh on linux strech causes:
Installing ESP-IDF tools
Installing tools: xtensa-esp32-elf, esp32ulp-elf, openocd-esp32
Installing xtensa-esp32-elf@esp32-2019r1-8.2.0
Downloading xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-amd64.tar.gz to /root/.espressif/dist/xtensa-esp32-elf-gcc8_2_0-esp32-2019r1-linux-amd64.tar.gz.tmp
Traceback (most recent call last):
  File "/opt/esp/esp-idf/tools/idf_tools.py", line 1317, in <module>
    main(sys.argv[1:])
  File "/opt/esp/esp-idf/tools/idf_tools.py", line 1313, in main
    action_func(args)
  File "/opt/esp/esp-idf/tools/idf_tools.py", line 1107, in action_install
    tool_obj.download(tool_version)
  File "/opt/esp/esp-idf/tools/idf_tools.py", line 541, in download
    urlretrieve(url, local_temp_path, report_progress if not global_non_interactive else None)
  File "/usr/lib/python2.7/urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "/usr/lib/python2.7/urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "/usr/lib/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/usr/lib/python2.7/urllib.py", line 443, in open_https
    h.endheaders(data)
  File "/usr/lib/python2.7/httplib.py", line 1038, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 882, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 844, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 1263, in connect
    server_hostname=server_hostname)
  File "/usr/lib/python2.7/ssl.py", line 363, in wrap_socket
    _context=self)
  File "/usr/lib/python2.7/ssl.py", line 611, in __init__
    self.do_handshake()
  File "/usr/lib/python2.7/ssl.py", line 840, in do_handshake
    self._sslobj.do_handshake()
IOError: [Errno socket error] [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:661)