espressif/arduino-esp32 #975: Can't read input data form pins if use WiFi

stas0, 2018-01-07:
Hi!

I one pin for read temperature from several ds18b20 sensors and one pin for read data from analog input.

All errors starting after I started use WiFi connection. Before using WiFi I used G22 and G4 pins and all works great. After I started use  WiFi(WiFi.begin(ssid, password)), none of pins not working. 
I found that this link https://github.com/espressif/arduino-esp32/issues/825
Its says that if I use WiFi I need 

> use pins 32 and above when WiFi is ON

Why???
What about rest pins???
Ok. For test I use G34 and G35 pins. Analog pin works great. but pin for temperature not woking. 
I aloso added cli() and sei() between reading sensors data, but it also not working.

Whats wrong?

stickbreaker, 2018-01-07:
@stas0 
`cli()` and `sei()` have no effect on the ESP32. I have modified the `OneWire.h` library to be compatible with the ESP32 here: [esp32 OneWire.h](https://github.com/stickbreaker/OneWire) 

Try it. 

Don't know about your Analog input problems.

Chuck.

stas0, 2018-01-07:
@stickbreaker 
http://prntscr.com/hxb53s
Very nice, nice!
At screen you can see that all works great! I used previous version and got some strange effect. Because I use only one pin for several sensors old version return wrong data. Current version working great.

stas0, 2018-01-07:
It takes years of testing the board, it's still very raw.

closed 2018-04-19 by everslick