espressif/arduino-esp32 #976: mbedtls with hardware SHA engine

yoursunny, 2018-01-07:
### Hardware:
Board:							Heltec WiFi Kit 32
Core Installation/update date:			70d0d4648766cea047613062ed50a9ecfc9de31f
IDE name:							Arduino 1.8.5
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
[`CONFIG_MBEDTLS_HARDWARE_SHA`](https://github.com/espressif/arduino-esp32/blob/70d0d4648766cea047613062ed50a9ecfc9de31f/tools/sdk/sdkconfig#L428) option is disabled.
As a result, SHA256 computation is slow.

### Sketch:
```cpp
#include <mbedtls/sha256.h>

void setup() {
  Serial.begin(115200);
  Serial.println();
}

void loop() {
  unsigned long t0 = micros();
  mbedtls_sha256_context ctx;
  mbedtls_sha256_init(&ctx);
  for (int i = 0; i < 16777216; ++i) {
    Serial.printf("mode=%d\n", ctx.mode);
    mbedtls_sha256_update(&ctx, reinterpret_cast<const uint8_t*>("abcdefghbcdefghicdefghijdefghijkefghijklfghijklmghijklmnhijklmno"), 64);
  }
  uint8_t digest[32];
  mbedtls_sha256_finish(&ctx, digest);
  mbedtls_sha256_free(&ctx);
  unsigned long t1 = micros();

  for (int i = 0; i < sizeof(digest); ++i) {
    Serial.printf("%02x", digest[i]);
  }
  Serial.println();
  Serial.printf("duration=%lu\n", t1 - t0);
}
```

### Debug Messages:
In "alt" version of `mbedtls_sha256_context`, there is a `mode` member that indicates whether hardware or software SHA engine is selected.
The sketch does not compile because the `mbedtls_sha256_context` only supports software implementation.
```
sha256-mbedtls:13: error: 'struct mbedtls_sha256_context' has no member named 'mode'

     Serial.printf("mode=%d\n", ctx.mode);
```

copercini, 2018-01-08:
It has disabled due an old bug in hardware acceleration that crash mbedltls sometimes (https://github.com/espressif/esp-idf/issues/630)
It's fixed on IDF now and just waiting for the update for Arduino here: https://github.com/espressif/arduino-esp32/issues/964

yoursunny, 2018-08-18:
Core Installation/update date: 172802b18e3540e39371d5125ba4dc70d07046e7
Now that #964 is closed, but `CONFIG_MBEDTLS_HARDWARE_SHA` [stays disabled](https://github.com/espressif/arduino-esp32/blob/172802b18e3540e39371d5125ba4dc70d07046e7/tools/sdk/sdkconfig#L525).
Can this be enabled?

I tried modifying `tools/sdk/sdkconfig` but it does not seem to be effective. I don't know how to regenerate binaries in there.

yoursunny, 2019-08-01:
This issue should remain open. Enabling hardware SHA engine is a useful performance improvement.

TomZanna, 2020-04-11:
Please, enable the hardware SHA engine.

lbernstone, 2020-04-11:
While it certainly does accelerate the hashing by about 4x, I get stack smashing issues when I compile with CONFIG_MBEDTLS_HARDWARE_SHA=y and test @copercini 's code.

yoursunny, 2020-06-11:
.

yoursunny, 2020-08-10:
.

yoursunny, 2020-10-09:
.

omerahmedd, 2020-12-03:
any update on this ?

how to enable sha256 hardware acceleration on esp32

yoursunny, 2021-02-02:
.

yoursunny, 2021-06-20:
.

yoursunny, 2021-08-21:
.

hash13, 2022-03-17:
pls solve

igrr, 2022-03-18:
Should be fixed by https://github.com/espressif/esp32-arduino-lib-builder/pull/62/commits/d65e071c5258b7228fef05e274ba2e7ca9df7a49 — this will be merged into arduino-esp32 as part of https://github.com/espressif/arduino-esp32/pull/6341 and will be part of the next release.

VojtechBartoska, 2022-03-30:
I'm closing this as mentioned PR was merged. If you still face some issues regarding this, you can reopen it. Thanks to all for your contribution.

closed 2022-03-30 by VojtechBartoska