espressif/arduino-esp32 #977: Simultaneously use the two I2C bus of the ESP32.

BlackTiger44, 2018-01-07:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Hello.

I am doing a test program in Arduino for ESP32 in which I would like to use the two I2C buses that I have, separately I can read and write data, but what I can not do in my program is being able to change from one I2C bus to another to send or receive data, I have used the wire.h library that comes for it in Arduino.

Greetings BlackTiger44.

beegee-tokyo, 2018-01-08:
You might define two Wire ports, and then access them as needed by their name.
```cpp
TwoWire I2Cone = TwoWire(0);
TwoWire I2Ctwo = TwoWire(1);

void setup() {
  I2Cone.begin(21,22,100000); // SDA pin 21, SCL pin 22, 100kHz frequency
  I2Ctwo.begin(16,17,400000); // SDA pin 16, SCL pin 17, 400kHz frequency
}

void loop() {
  I2Cone.beginTransmission(0x38);
  I2Cone.write(0x00 | 0x80);
  I2Cone.write(0x03);
  I2Cone.endTransmission();

  I2Ctwo.beginTransmission(0x42);
  I2Ctwo.write(140);
  I2Ctwo.endTransmission();
  I2Ctwo.requestFrom(0x42,1);
  if (I2Ctwo.available() == 1) {
    byte value = I2Ctwo.read();
  }
}
```
Code is not tested, but that is how I would approach it.

TheFlyingZephyr, 2018-01-08:
Have you tried running each bus in its own task? If that doesn't work you might want to look into bitbanging the buses yourself, it's not as hard as it might at first seem.

- Mark West

BlackTiger44, 2018-01-09:
Hello again.
beegee-tokyo I have tested your code and it works perfectly, thank you very much for your help.
TheFlyingZephyr if I use the two buses in the same task, as my level is basic I just do not understand the idea you want to convey to me.
Although I did not explain the reason why I wanted to use the two I2C bus, I feel obliged to say it, I am making an application that uses the Blynk APP (http://www.blynk.cc/) on the one hand and on the other hand the ESP32 with two ports PCF8574, one used as digital inputs and the other as outputs, to read the input port I use an interruption, and when I perform a write operation on the output port and immediately I get the interruption of the port entries, reading the port of entries is wrong, so I want to try to put each of them on a different I2C bus.

Regards, and thank you very much
BlackTiger 44

Kutscher07, 2018-02-14:
Thanks beegee-tokyo. With your help, i could modify a I2C Scanner to this dual channel I2C Scanner. Tested with ESP32 TTGO TQ. OLED on second,  BME280 on first.
```c
/* I2C slave Address Scanner
for 5V bus
 * Connect a 4.7k resistor between SDA and Vcc
 * Connect a 4.7k resistor between SCL and Vcc
for 3.3V bus
 * Connect a 2.4k resistor between SDA and Vcc
 * Connect a 2.4k resistor between SCL and Vcc
Kutscher07: Modified for TTGO TQ board with builtin OLED
 */

#include <Wire.h>

#include <dummy.h>  //for esp32

#define SDA1 21
#define SCL1 22

#define SDA2 5
#define SCL2 4

TwoWire I2Cone = TwoWire(0);
TwoWire I2Ctwo = TwoWire(1);

void scan1(){
Serial.println("Scanning I2C Addresses Channel 1");
uint8_t cnt=0;
for(uint8_t i=0;i<128;i++){
  I2Cone.beginTransmission(i);
  uint8_t ec=I2Cone.endTransmission(true);
  if(ec==0){
    if(i<16)Serial.print('0');
    Serial.print(i,HEX);
    cnt++;
  }
  else Serial.print("..");
  Serial.print(' ');
  if ((i&0x0f)==0x0f)Serial.println();
  }
Serial.print("Scan Completed, ");
Serial.print(cnt);
Serial.println(" I2C Devices found.");

}
void scan2(){
Serial.println("Scanning I2C Addresses Channel 2");
uint8_t cnt=0;
for(uint8_t i=0;i<128;i++){
  I2Ctwo.beginTransmission(i);
  uint8_t ec=I2Ctwo.endTransmission(true);
  if(ec==0){
    if(i<16)Serial.print('0');
    Serial.print(i,HEX);
    cnt++;
  }
  else Serial.print("..");
  Serial.print(' ');
  if ((i&0x0f)==0x0f)Serial.println();
  }
Serial.print("Scan Completed, ");
Serial.print(cnt);
Serial.println(" I2C Devices found.");

}
void setup(){
Serial.begin(115200);
I2Cone.begin(SDA1,SCL1,400000); // SDA pin 21, SCL pin 22 TTGO TQ
I2Ctwo.begin(SDA2,SCL2,400000); // SDA pin 5, SCL pin 4 builtin OLED

}
void loop(){
scan1();
Serial.println();
delay(100);
scan2();
Serial.println();
delay(5000);

}

```

beegee-tokyo, 2018-02-15:
Added to the [Wiki](http://desire.giesecke.tk/index.php/2018/01/30/simultaneously-use-of-two-i2c-buses-twowire-interfaces/)

BlackTiger44, 2018-02-15:
Thanks for sharing your work Kutscher07.

Greetings BlackTiger

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

MrLight, 2018-05-07:
Can anyone confirm that the second i2c channel is really working? When I try to use the I2C Scanner example above I can only use the first TwoWire(0) Channel. When I change the SDA1 / 2 and SCL1 / 2 the scanner detects my second I2C String but I can't use two at the same time. Second one is always empty. Even when I Change 
```
TwoWire I2Cone = TwoWire(0);
TwoWire I2Ctwo = TwoWire(1);
```

to

```
TwoWire I2Cone = TwoWire(1);
TwoWire I2Ctwo = TwoWire(0);
```
only the I2Cone Channel is successfully used.

Hobelbank, 2019-09-17:
I keep trying to get ESP32 and BME_280 working on a second I2C-Bus using <Adafruit_BME280.h>
and <Wire.h>. Scanning both buses works fine. BME_280.begin(uint8_t addr, TwoWire *theWire) does not work --> no matching function for call to 'Adafruit_BME280::begin(uint8_t&, TwoWire&)'

Any solutions?

I've tried different BME_280 libraries. They all do not seem to work for ESP32.

TimTurnerResearch, 2020-02-21:
In the Wire.cpp lib for the ESP32 there are already two available I2C ports:
TwoWire Wire = TwoWire(0);
TwoWire Wire1 = TwoWire(1);

To use in your project just initialise  separately e.g.
Wire.begin(16, 17, 400000);
Wire1.begin(21, 22, 400000);

then use as usual.

romatou18, 2020-05-04:
Thanks guys ! I was almost there this put me on the way !

dburenok, 2020-07-30:
> In the Wire.cpp lib for the ESP32 there are already two available I2C ports:
> TwoWire Wire = TwoWire(0);
> TwoWire Wire1 = TwoWire(1);
> 
> To use in your project just initialise separately e.g.
> Wire.begin(16, 17, 400000);
> Wire1.begin(21, 22, 400000);
> 
> then use as usual.

This solved the problem for me! Thank you Tim!!

poojan88, 2021-01-07:
I am trying to use two TCA9548 multiplexers for 11 ADXL on esp32 in two I2c ports .
I get values only for the first multiplexer with default GPIO 21(SDA) and GPIO 22(SCL). While I try for the other I2c port I do not get values for ADXLs. I am not able to figure out where the problem lies.
`#include "Wire.h"
#include <SparkFun_ADXL345.h>        

#define SDA_1 21
#define SCL_1 22

#define SDA_2 33
#define SCL_2 32

TwoWire I2Cone = TwoWire(0);
TwoWire I2Ctwo = TwoWire(1);

/*********** COMMUNICATION SELECTION ***********/
ADXL345 adxl1 = ADXL345();         
ADXL345 adxl2 = ADXL345();
ADXL345 adxl3 = ADXL345();         
ADXL345 adxl4 = ADXL345();
ADXL345 adxl5 = ADXL345();         
ADXL345 adxl6 = ADXL345();
ADXL345 adxl7 = ADXL345();         
ADXL345 adxl8 = ADXL345();
ADXL345 adxl9 = ADXL345();         
ADXL345 adxl10 = ADXL345();
ADXL345 adxl11 = ADXL345();  

#define TCAADDR1 0x70
void tcaselect_1(uint8_t bus) {
  if (bus > 7) return;
 
  Wire.beginTransmission(TCAADDR1);
  Wire.write(1 << bus);
  Wire.endTransmission();  
}

#define TCAADDR2 0x7
void tcaselect_2(uint8_t i) {
  if (i > 7) return;
 
  I2Ctwo.beginTransmission(TCAADDR2);
  I2Ctwo.write(1 << i);
  I2Ctwo.endTransmission();  
}

/******************** SETUP ********************/
/*          Configure ADXL345 Settings         */
void setup(){
      
  Serial.begin(115200); 

  I2Cone.begin(SDA_1,SCL_1,400000); // SDA pin 21, SCL pin 22 TCA1
  I2Ctwo.begin(SDA_2,SCL_2,400000); // SDA pin 33, SCL pin 32 TCA2
                  
/***** MUX 1 *****/

  tcaselect_1(1);
  adxl1.powerOn();                    
  adxl1.setRangeSetting(2);
  
  tcaselect_1(2);
  adxl2.powerOn();                     
  adxl2.setRangeSetting(2); 

  tcaselect_1(3);
  adxl3.powerOn();                    
  adxl3.setRangeSetting(2);
  
  tcaselect_1(4);
  adxl4.powerOn();                     
[... 138 lines cut ...]
   
  tcaselect_2(4);
  // Accelerometer Readings
  int x10,y10,z10;   
  adxl10.readAccel(&x10, &y10, &z10);        
  // Output Results to Serial
  Serial.print("X10: "); Serial.print(x10); Serial.print("  ");
  Serial.print("y10: "); Serial.print(y10); Serial.print("  ");
  Serial.print("z10: "); Serial.print(z10); Serial.print("  ");
  Serial.println("m/s^2 ");
   delay(500);

   tcaselect_2(5);
  // Accelerometer Readings
  int x11,y11,z11;   
  adxl11.readAccel(&x11, &y11, &z11);         // Read the accelerometer values and store them in variables declared above x,y,z
  // Output Results to Serial
  Serial.print("X11: "); Serial.print(x11); Serial.print("  ");
  Serial.print("y11: "); Serial.print(y11); Serial.print("  ");
  Serial.print("z11: "); Serial.print(z11); Serial.print("  ");
  Serial.println("m/s^2 ");
   delay(500);

} `
I have also tried to interface 6 adxls separately with the twoWire instance but ,no luck! Please do respond

andressakreutz, 2021-04-14:
> Thanks beegee-tokyo. With your help, i could modify a I2C Scanner to this dual channel I2C Scanner. Tested with ESP32 TTGO TQ. OLED on second, BME280 on first.
>
> ```c
> /* I2C slave Address Scanner
> for [...]

Hi, I am trying to use this code with a board ESP32 38 pins (like this https://produto.mercadolivre.com.br/MLB-1717757954-esp-32-38-pinos-_JM). I'm sorry my ignorance, but when you defined "#define SDA2 5, #define SCL2 4", that's mean that I must connect the SDA from display in the GIOP5 (pin 29 in this board) or in the pin 5 (GIOP34 in this board)? Thanks.

JonathanMortes, 2021-06-25:
Hi, I'm having the same issues. Have you guys fixed it?

atipezda, 2021-07-01:
Same here using ESP32 Devkit v1 using port 33 and 32

Yassinebouothman, 2023-05-01:
@BlackTiger44 You just need to create 2 I2C objects like this and specify port and frequency, like this : 
```
// 1 is first port and 2 is second port
#include <Wire.h>

TwoWire WIRE = TwoWire(1, I2C_FAST_MODE); 
TwoWire WIRE1 = TwoWire(2, I2C_FAST_MODE);
```
then you can use WIRE and WIRE1 in your code.