espressif/arduino-esp32 #979: ESP32 Deep Sleep/TimerWakeUp Example stops after 2-3 few hours

mike-rankin, 2018-01-08:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		8/JAN/2018
IDE name:						Arduino IDE 1.8.0
Flash Frequency:					80Mhz
Upload Speed:					115200

### Description:
Deep Sleep Timer WakeUp Sketch wakes up the ESP32 every 15 minutes for only two-three hours then stops. I have a 3.3V LDO, ESP32 and an LED on an I/O pin that blinks the led every 15 minutes. Two identical pieces of hardware are assembled but both stop blinking the led around the three hour mark. LiPO battery voltage is good and sleep/awake current measurements are good. If you could connect an LED to any I/O pin and run this blink sketch over a few hours you should see this problem.

### Sketch:

/*
Simple Deep Sleep with Timer Wake Up
=====================================
ESP32 offers a deep sleep mode for effective power
saving as power is an important factor for IoT
applications. In this mode CPUs, most of the RAM,
and all the digital peripherals which are clocked
from APB_CLK are powered off. The only parts of
the chip which can still be powered on are:
RTC controller, RTC peripherals ,and RTC memories

This code displays the most basic deep sleep with
a timer to wake it up and how to store data in
RTC memory to use it over reboots

This code is under Public Domain License.

Author:
Pranav Cherukupalli <cherukupallip@gmail.com>
*/

#define uS_TO_S_FACTOR 1000000  /* Conversion factor for micro seconds to seconds */
#define TIME_TO_SLEEP  30        /* Time ESP32 will go to sleep (in seconds) */

#define Orange_Led 33

RTC_DATA_ATTR int bootCount = 0;

/*
Method to print the reason by which ESP32
has been awaken from sleep
*/
void print_wakeup_reason(){
  esp_sleep_wakeup_cause_t wakeup_reason;

  wakeup_reason = esp_sleep_get_wakeup_cause();

  switch(wakeup_reason)
  {
    case 1  : Serial.println("Wakeup caused by external signal using RTC_IO"); break;
    case 2  : Serial.println("Wakeup caused by external signal using RTC_CNTL"); break;
    case 3  : Serial.println("Wakeup caused by timer"); break;
    case 4  : Serial.println("Wakeup caused by touchpad"); break;
    case 5  : Serial.println("Wakeup caused by ULP program"); break;
    default : Serial.println("Wakeup was not caused by deep sleep"); break;
  }
}

void setup()
{
  Serial.begin(115200);
  delay(1000); //Take some time to open up the Serial Monitor

  pinMode(Orange_Led, OUTPUT);

  //Increment boot number and print it every reboot
  ++bootCount;
  Serial.println("Boot number: " + String(bootCount));

  //Print the wakeup reason for ESP32
  print_wakeup_reason();

  /*
  First we configure the wake up source
  We set our ESP32 to wake up every 5 seconds
  */
  esp_sleep_enable_timer_wakeup(TIME_TO_SLEEP * uS_TO_S_FACTOR);
  Serial.println("Setup ESP32 to sleep for every " + String(TIME_TO_SLEEP) +
  " Seconds");

  /*
  Next we decide what all peripherals to shut down/keep on
  By default, ESP32 will automatically power down the peripherals
  not needed by the wakeup source, but if you want to be a poweruser
  this is for you. Read in detail at the API docs
  http://esp-idf.readthedocs.io/en/latest/api-reference/system/deep_sleep.html
  Left the line commented as an example of how to configure peripherals.
  The line below turns off all RTC peripherals in deep sleep.
  */
  //esp_deep_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_OFF);
  //Serial.println("Configured all RTC Peripherals to be powered down in sleep");

  /*
  Now that we have setup a wake cause and if needed setup the
  peripherals state in deep sleep, we can now start going to
  deep sleep.
  In the case that no wake up sources were provided but deep
  sleep was started, it will sleep forever unless hardware
  reset occurs.
  */
  
  Fast_Blink_LED();
  
  Serial.println("Going to sleep now");
  esp_deep_sleep_start();
  Serial.println("This will never be printed");
}

void loop()
{
  //This is not going to be called
}

void Fast_Blink_LED(void)
{
 digitalWrite(Orange_Led, HIGH);   
 delay(200);                       
 digitalWrite(Orange_Led, LOW);    
 delay(200);
 digitalWrite(Orange_Led, HIGH);   
 delay(200);                      
 digitalWrite(Orange_Led, LOW);    
 delay(200); 
}

mike-rankin, 2018-01-08:
#define TIME_TO_SLEEP 900 //changing the TIME_TO_SLEEP to 900 should = 15 minutes

cristianhumelnicu, 2018-01-10:
@mike-rankin : Try to read this ... 

https://github.com/espressif/arduino-esp32/issues/796

closed 2019-08-15 by stale[bot]