espressif/arduino-esp32 #980: esp32 tone

gemforce, 2018-01-08:
Hi
Why not working tone in esp32 ?

       tone(SOUND_PIN, 1000, 1000);
       delay(1000);
        noTone(SOUND_PIN)

lbernstone, 2018-01-08:
I think tone() will be implemented in the ESP-IDF, rather than arduino.  In the meantime, try https://techtutorialsx.com/2017/07/01/esp32-arduino-controlling-a-buzzer-with-pwm/

ReanimationXP, 2018-09-08:
This is part of the Arduino core library and needs to be implemented.

lbernstone, 2018-09-08:
I don't think this is of a quality to be added to the code base, but it is a start, anyhow.
https://github.com/lbernstone/Tone

Thomascountz, 2021-02-21:
Although this functionality is unavailable in Espressif's [arduino-esp32](https://github.com/espressif/arduino-esp32) library, members of the community have found various [work-arounds](https://github.com/lbernstone/Tone) such as using the native [LED Control](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/ledc.html) functions to generate PWM signals.

However, looking more closely at [arduino-esp32](https://github.com/espressif/arduino-esp32) library, not only has Espressif provided a clean API for generating tones, they've provided an interface for generating specific PWM frequencies for specific notes on the chromatic scale in different octaves.

https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-ledc.c#L255-L266

```c
double ledcWriteNote(uint8_t chan, note_t note, uint8_t octave){
  const uint16_t noteFrequencyBase[12] = {
  //   C        C#       D        Eb       E        F       F#        G       G#        A       Bb        B
      4186,    4435,    4699,    4978,    5274,    5588,    5920,    6272,    6645,    7040,    7459,    7902
  };

  if(octave > 8 || note >= NOTE_MAX){
      return 0;
  }
  double noteFreq =  (double)noteFrequencyBase[note] / (double)(1 << (8-octave));
  return ledcWriteTone(chan, noteFreq);
}

```

Although not directly compatible with Arduino's `tone()`, the function allows for a clearer interface by providing named frequencies out of the box via the `note_t` type.

```c
typedef enum {
    NOTE_C, NOTE_Cs, NOTE_D, NOTE_Eb, NOTE_E, NOTE_F, NOTE_Fs, NOTE_G, NOTE_Gs, NOTE_A, NOTE_Bb, NOTE_B, NOTE_MAX
} note_t;
```

## Example Usage
Here is an example of generating PWM frequencies at 50% duty cycle to play the C-major scale.

```c
void playScale(uint8_t pin, uint8_t channel) {
  ledcAttachPin(pin, channel);
  ledcWriteNote(pin, NOTE_C, 4);
  delay(500);
  ledcWriteNote(pin, NOTE_D, 4);
  delay(500);
  ledcWriteNote(pin, NOTE_E, 4);
  delay(500);
  ledcWriteNote(pin, NOTE_F, 4);
  delay(500);
  ledcWriteNote(pin, NOTE_G, 4);
  delay(500);
  ledcWriteNote(pin, NOTE_A, 4);
  delay(500);
  ledcWriteNote(pin, NOTE_B, 4);
  delay(500);
  ledcDetachPin(pin)
}
```

lbernstone, 2021-02-21:
analogWrite and tone are both available in the [ESP32Servo library](https://github.com/madhephaestus/ESP32Servo).  This can be found in the Arduino Library Manager.

mmackh, 2021-04-18:
For anyone coming across @Thomascountz Example Usage code, the first argument in ledcWriteNote is a channel, not the pin. So the code could look like:

```
void playScale(uint8_t pin, uint8_t channel) {
  uint8_t octave = 4;
  
  ledcAttachPin(pin, channel);
  ledcWriteNote(channel, NOTE_C, octave);
  delay(500);
  ledcWriteNote(channel, NOTE_D, octave);
  delay(500);
  ledcWriteNote(channel, NOTE_E, octave);
  delay(500);
  ledcWriteNote(channel, NOTE_F, octave);
  delay(500);
  ledcWriteNote(channel, NOTE_G, octave);
  delay(500);
  ledcWriteNote(channel, NOTE_A, octave);
  delay(500);
  ledcWriteNote(channel, NOTE_B, octave);
  delay(500);
  ledcDetachPin(pin);
}
```

aldityarafi, 2022-01-11:
> analogWrite and tone are both available in the [ESP32Servo library](https://github.com/madhephaestus/ESP32Servo). This can be found in the Arduino Library Manager.

hey man, you just saved me from depression because of this...buzzer . Thanks

PR #6402 (2022-03-10): Implemented tone and noTone; fixes #980
Issue #980 requested tone to be implemented.
[tone in Arduino documentation](https://www.arduino.cc/reference/en/language/functions/advanced-io/tone/)
[noTone in Arduino documentation](https://www.arduino.cc/reference/en/language/functions/advanced-io/notone/)

## Summary
This PR provides new implementation of functions tone and noTone to be used with esp32.

## Impact
New functions are written with almost the same API as Arduino original functions. This implementation adds the option to change duty (default is 50% the same as in Arduino).

Closes #980
commit 68daea4a (2022-03-10): Implemented tone and noTone; fixes #980 (#6402)

* Implemented tone
* Tone uses queue; implemented setToneChannel
closed 2022-03-10 by me-no-dev