espressif/arduino-esp32 #982: Arduino not compatible as a component with esp-idf v2.1.1

forthlightning, 2018-01-09:
Compile error for arduino as a component, serial output as follows:

```
CC cores/esp32/libb64/cencode.o
CC cores/esp32/esp32-hal-adc.o
CC cores/esp32/esp32-hal-bt.o
CC cores/esp32/esp32-hal-dac.o
CC cores/esp32/esp32-hal-gpio.o
CC cores/esp32/esp32-hal-i2c.o
CC cores/esp32/esp32-hal-ledc.o
CC cores/esp32/esp32-hal-matrix.o
CC cores/esp32/esp32-hal-misc.o
CC cores/esp32/esp32-hal-sigmadelta.o
CC cores/esp32/esp32-hal-spi.o
CC cores/esp32/esp32-hal-time.o
CC cores/esp32/esp32-hal-timer.o
CC cores/esp32/esp32-hal-touch.o
CC cores/esp32/esp32-hal-uart.o
CC cores/esp32/stdlib_noniso.o
CC cores/esp32/wiring_pulse.o
CC cores/esp32/wiring_shift.o
CC libraries/SD/src//sd_diskio_crc.o
CXX cores/esp32/Esp.o
/Users/maxfieldmorrison/NSR/solsense-esp32/components/arduino-esp32/cores/esp32/Esp.cpp:23:23: fatal error: esp_sleep.h: No such file or directory
compilation terminated.
```

I tried to change the `esp_sleep.h` to `esp_deep_sleep.h` which does exist, but then it ran into several other build errors.

I have checked out v2.1.1 for esp-idf and using the most recent arduino commit... not sure what I am missing here, any help would be appreciated :D

everslick, 2018-04-19:
Is this problem still unsolved for you?

copercini, 2018-04-21:
Solved in the last update =)

closed 2018-04-21 by copercini