espressif/arduino-esp32 #983: upload hangs on older Mac configuration (Mac Pro 4,1 w/ El Capitan)

JohnGEyles, 2018-01-09:
### Hardware:
Board:							Adafruit HUZZAH32
Core Installation/update date:			8Jan2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
Got core working on mid-2011 Macbook Air running High Sierra with no difficulties.   Trying to get it running on Mac Pro 2009 4,1 w/ El Capitan now.   When I upload sketch it hangs.   The upload never completes, and odd effects are seen on the computer.   For example, System Profiler ("About This Mac")  fails to run, even after I quit Arduino IDE (Activity Monitor says it's "not responding").   Activity Monitor also shows "kernel_task" (PID 0) is using 100+% of CPU (even after I quit Arduino IDE).     Rebooting seems to be only way to get OSX back to normal.

### Sketch:

void setup() {
    Serial.print ("Hello world");
}

void loop() {
}

### Debug Messages:
nothing

lbernstone, 2018-01-10:
This sounds like a problem with esptool, not arduino.  Try running `python esptool.py chip_id` from the command line and see if it gives you more info about the failure.

JohnGEyles, 2018-01-11:
Thanks.  Yeah, I figured with the ESP stuff too.   It was suggested at an Adafruit forum that it could be a version compatibility issue, but it's not like El Capitan is that old a version of OSX.   

I tried that command (in the directory Documents//Arduino/hardware/espressif/esp32/tools) and got:

```
python esptool.py chip_id
Traceback (most recent call last):
  File "esptool.py", line 25, in <module>
    import serial
ImportError: No module named serial
```

So I commented out 'import serial' is esptool.py and got:

```
python esptool.py chip_id
esptool.py v2.1
Traceback (most recent call last):
  File "esptool.py", line 2524, in <module>
    _main()
  File "esptool.py", line 2517, in _main
    main()
  File "esptool.py", line 2240, in main
    esp = ESPLoader.detect_chip(args.port, initial_baud, args.before)
  File "esptool.py", line 203, in detect_chip
    detect_port = ESPLoader(port, baud)
  File "esptool.py", line 174, in __init__
    if isinstance(port, serial.Serial):
NameError: global name 'serial' is not defined

```
I got exactly the same results on the High Sierra Macbook AIr where things work fine and the El Capitan Mac Pro where the upload hangs.

lbernstone, 2018-01-11:
No, it queries the chip for its id (which I believe is just the base MAC in ESP32).  It's a non-destructive test of your serial connection.  I'm afraid I don't know enough about Mac serial and python to help you out.

JohnGEyles, 2018-01-11:
Ok, thanks for trying.   Seems like if I could just get ahold of the code from when El Cap was the latest OSX, but I can't figure out how to do that.

JohnGEyles, 2018-01-18:
BTW, I've asked about this at stackexchange:

https://arduino.stackexchange.com/questions/48853/adafruit-esp32-device-working-on-osx-10-13-but-not-on-osx-10-11-on-older-machine?noredirect=1#comment97665_48853

... and at the Adafruit forums:

https://forums.adafruit.com/viewtopic.php?f=57&t=128986

JohnGEyles, 2018-02-27:
Ok, new information: It turns out that it works on SOME of the USB ports on the MacPro, and not on others.   Was hanging (as described in OP) on the ones on the front of the machine (tower) but DOES work on the ones on the rear of the Apple monitor (which in turn is connected to the ports on the rear of the machine).   However, it does not work if I plug the monitor into a different USB port (on the back of the machine) and plug the Arduino directly into the USB port that the monitor WAS plugged into when plugging the Arduino into the back of the monitor worked.

This is REALLY strange behavior, IMHO, and hopefully a smoking gun for someone who's smarter about this stuff than me.

FWIW, this is what my USB setup looks like in System Profiler (with HUZZAH32 plugged into rear of Apple monitor, the only confuguration that seems to work):

[USB 2.0 Bus:

  Host Controller Driver:	AppleUSBEHCIPCI
  PCI Device ID:	0x3a3a 
  PCI Revision ID:	0x0000 
  PCI Vendor ID:	0x8086 

Hub:

  Product ID:	0x9120
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	0.01
  Speed:	Up to 480 Mb/sec
  Location ID:	0xfd100000 / 1
  Current Available (mA):	1000
  Current Required (mA):	500
  Extra Operating Current (mA):	0

Hub in Apple Pro Keyboard:

  Product ID:	0x1002
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	1.22
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Mitsumi Electric
  Location ID:	0xfd120000 / 3
  Current Available (mA):	1000
  Current Required (mA):	50
  Extra Operating Current (mA):	0

Apple Optical USB Mouse:

  Product ID:	0x0302
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	16.02
  Speed:	Up to 1.5 Mb/sec
  Manufacturer:	Logitech
  Location ID:	0xfd122000 / 5
  Current Available (mA):	1000
  Current Required (mA):	100
  Extra Operating Current (mA):	0

Apple Pro Keyboard:

  Product ID:	0x0204
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	1.22
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Mitsumi Electric
  Location ID:	0xfd121000 / 4
  Current Available (mA):	1000
  Current Required (mA):	50
  Extra Operating Current (mA):	0

CP2104 USB to UART Bridge Controller:

  Product ID:	0xea60
  Vendor ID:	0x10c4  (Silicon Laboratories, Inc.)
  Version:	1.00
  Serial Number:	01435454
  Speed:	Up to 12 Mb/sec
  Manufacturer:	Silicon Labs
  Location ID:	0xfd110000 / 6
  Current Available (mA):	1000
  Current Required (mA):	100
  Extra Operating Current (mA):	0

Apple Cinema HD Display:

  Product ID:	0x9220
  Vendor ID:	0x05ac  (Apple Inc.)
  Version:	1.33
  Speed:	Up to 1.5 Mb/sec
  Location ID:	0xfd130000 / 2
  Current Available (mA):	1000
  Extra Operating Current (mA):	0

](url)

JohnGEyles, 2018-02-27:
[image]

llemtt, 2018-02-28:
Update your CP210X drivers to latest version 5 -> it should work.

Anyway it usually works also if you connect it through a passive usb hub...

llemtt, 2018-03-01:
Updated CP210X drivers on my El Capitan -> doesn't work anymore!! And I can't easily undo...

It seems that esptool can't actually read input from /dev/cu.SLAB_USBtoUART because opening the device after esptool failure reads all the "buffered" input.

JohnGEyles, 2018-03-01:
I don't think CP210X drivers is my problem, because it works on some of my ports.

Sorry about what happened with you, but thanks for the warning about the update.

Man, this stuff is flakey ...

llemtt, 2018-03-01:
Ok but those port are different from on board ports...

BTW I traced my problem to python serial flushInput() which doesn't flush anything but blocks subsequent read calls :worried:

JohnGEyles, 2018-03-01:
Not sure what you mean by "Ok but those port are different from on board ports...", can you explain more ?   Somehow the drivers could still be my problem ?

closed 2018-04-19 by everslick