espressif/arduino-esp32 #985: Wifi.softAPConfig() sometimes set the wrong IP address

henricazottes, 2018-01-10:
### Hardware:
Board:							Lolin32 Lite
Core Installation/update date:			?11/jul/2017?
IDE name:							Platform.Io
Flash Frequency:					?
Upload Speed:						?

### Description:
I'm using my esp as an AP to serve a little webserver. To do so, I start the ESP in AP mode like this:

```cpp
int startAP() {
  digitalWrite(LED_BUILTIN, LOW);
  Serial.println("Running config server");
  delay(500);
  digitalWrite(LED_BUILTIN, HIGH);
  WiFi.mode(WIFI_OFF);
  delay(1000);
  WiFi.mode(WIFI_AP);
  if(!WiFi.softAPConfig(IPAddress(192, 168, 5, 1), IPAddress(192, 168, 5, 1), IPAddress(255, 255, 255, 0))){
      Serial.println("AP Config Failed");
  }
  if(WiFi.softAP(gw_ssid, gw_pwd)){
    Serial.println("");
    IPAddress myIP = WiFi.softAPIP();
    Serial.println("Network " + String(gw_ssid) + " running");
    Serial.print("AP IP address: ");
    Serial.println(myIP);
    return 0;
  } else {
      Serial.println("Starting AP failed.");
      return 1;
  }
}
```
As you can see I'm giving a static IP to the ESP which is supposed to be `192.168.5.1`. Moreover, the sub network mask is set to /24. However despite this given configuration I sometimes get my ESP adressed as `192.168.4.1` (which is confirmed when I try to access the webserver).

Am I missing a configuration option ? Thanks :)

henricazottes, 2018-01-15:
Anyone experiencing the same ?
(ping @me-no-dev ?)

everslick, 2018-01-20:
you have to wait until event SYSTEM_EVENT_AP_START has fired, before you can set its configuration.

a crude hack would be to put delay(100); after WiFi.mode(WIFI_AP);

kharar, 2018-01-24:
I can confirm that the following is working for me, running Arduino with the latest ESP32 library
```
#include <WiFi.h>

const char *ssid = "ESP32ap";
const char *password = "12345678";

void setup() {
  Serial.begin(115200);
  Serial.println("Configuring access point...");

  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid, password);
  Serial.println("Wait 100 ms for AP_START...");
  delay(100);
  
  Serial.println("Set softAPConfig");
  IPAddress Ip(192, 168, 1, 1);
  IPAddress NMask(255, 255, 255, 0);
  WiFi.softAPConfig(Ip, Ip, NMask);
  
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);
}

void loop() {
}
```

henricazottes, 2018-01-24:
I also tried today and so far it has been working correctly. I'll report it if it behaves weird again. Thanks guys !

closed 2018-01-24 by henricazottes

williamesp2015, 2019-06-15:
I have problem for one week using WIFI_AP_STA specially AP mode. I am using ESP32 and PlatformIo and after connect to AP I can not connect to websocket. I had no problem before.

WiFi.disconnect(true); //Disable STA
WiFi.mode(WIFI_OFF);
delay(1000);
WiFi.mode(WIFI_AP_STA);
 while (WiFi.getMode() != WIFI_AP_STA) {
      delay(50);
      }
Serial.println("Configuring access point...");
WiFi.softAP(apssid, appassword);
Serial.println("Wait 100 ms for AP_START...");
delay(100);
IPAddress myIP = WiFi.softAPIP();
Serial.print("AP IP address: ");
Serial.println(myIP);
Serial.print("Soft-AP IP address = ");
Serial.println(WiFi.softAPIP());

thEpisode, 2020-01-28:
> you have to wait until event **SYSTEM_EVENT_AP_START** has fired, before you can set its configuration. and the AP is started by `WiFi.softAP(gw_ssid, gw_pwd);` so the order is:
> 
> ```
> WiFi.mode(WIFI_AP);
> WiFi.softAP(...);
> wait for SYSTEM_EVENT_AP_START
> WiFi.softAPConfig(...)
> ```
> 
> a crude hack would be to put delay(100); after WiFi.softAP(...);

Awesome! It works!

sanfx, 2020-06-04:
> I can confirm that the following is working for me, running Arduino with the latest ESP32 library
>
> ```
> #include <WiFi.h>
>
> const char *ssid = "ESP32ap";
> const char *password = "12345678";
>
> void [...]

what is `#include <WiFi.h>` I believe everyone is using `ESP8266Wifi.h`

atefbhuiyan, 2024-10-07:
that is for esp8266, he was using esp32.