espressif/arduino-esp32 #988: I2C port expander

fredykoci, 2018-01-10:
Hi, I have problem with I2c io port expander and esp32. i2c scanner find i2c address but can't comunicate with it. here is scatch:

#include<Wire.h>

void setup() {
  // put your setup code here, to run once:
  Wire.begin(21, 22);
  Serial.begin(115200);
  Wire.setClock(400000); 
  Wire.beginTransmission(0x20); 
  Wire.write(0x00); 
  Wire.write(0x00); 
  Wire.endTransmission(false); 
}

void loop() {
  // put your main code here, to run repeatedly:
  Wire.beginTransmission(0x20); 
  Wire.write(0x09); 
  Wire.write(00000011); 
  Wire.endTransmission(false); 
}

and error code is always 0.

stickbreaker, 2018-01-10:
@fredykoci 

* Are you using my I2C fork [stickbreaker/arduino-esp32](https://github.com/stickbreaker/arduino-esp32) or this main branch?
* What is the actual part your are using?
* What is the resistor value of your pullups?
* Why do you have `Wire.endTransmission(false);` 
Try `Wire.endTransmission();` or `Wire.endTransmission(true);` 

Chuck.

fredykoci, 2018-01-10:
this main branch, I have not resistors because I read that esp32 need not have resistors

stickbreaker, 2018-01-10:
@fredykoci The built in pullups on the ESP32 are ~50k..100k ohm. I2C with 3.3V needs 2.4k to 3.3k

With this branch `Wire.endTransmission(false);` will cause timeout problems with the hardware. 

a link to my thread discussing [pullup values ](https://github.com/stickbreaker/arduino-esp32/wiki/Pullup-Values)

I2C transactions must have START, data, STOP.  
```c++
Wire.beginTransaction(addr); // generates the start, and one data byte (the address)
Wire.write(value); // generates additional data bytes
Wire.endTransmission(false); // does NOT generate a STOP, it just pauses the I2C hardware waiting for
// additional commands, usually a Wire.requestFrom(addr,len);
// if the Wire.requestFrom() is not processed within ~12ms the I2C hardware Fails with a timeout
```

Chuck.

fredykoci, 2018-01-13:
Hi,
now i have 2,4K Ohm resistor between 3,3 and sda , 3,3 and scl

code looks like this:
#include "Wire.h"

void setup() {
  // put your setup code here, to run once:
  Wire.begin(21, 22);
  Wire.setClock(400000);
  Serial.begin(115200);
  Wire.beginTransmission(0x20);
  Wire.write(0x00);
  Wire.write(0x00);
  Wire.endTransmission();
}

void loop() {

  Wire.beginTransmission(0x20);
  Wire.write(0x09);
  Wire.write(11111111);
  byte error = Wire.endTransmission();
  Serial.println(error);
  delay(1000);
}

error code is always 0 i2c doesn't work.
whats now?

stickbreaker, 2018-01-13:
@fredykoci I need more info from you:
* What is the part number of your I/O expander?
* What doesn't work?  Error code 0 says Wire successfully executed,

try this sketch:
```c++
#include <Wire.h>

void scan(){
Serial.println("\n Scanning I2C Addresses");
uint8_t cnt=0;
for(uint8_t i=0;i<0x7F;i++){
  Wire.beginTransmission(i);
  uint8_t ec=Wire.endTransmission(true); // if device exists on bus, it will aCK
  if(ec==0){ // Device ACK'd
    if(i<16)Serial.print('0');
    Serial.print(i,HEX);
    cnt++;
  }
  else Serial.print(".."); // no one answered
  Serial.print(' ');
  if ((i&0x0f)==0x0f)Serial.println();
  }
Serial.print("Scan Completed, ");
Serial.print(cnt);
Serial.println(" I2C Devices found.");
}

void setup(){
Serial.begin(115200);
Wire.begin();
scan();
}

void loop(){
}
```

Chuck.

fredykoci, 2018-01-14:
part number is MCP23008

copercini, 2018-07-01:
The I2C core was changed to @stickbreaker code officially after https://github.com/espressif/arduino-esp32/commit/13dcfe5c13bc882f8923d137ed604aa65b6db057 (thanks @stickbreaker), Let's close issues with old code and if have problems with the new code, open new issues =)

closed 2018-07-01 by copercini