espressif/arduino-esp32 #989: ESP32 use of Thingspeak Library

tolo69, 2018-01-10:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			30/Oct/2017
IDE name:						Arduino IDE 1.8.3
Flash Frequency:					80MHz
Upload Speed:						921600?

Description of Problem:

Compiler  Errors:
Arduino: 1.8.3 (Windows 10), Board: "ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, Debug"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\tolo6\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\tolo6\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\projects\ArduinoSketches\libraries -fqbn=esp32:arduino-esp32-master:esp32:FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=debug -ide-version=10803 -build-path C:\Users\tolo6\AppData\Local\Temp\arduino_build_318008 -warnings=none -build-cache C:\Users\tolo6\AppData\Local\Temp\arduino_cache_796384 -prefs=build.warn_data_percentage=75 -verbose C:\projects\ArduinoSketches\ESP32\Thingspeak\Thingspeak\Thingspeak.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\tolo6\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\tolo6\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\projects\ArduinoSketches\libraries -fqbn=esp32:arduino-esp32-master:esp32:FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=debug -ide-version=10803 -build-path C:\Users\tolo6\AppData\Local\Temp\arduino_build_318008 -warnings=none -build-cache C:\Users\tolo6\AppData\Local\Temp\arduino_cache_796384 -prefs=build.warn_data_percentage=75 -verbose C:\projects\ArduinoSketches\ESP32\Thingspeak\Thingspeak\Thingspeak.ino
Using board 'esp32' from platform in folder: C:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master
Using core 'esp32' from platform in folder: C:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/config" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/bluedroid" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/app_trace" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/app_update" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/bootloader_support" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/bt" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/driver" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/esp32" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/esp_adc_cal" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/ethernet" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/fatfs" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/freertos" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/heap" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/jsmn" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/log" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/mdns" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/mbedtls" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/mbedtls_port" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/newlib" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/nvs_flash" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/openssl" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/soc" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/spi_flash" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/sdmmc" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/spiffs" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/tcpip_adapter" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/ulp" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/vfs" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/wear_levelling" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/xtensa-debug-module" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/console" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/newlib" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/coap" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/wpa_supplicant" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/expat" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/json" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/nghttp" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/lwip" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10803 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ARDUINO-ESP32-MASTER  -DESP32 -DCORE_DEBUG_LEVEL=4 "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master\cores\esp32" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master\variants\esp32" "C:\Users\tolo6\AppData\Local\Temp\arduino_build_318008\sketch\Thingspeak.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/config" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/bluedroid" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/app_trace" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/app_update" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/bootloader_support" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/bt" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/driver" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/esp32" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/esp_adc_cal" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/ethernet" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/fatfs" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/freertos" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/heap" "-IC:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master/tools/sdk/include/jsmn" "-IC:\Program Files (x86)\Arduino\hardware\esp32\ardu [... cut ...]
   #error Only Arduino MKR1000, Yun, Uno/Mega/Due with either WiFi101 or Ethernet shield. ESP8266 and ESP32 are also supported.

    ^

C:\projects\ArduinoSketches\libraries\ThingSpeak\src/ThingSpeak.h:87:3: error: #error "Platform not supported"

  #error "Platform not supported"

   ^

Bibliothek ThingSpeak in Version 1.3.2 im Ordner: C:\projects\ArduinoSketches\libraries\ThingSpeak  wird verwendet
Bibliothek WiFi in Version 1.0 im Ordner: C:\Program Files (x86)\Arduino\hardware\esp32\arduino-esp32-master\libraries\WiFi  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board ESP32 Dev Module.

tolo69, 2018-01-10:
Description of Problem:
I try to use the Thingspeak library "Thingspeak.h" für ESP32. But the compiler throws an error, refer below.
Can this library be adapted to ESP32? 
Or is there any other way to read values from Thingspeak client ?

lbernstone, 2018-01-10:
Add this to the top of your sketch:
#define ARDUINO_ARCH_ESP32 1

stickbreaker, 2018-01-10:
@lbernstone your 
`#define ARDUINO_ARCH_ESP32`

Should not be necessary.  Unless the selected board has an incorrect definition in boards.txt?

which it has "-DARDUINO_ARCH_ARDUINO-ESP32-MASTER"

Somewhere in the build recipes `build.arch=ARDUINO-ESP32-MASTER` exist.
But it should be `build.arch=ESP32`

Chuck.

tolo69, 2018-01-10:
Thanks for the hint. I tried the #define on top of my sketch. But this doesn't change the compiler error :|
It tells that thingspeak.h is incompatible with arduino-esp32-master. What would you suggest ?

lbernstone, 2018-01-10:
His code very clearly states that the ESP32 is supported, and it compiles for me. Have you tried the example sketches included with the library?

tolo69, 2018-01-10:
The example sketches can be compiled w/o problem. Only when I #include <ThingSpeak.h> the make stops with incompatibility error.

closed 2019-08-15 by stale[bot]