espressif/arduino-esp32 #990: Improvmenet on dhcp config.

DasVoelkel, 2018-01-11:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			9/jul/2017
IDE name:							Arduino IDE, IDF_PATHS, 
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
I don'T have an issue, i just think it would be neat if WiFi had a function (which uses config) to get into dhcp mode.
Personally i had to search for quite a while what the error meant i was getting and fount out that
```
WiFi.config({ 0,0,0,0 }, { 0,0,0,0 }, { 0,0,0,0 }); 
```
get's the whole thing back into dhcp mode if it was preconfigured before.

My issue was that if the user entered a wrong ip , it would have to go for normal dhcp instead.

Something like:
```
bool WiFi.config_dhcp();

WiFi.config_dhcp(){
WiFi.config({ 0,0,0,0 }, { 0,0,0,0 }, { 0,0,0,0 }); 
return true;

//i am not sure if or what could go wrong with config.
return false

}
```
//not too sure on how exactly to write this ^^ sorry (and i don'T know hot to tag it as 'enhancement' if it is any)

Loufylouf, 2018-01-20:
Just to be clear, you would like a function that deletes the config entered previously (ip, gateway, mask and dns addresses) to fallback on DHCP mode when trying to reconnect ?

DasVoelkel, 2018-01-21:
My current issue is, if you set a custom config , and then use WiFi.begin() and later disconnect and reconnect to another network, it will still keep the config.
In order to get rid of the config entirely (in its current state) you have to WiFi.disconnect(true). 
First of, i didn't find this anywhere documented and stumbled across it when i read through the libraries source code. 
Secondly this only works because it completly frees any space associated with the WiFi library, effectivly deleting the previous config and everything else. If you use WiFi.begin() after that, the library gets reassigned and the config is empty.

It seems a little storage and process intensive to free the entire API in order to just reset the config, and was wondering if it was possible to reduce the load by makeing a process to delete the previous config settings.

(i didn't know about wifi.disconnect(true) when i wrote the previous poste, so essentially my question changed from, "can there be a reset to dhcp" to "can i delete a previous config" because if there is no config, wifi.begin uses it'S dhcp mode anyways)

if i understood your question wrong or am unclear, sorry and i'll try to explain it better next time ^^

Loufylouf, 2018-01-22:
Thanks, that's a lot clearer. I took a look at the code for the different functions involved and I agree there is no way to clear a config, except `WiFi.config({ 0,0,0,0 }, { 0,0,0,0 }, { 0,0,0,0 });` which is a bit tricky. I see a few options : 

- Add an optional parameter to the begin function to make a fresh start
- Add an optional parameter to the disconnect function to also erase the config
- Add a function to erase the config

The first one feels a bit odd to me. The second feels quite normal : if you disconnect from a network, your configuration probably means nothing on the next network you'll connect to, so by default, I would erase the config but have a parameter to keep it, in odd cases.

DasVoelkel, 2018-01-22:
Second option seems the best ^^ and yes that is what i meant.
to make a fresh start with disconnect(true) is just my workaround , and i aggree it's dirty.

best would be to erase the config. 
should i close the issue ?

dpharris, 2018-01-22:
Disconnect with optional parameter to *not* erase the config.  ?

DasVoelkel, 2018-01-22:
I mean ... it should be fine, it's just a boolean.
The current way it is implemented it deleted everything, including the config.

It would be nice to have aa choice to just delete the config without the rest (my initial problem) 
but while one is at it i cna'T imagine it being too difficult to make it optional to disconnect without deleting the config (currently it is implemented this way, the normal wifi.disconnect() function does not delete the config)

But like @Loufylouf said, it makes very little sense to keep a config when joining a new network, i found a way to test if you have a valid ip address for the new network, but really there is no integration for this yet.
so IF you have an ip address in the configs, which works on one network, and you switch, the dhcp ignores your address completly, and you won't be able to do anything in the network, but WiFi.status() still thinks everything is in order,since you are technically "connected" the ip address just ain'T valid.

Loufylouf, 2018-01-23:
I have created a branch on my fork of the code (https://github.com/Loufylouf/arduino-esp32/tree/Fix-issue-990), if you could test it with your use case @DasVoelkel , that would be great.

DasVoelkel, 2018-01-23:
looks great ! i will test it tomorrow and reply !

DasVoelkel, 2018-01-24:
Error in line 310 WiFiSTA.cpp ;
You forgot to define err. `esp_err_t err = ESP_OK;`

Also , when i tested it i get a lot of core crashing and overflows.
I got core crashes galore and couldn't even trace them back xD, when i switched back to the old library, it worked again. my feeling is that some pointer is wrong somewhere.

Loufylouf, 2018-01-24:
Did you fetch the last commit ? Because the error you got had been fixed by a following commit.

DasVoelkel, 2018-01-24:
it says i am up to date with "origin/master" ,but i really don't know what changed because i missed the last week. 
My code worked before i messed around with the libraries (trying yours and then back to the original libraries master) and now i get a stackoverflow error and core panic just before starting any kind of socket server ( i think). The Backtrace , even decoded , is unhelpful as all hell.

closed 2019-03-04 by DasVoelkel