espressif/arduino-esp32 #991: Serial(2) only works first loop after reboot, Serial(1) works fine

TLS1000, 2018-01-11:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?DOIT V1 ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
I'm using modbusmaster library to communicate with some Modbus slave. I initially used hardwareserial(2) because I thought serial1 was for communicating to Arduino IDE. With serial2 I only got response from the Modbus slave the first time after reboot. When I use Serial1 UART it works fine.

this has been reported earlier: [https://github.com/espressif/arduino-esp32/issues/650]

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

TLS1000, 2018-01-11:
I tried using the default UART2 pins 16 and 17, no improvement.

bertvaneyken, 2018-01-14:
Wich modbusmaster library are you using?
You should include the important parts of your code.

TLS1000, 2018-01-15:
Hi,

I'm using the [https://github.com/4-20ma/ModbusMaster)](https://github.com/4-20ma/ModbusMaster)
these are parts of my code:
global:
```
`#include <HardwareSerial.h>
#include <ModbusMaster.h>
HardwareSerial Serial2(1);
#define Serial2TxControl 4   //RS485 Direction control 
#define RS485Transmit    HIGH
#define RS485Receive     LOW
ModbusMaster node;
```setup:
```
`Serial.begin(115200);
  pinMode(Serial2TxControl, OUTPUT);
  //delay(10);
  Serial2.setDebugOutput(HIGH);
  Serial2.begin(19200,SERIAL_8N1,16,17);
  node.begin(4,Serial2);`
node.preTransmission(preTransmission);
  node.postTransmission(postTransmission);

loop:
```
`node.readHoldingRegisters(20100,64)
node.getResponseBuffer(i)
```

Now when I change
HardwareSerial Serial2(1); to HardwareSerial Serial2(2);   The code runs once after reboot, it sends out data, but I think it's corrupt because Modbus slave doesn't respond. The hardwareserial (1) runs fine for several days now. Important: you need to add a delay in the  'posttransmission' routine so the driver stays on a few  ms longer, depending on your baudrate speed.

And although the common MAX485 Arduino board is specified for TTL use, it runs fine at CMOS levels (and powered by 3.3V).

baggior, 2018-02-04:
same issue for me
only the first message is sent correctly throught UART2, followings packet are garbled with other data
I dont't what this data is 

usign UART1 works forever

shimarin, 2018-02-06:
Most likely it's related to hardware issue of UART2.
https://github.com/espressif/esp-idf/issues/1202#issuecomment-342691402

RS485 communication using ESP32's UART2 causes this problem as it needs to flush send buffer to finish RTS state.

In esp-idf, a workaround has been applied but not in Arduino-ESP32.

baggior, 2018-02-06:
@shimarin  maybe you are right about the hardware bug.. could you extend and apply the workaround on the Arduino HardwareSerial library too?

shimarin, 2018-02-06:
@baggior Well, I wish I was someone who could do it. Rewriting my code to use esp-idf's uart library was the best I could do.

baggior, 2018-02-06:
who could correct the **HardwareSerial** library and include the uart fix, then?

TLS1000, 2018-02-06:
yes, having only UART1, (most of us use UART0 for IDE communication) feels like a limitation on ESP32. Hopefully someone with the right skillset can fix this.

everslick, 2018-04-19:
I think there have been some changes lately, does this problem still persist?

dtboy1995, 2018-05-31:
this problem still persist

TLS1000, 2018-05-31:
JacoFourie got all three UARTS working, you need to add a serial.flush() in the pretransmission routine. I haven't tested it.

https://github.com/4-20ma/ModbusMaster/issues/96#issuecomment-368253172

dtboy1995, 2018-05-31:
@TLS1000 thank you for your reply.

below works well.
```c
HardwareSerial uart(2);
...
uint8_t data[11] = {0x01, 0x10, 0x01, 0x02, 0x00, 0x01, 0x02, 0x00, 0x09, 0x77, 0x74};
uart.write(data, 11);
```
but let me add a line of code., it output garbled data.
```c
HardwareSerial uart(2);
...
uint8_t data[11] = {0x01, 0x10, 0x01, 0x02, 0x00, 0x01, 0x02, 0x00, 0x09, 0x77, 0x74};
uart.write(data, 11);
uart.flush(); // next loop output garbled data.
```

Kirk-1, 2018-06-11:
Another clue for finding this issue. I am using Arduino ide with a Wrover module, the uart2.available() function will only respond every 10 seconds if I am initialising uart2 non-inverted. When I enable invert the uart2.available() responds normally. 

I tried commenting out these lines as @JacoFourie suggested in #96 but didn't help.

//uart->dev->conf0.txfifo_rst = 1;
    //uart->dev->conf0.txfifo_rst = 0;

    //uart->dev->conf0.rxfifo_rst = 1;
    //uart->dev->conf0.rxfifo_rst = 0;

 Anyone out there that can help!

TLS1000, 2018-11-30:
With my new project I need all three UARTS, problem still persists...

I can use Serial(1) with no problem, when I change to Serial(2) the code only runs once after startup...

I tried the ESP32Serial library that uses the IDF hardwareserial, but it gives the same problem.

please help

TLS1000, 2018-11-30:
I managed to get Serial(2) working with the ESP32Serial.h library
https://github.com/9a4gl/ESP32Serial

closed 2018-11-30 by TLS1000