espressif/arduino-esp32 #993: Unable to find my board on network port

LeoLaiFookSiang, 2018-01-13:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Unable to find my board on network port. The network port session don't even show out.

### Code:
The code is using the sample code BasicOTA

bill-orange, 2018-01-13:
Is the PC that is looking for the board hooked up by Ethernet or WiFi?  If it is connected to the network via Ethernet try switching to WiFi.

  I can only see my boards if I switch the PC to WiFi by disabling Ethernet and plugging in a Wifi USB device.  Apparently, my lousy Time Warner router blocks UDP broadcasts between WiFi and  Ethernet for security reasons.  There is no way to turn off this "feature".

LeoLaiFookSiang, 2018-01-14:
Yes, the PC using WiFi connection. But I still unable to see network port in arduino ide

bill-orange, 2018-01-14:
Are you running bonjour? It is part of iTunes and manages UDP on PCs. It seems weird that you have to install this Apple product. 

http://smallbusiness.chron.com/really-need-bonjour-computer-73335.html

LeoLaiFookSiang, 2018-01-14:
Unfortunately, I am develop in Window OS

bill-orange, 2018-01-14:
It runs on Windows 7, 10, etc. as do many Apple products. It’s free.

LeoLaiFookSiang, 2018-01-15:
Solved but not sure why. I use another place WiFi connection and it works.

closed 2018-01-15 by LeoLaiFookSiang