espressif/arduino-esp32 #994: Set RTC without connecting to SNTP server

felixzhang97, 2018-01-13:
Hi. I'm currently developing an automation program and need to use real time clock. As i read, esp32 has built-in RTC and there are way to config time using NTP server. My problem is i'm not using wifi and my program will run totally in offline mode. Is there any way i can set the time without connecting to any server?(i planned to set time via bluetooth)

Thankyou.

marian-craciunescu, 2018-02-13:
any ideas?

lrmoreno007, 2018-02-13:
The problem:
RTC in esp32 is made for internal timing process, and it have a bad accuracy. https://www.esp32.com/viewtopic.php?t=3715

As a solution, you can determine what the error is on each board and establish a correction mechanism.

Any ideas?:
You can set the time with in the same way as in the example ESP8266/NTP-TZ-DST (see constant RTC_TEST) (it's epoch time - seconds since  January 1, 1970), for this you can use two or three button (SET, + or/and -) and a display, as in many clocks and home appliances.

You can get time and date too when upload a new sketch with the Arduino variables: " __DATE__ " and " __TIME__ ", but this is a fixed time when you upload the sketch and if you loose the power loose the time.

Regards.

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

fbiego, 2021-01-17:
you can set the esp32 time as follows,
```
  struct tm t = {0};  // Initalize to all 0's
  t.tm_year = year-1900;  // This is year-1900, so 121 = 2021
  t.tm_mon = month-1;
  t.tm_mday = day;
  t.tm_hour = hour;
  t.tm_min = minute;
  t.tm_sec = second;
  time_t timeSinceEpoch = mktime(&t); //create epoch time

  struct timeval tv;
  tv.tv_sec = timeSinceEpoch;  // ie 1609459200 -> 1st Jan 2021 in epoch time
  tv.tv_usec = 0;
  settimeofday(&tv, NULL);
```

and read like this

```
  struct tm timeinfo;
  getLocalTime(&timeinfo);
  int hour = timeinfo.tm_hour;
  int minute = timeinfo.tm_min;
  int second = timeinfo.tm_sec;
  int day_of_month = timeinfo.tm_mday;
  int month = timeinfo.tm_mon;
  int year = timeinfo.tm_year;
  int day_of_week = timeinfo.tm_wday;
```

lbernstone, 2021-01-17:
https://github.com/lbernstone/ESP32_settimeofday