espressif/arduino-esp32 #997: How High is the SDA/SCL High of the ESP32?

Skel0t, 2018-01-14:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					?40Mhz?
Upload Speed:						115200

### Description:
Hey, I want to use a i2c sensor that can't handle 5V SDA/SCL so I'd like to know how high the high of the esp32 is. Googling it didn't help, I looked for more than an hour now so I would appreciate every answer.

stickbreaker, 2018-01-14:
@Skel0t The ESP32 is a 3.3v device all of it's I/O's are limited to 0 .. 3.3V

The I2C pullups must be connect to 3.3v.

Chuck.

Skel0t, 2018-01-14:
So I don't need to use any resistors? From what I've read researching until now, I would need a 4.7k resistor between SDA & Vcc and SCL & Vcc for 5V and 2.4k instead of 4.7k for a 3.3V device.

stickbreaker, 2018-01-14:
@Skel0t yes, you need to add Pullups, with 3.3v use between 2.4k and 3.3k to 3.3V.

Here is a captured thread how I justified these values [Pullup Resistor Values](https://github.com/stickbreaker/arduino-esp32/wiki/Pullup-Values).

Chuck.

Skel0t, 2018-01-14:
Thank you gor your help! 
Problem solved then

closed 2018-01-14 by Skel0t