espressif/arduino-esp32 #998: How to send an audio file to a Bluetooth Speaker

williamesp2015, 2018-01-14:
### Hardware:
Board:							ESP32 Dev Module?
IDE name:							Platform.io Arduino-ESP32
I need to use ESP32 to send audio file stored on a SD card to Bluetooth Speaker but I don't know how to do it using Arduino-ESP32. Using smartphone is easy just connect to Speaker and then play sound. Is it easy just like that?

joeybab3, 2018-01-16:
It's a lot more difficult than that, and as far as i know this can't be done using the arduino IDE, only the actual Espressif sdk.

joeybab3, 2018-01-16:
I've attempted this myself or at least getting audio to work on the esp32 and was able to get this example playing audio: https://github.com/MrBuddyCasino/ESP32_MP3_Decoder

closed 2018-04-19 by everslick