espressif/arduino-esp32 #1000: Get new IP address each time the ESP32 gets started/reboot

T94T, 2018-01-15:
### Hardware:
Board:							Various ESP32 modules
Core Installation/update date:			2.4.0
IDE name:							Arduino IDE 1.8.3
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Everytime I startup/reboot my ESP32 I get a new IP address from the router (Fritz!Box Fon WLAN 7390).

### Sketch:
For simplicity I use the WiFiClient example.

T94T, 2018-01-15:
By the way: My ESP8266 devices get the same IP addresses each time they connect.

stickbreaker, 2018-01-15:
@T94T the DHCP server reuses IP based on MAC address, if the ESP32 is sending a different MAC it will receive a different IP.  Modify your test code to display the MAC address.  Verify that it is sending the same MAC when it asks for an IP.

Chuck.

T94T, 2018-01-15:
This was my first thought also. But the ESP32 sends the same MAC address each time it connects.

lrmoreno007, 2018-01-15:
Take a look at this:

https://en.avm.de/service/fritzbox/fritzbox-7390/knowledge-base/publication/show/1469_The-wireless-guest-access-cannot-be-used-to-access-the-internet/

Let's see what he puts in the field: "Valid for"

stickbreaker, 2018-01-15:
@lrmoreno007 , he, @T94T is getting an IP.  Your weblink is solving an IP pool exhaustion because of extending lease time problem.  His problem is that every time his ESP32 reconnects it is given a different IP. I believe that this wandering IP is a problem because he is using the ESP32 as some type of a server that other devices access.  Whenever the ESP32 reconnects all of his other devices have to 'find it' at a new address.

Chuck.

lrmoreno007, 2018-01-15:
Thanks for clarifying the problem, but it was already clear. My always assigned the same IP with the cited sketch, so the problem seems to be in the router configuration and I was trying to discard things.

T94T, 2018-01-19:
I made a git pull on the arduino-esp32 today. Since then, the ESP32 gets the same IP address on each reboot.

everslick, 2018-01-20:
@T94T  then you should consider closing this issue.

closed 2018-01-20 by T94T