espressif/arduino-esp32 #1001: SmartConfig stopped working.

CelliesProjects, 2018-01-15:
Board:  MH-ET LIVE ESP32MiniKit
Core Installation/update date:	v3.0-dev-1325-ge5b2c1cf
IDE name: Arduino IDE 1.8.4
Flash Frequency:	80Mhz
Upload Speed:	115200

SmartConfig does not work anymore.
To reproduce the problem, just run the example `WiFiSmartConfig.ino`.

Tested on 4 different WiFi routers, none of them was able to connect via SC.

Anyone else notice this?

### Sketch:
```cpp
#include "WiFi.h"

void setup() {
  Serial.begin(115200);

  //Init WiFi as Station, start SmartConfig
  WiFi.mode(WIFI_AP_STA);
  WiFi.beginSmartConfig();

  //Wait for SmartConfig packet from mobile
  Serial.println("Waiting for SmartConfig.");
  while (!WiFi.smartConfigDone()) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("SmartConfig received.");

  //Wait for WiFi to connect to AP
  Serial.println("Waiting for WiFi");
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("WiFi Connected.");

  Serial.print("IP Address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
```
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 13 - AP_STOP
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 14 - AP_STACONNECTED
[D][WiFiGeneric.cpp:265] _eventCallback(): Event: 14 - AP_STACONNECTED
Waiting for SmartConfig.
.........[D][WiFiSTA.cpp:624] _smartConfigCallback(): Status: FIND_CHANNEL
............
```

CelliesProjects, 2018-01-17:
@me-no-dev Is this problem known by Espressif?

jack0c, 2018-02-02:
@CelliesProjects  
Can you please use : `WiFi.mode(WIFI_STA);`?

CelliesProjects, 2018-02-02:
@jack0c 
As in
````cpp
...
  //Init WiFi as Station, start SmartConfig
  WiFi.mode(WIFI_STA);
  WiFi.beginSmartConfig();
...
````
Just tried that, but am currently on `ESP32 SDK: v3.0-dev-1325-ge5b2c1cf`
where it makes no difference.

Later commits leave me with tons of errors on all kind of libs.

jack0c, 2018-02-04:
What do you mean？STA mode worked or not?

CelliesProjects, 2018-02-04:
@jack0c 
Works. Thanks for the tip.
So the example sketch is in error?

jack0c, 2018-02-05:
Smartconfig need to change channel, but AP need to fix channel, they are a little contradictory. There is a way support both: stop smartconfig when some station connected to AP, stop AP when smartconfig first stage success.

CelliesProjects, 2018-02-12:
Just a ping to the other thread:
https://github.com/espressif/arduino-esp32/issues/674

CelliesProjects, 2018-04-15:
Closed as it seems 69f72eca849a44688e2da38028eb9aa707e91f08 solves this issue.

closed 2018-04-15 by CelliesProjects