espressif/arduino-esp32 #1004: Reboot when program reaches semaphore

Sonnenschwein, 2018-01-16:
### Hardware:
Board:							Adafruit HUZZAH32 Feather
Core Installation/update date:			?11/jul/2017?
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Whenever I try to use Semaphores in my code I get the following error:

 `/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40087889 on core 1

Backtrace: 0x4008a528:0x3ffd6320 0x4008a627:0x3ffd6340 0x40087889:0x3ffd6360 0x400d371d:0x3ffd63a0

Rebooting...
`
I tried it in several different cases but I never got semaphores to work on my code. The RepeatTimer example works fine though. I read that this has to do with serial connection, but for me everything works fine except when I insert a semaphore.

### Sketch:
```cpp

/**
 * \file sketch_nov23a.ino
 */
#include <Adafruit_GFX.h>    // Core graphics library
#include <SPI.h>       // this is needed for display
#include <Adafruit_ILI9341.h>
#include <Wire.h>      // this is needed for FT6206
#include <Adafruit_FT6206.h>

#include "BLEDevice.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID("b5fea5fb-2467-49d4-8721-26a7b124c9dc");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID("8182da9a-9ed6-4ca3-849c-e0adf73b6a7c");
static BLEUUID    batLowUUID("5795259b-9093-4b74-8337-fbfe4964949e");

// The FT6206 uses hardware I2C (SCL/SDA)
Adafruit_FT6206 ctp = Adafruit_FT6206();

// The display also uses hardware SPI, plus #9 & #10
#define TFT_CS 21
#define TFT_DC 14
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);

#define BITSFORBITMAPS 24
#define BATTERYLOW "Battery Low"

static BLEAddress *pServerAddress;
static boolean doConnect = false;
static boolean connected = false;
static boolean firstConnected = false;
static boolean batChanged = false;
static boolean batLow = false;
static boolean myAlarm = false;
static uint8_t batValue = 99;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLERemoteCharacteristic* pBLRemoteCharacteristic;

byte screen, rotate;
const byte vibratePin = 4;
const int strength = 255;

typedef struct box box;

TaskHandle_t Task1;
TaskHandle_t Task2;

[... 567 lines cut ...]
[D][BLERemoteCharacteristic.cpp:173] gattClientEventHandler(): Invoking callback for notification on characteristic Characteristic: uuid: 8182da9a-9ed6-4ca3-849c-e0adf73b6a7c, handle: 42 0x2a, props: broadcast: 0, read: 1, write_nr: 0, write: 1, notify: 1, indicate: 0, auth: 0
Notify callback for characteristic 8182da9a-9ed6-4ca3-849c-e0adf73b6a7c of data length 2
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_NOTIFY_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_NOTIFY_EVT
[D][BLEUtils.cpp:1451] dumpGattClientEvent(): [conn_id: 0, remote_bda: 30:ae:a4:26:37:82, handle: 42 0x2a, value_len: 2, is_notify: 1]
[D][BLERemoteCharacteristic.cpp:173] gattClientEventHandler(): Invoking callback for notification on characteristic Characteristic: uuid: 8182da9a-9ed6-4ca3-849c-e0adf73b6a7c, handle: 42 0x2a, props: broadcast: 0, read: 1, write_nr: 0, write: 1, notify: 1, indicate: 0, auth: 0
Notify callback for characteristic 8182da9a-9ed6-4ca3-849c-e0adf73b6a7c of data length 2
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_NOTIFY_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_NOTIFY_EVT
[D][BLEUtils.cpp:1451] dumpGattClientEvent(): [conn_id: 0, remote_bda: 30:ae:a4:26:37:82, handle: 47 0x2f, value_len: 1, is_notify: 1]
[D][BLERemoteCharacteristic.cpp:173] gattClientEventHandler(): Invoking callback for notification on characteristic Characteristic: uuid: 5795259b-9093-4b74-8337-fbfe4964949e, handle: 47 0x2f, props: broadcast: 0, read: 1, write_nr: 0, write: 1, notify: 1, indicate: 0, auth: 0
Notify callback for characteristic 5795259b-9093-4b74-8337-fbfe4964949e of data length 1
[D][BLERemoteCharacteristic.cpp:433] readValue(): >> readValue(): uuid: 5795259b-9093-4b74-8337-fbfe4964949e, handle: 47 0x2f
[D][BLEDevice.cpp:105] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 3] ... ESP_GATTC_READ_CHAR_EVT
[D][BLEUtils.cpp:1344] dumpGattClientEvent(): GATT Event: ESP_GATTC_READ_CHAR_EVT
[D][BLEUtils.cpp:1493] dumpGattClientEvent(): [status: ESP_GATT_OK, conn_id: 0, handle: 47 0x2f, value_len: 1]
[D][BLERemoteCharacteristic.cpp:455] readValue(): << readValue(): length: 1
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40087885 on core 1

Backtrace: 0x4008a524:0x3ffd6290 0x4008a623:0x3ffd62b0 0x40087885:0x3ffd62d0 0x400d3578:0x3ffd6310

Rebooting...

```

Sonnenschwein, 2018-01-16:
Additionally, I am not ficeto

stickbreaker, 2018-01-16:
@Sonnenschwein You need to create the semaphore before you use it.

And since you are using it from an ISR it needs to be a binary semaphore;
[xSemaphoreCreateBinary](https://www.freertos.org/xSemaphoreCreateBinary.html)
```timerSemaphore = xSemaphoreCreateBinary();```

Chuck.

Sonnenschwein, 2018-01-16:
@stickbreaker Sounds right. It still doesn't work though...

```
/**
 * \file sketch_nov23a.ino
 */
#include <Adafruit_GFX.h>    // Core graphics library
#include <SPI.h>       // this is needed for display
#include <Adafruit_ILI9341.h>
#include <Wire.h>      // this is needed for FT6206
#include <Adafruit_FT6206.h>

// The FT6206 uses hardware I2C (SCL/SDA)
Adafruit_FT6206 ctp = Adafruit_FT6206();

// The display also uses hardware SPI, plus #9 & #10
#define TFT_CS 21
#define TFT_DC 14
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);

static boolean myAlarm = false;

byte screen, rotate;
const byte vibratePin = 4;
const int strength = 255;

typedef struct box box;

TaskHandle_t Task1;
TaskHandle_t Task2;

hw_timer_t *timer = NULL;
volatile SemaphoreHandle_t timerSemaphore;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

volatile uint32_t isrCounter = 0;
volatile uint32_t lastIsrAt = 0;

void IRAM_ATTR onTimer(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  isrCounter++;
  lastIsrAt = millis();
  portEXIT_CRITICAL_ISR(&timerMux);
  // Give a semaphore that we can check in the loop
  xSemaphoreGiveFromISR(timerSemaphore, NULL);
  Serial.println("onTimer");
  // It is safe to use digitalRead/Write here if you want to toggle an output
}

void DisplayTask(void *pvParameters){
  tft.begin();
  while (! ctp.begin(40)) {  // pass in 'sensitivity' coefficient
    Serial.println("Couldn't start FT6206 touchscreen controller");
  }  
  
  Serial.println("Capacitive touchscreen started");
  showMainScreen();
  timerSemaphore = xSemaphoreCreateBinary(); //Created semaphore 
  while(1){

    // Create alarm
    if (myAlarm){
      if(timer){
        if (xSemaphoreTake(timerSemaphore, 0) == pdTRUE){
          uint32_t isrCount = 0, isrTime = 0;
          // Read the interrupt count and time
          portENTER_CRITICAL(&timerMux);
          isrCount = isrCounter;
          isrTime = lastIsrAt;
          portEXIT_CRITICAL(&timerMux);
          if (isrCount%2==0){
            sigmaDeltaWrite(0, 0);
          }
          else{
            sigmaDeltaWrite(0, strength);
          }
        }
      }
      else{
          // Use 1st timer of 4 (counted from zero).
          // Set 80 divider for prescaler (see ESP32 Technical Reference Manual for more
          // info).
          timer = timerBegin(0, 80, true);
        
          // Attach onTimer function to our timer.
          timerAttachInterrupt(timer, &onTimer, true);
        
          // Set alarm to call onTimer function every second (value in microseconds).
          // Repeat the alarm (third parameter)
          timerAlarmWrite(timer, 1000000, true);
        
          // Start an alarm
          timerAlarmEnable(timer);
      }
    }
    else{
      if(timer){
        timerEnd(timer);
        timer = NULL;
      }
    }
    if (! ctp.touched()) {
      delay(1);
      continue;
    }

    
    //analogWrite(vibratePin, strength);
    // Retrieve a point  
    TS_Point p = ctp.getPoint();
    p.x = map(p.x, 0, 240, 240, 0);
    p.y = map(p.y, 0, 320, 320, 0);

    showMainScreen();
    delay(1);
  }
}

/**
 * \brief Setup starts the serial communication and shows first main screen.
 */
void setup() {
 
  Serial.begin(115200);
  
  xTaskCreatePinnedToCore(
                    DisplayTask,   /* Function to implement the task */
                    "DisplayTask", /* Name of the task */
                    10000,      /* Stack size in words */
                    NULL,       /* Task input parameter */
                    2,          /* Priority of the task */
                    &Task1,       /* Task handle. */
                    1);  /* Core where the task should run */
    myAlarm = true;
    
}

void loop() {
  
  
}

/*****************************************************************
 * Display screens
 */
void showMainScreen(){
  if(myAlarm){
    tft.fillScreen(0);
  }
  else{
    tft.fillScreen(255);
  }
}
```
```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11392
entry 0x40078a9c
Capacitive touchscreen started
onTimer
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40084701 on core 1

Backtrace: 0x40087360:0x3ffcd050 0x4008745f:0x3ffcd070 0x40084701:0x3ffcd090 0x400d2850:0x3ffcd0d0 0x400d0a70:0x3ffcd0f0

Rebooting...
```
The backtrace looks a little different and this time it prints "onTimer" before crashing. 
By commenting out I tracked down the following peace of code as the one being problematic. 

```
          portENTER_CRITICAL(&timerMux);
          isrCount = isrCounter;
          isrTime = lastIsrAt;
          portEXIT_CRITICAL(&timerMux);
``` 
it works. When putting a "Serial.println(""); just before or after these lines the log looks like this:
``` 
Couldn't start FT6206 touchscreen controller
[E][esp32-hal-i2c.c:490] i2cInitFix(): Busy at initialization!
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 38
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 38
[W][esp32-hal-i2c.c:280] i2cRead(): Busy Timeout! Addr: 38
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 38
Couldn't start FT6206 touchscreen controller
[E][esp32-hal-i2c.c:490] i2cInitFix(): Busy at initialization!
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 38
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 38
[W][esp32-hal-i2c.c:280] i2cRead(): Busy Timeout! Addr: 38
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 38
``` 
I guess the MUX is somehow blocking the pins and preventing serial communication or the "sigmaDeltaWrite"?

stickbreaker, 2018-01-16:
@Sonnenschwein 
It looks like you are having I2C issues.  I rewrote the I2C Hardware abstraction Layer to make it more robust.
Try swapping these files from my fork for arduino-esp32 
cores\esp32\esp32-hal-log.h
cores\esp32\esp32-hal-i2c.c
cores\esp32\esp32-hal-i2c.h
libraries\Wire\src\Wire.cpp
libraries\Wire\src\Wire.h

[stickbreaker:arduino-esp32](https://github.com/stickbreaker/arduino-esp32)

Chuck.
tools\sdk\include\soc\soc\i2c_reg.h
tools\sdk\include\soc\soc\i2c_struct.h
libraries\Wire\src\Wire.cpp
libraries\Wire\src\Wire.h
README.md

stickbreaker:arduino-esp32

Chuck.

Sonnenschwein, 2018-01-16:
Thanks for your help. I also found a way to make it work: use ledcWrite insted of sigmaDeltaWrite. I will change the files anyway, thank you!

tferrin, 2018-01-19:
As @stickbreaker points out, you need to initialize your semaphore before you can use it. That's what's causing the assert failed errors.
SInce you have solved your problem you should close this thread.

closed 2018-01-20 by Sonnenschwein