espressif/arduino-esp32 #1006: Error while copiling to DOIT ESP32 DEVKIT V1 board

zsiroslaci, 2018-01-16:
## Hardware:
Board:							DOIT ESP32 DevKit V1 
Core Installation/update date:			16/01l/2018
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						?115200?

### Description:
Arduino IDE throw error that missing ".\\hardware\\espressif\\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp-elf-g++" file compiling to DOIT ESP32 DEVKIT V1 board.
But cloning the git hub repository really no such subdirectory and file in tools directory.
Code was not upload , just compile.   
### Sketch: "SimpleWiFiServer from"Examples"
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

JamesGKent, 2018-01-16:
if you follow the install from git instructions, after you clone the repo you have to get the toolchain. for windows the instructions are here:
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md
there are instructions for other platforms too, links here:
https://github.com/espressif/arduino-esp32#installation-instructions

zsiroslaci, 2018-01-17:
Thanks your answer. I realized I made mistake. I failed to run get.exe

closed 2018-01-17 by zsiroslaci