espressif/arduino-esp32 #1008: esp32 analogWrite HIGH is not constant curent, but pulsing, makes led blink

Edvinas1122, 2018-01-18:
### Hardware:
Board:							ESP-WROOM-32 
IDE name:							Arduino IDE 1.8.5
Upload Speed:						115200

### Description:
ESP32 analogWrite HIGH makes led blink by short about half second interval. Only indicator led on board blinks as programed.

void setup() {
pinMode(2, OUTPUT);

}

void loop() {
  digitalWrite(2, HIGH);
  delay(5000);
}

everslick, 2018-04-19:
This issue is closed, because it looks as if it is not a bug or problem with the ESP32 Arduino core or its support libraries. For general API usage questions or help on specific coding challenges, please visit the [arduino-esp32 Gitter channel](https://gitter.im/espressif/arduino-esp32). If you feel this issue was closed in error, reopen it and comment, why you think this is a bug in the Arduino-Core.

closed 2018-04-19 by everslick

gknops, 2019-01-13:
Were you ever able to figure out the underlying cause? I am having the same issue.