espressif/arduino-esp32 #1014: Firebeetle-esp32.json has bad address

kovachjk, 2018-01-20:
----------------------------- Remove above -----------------------------

### Hardware:
Board:							various
Core Installation/update date:			12/15/2018?
IDE name:							Arduino 1.8.5 Window Store 1.8.10.0
Flash Frequency:					40Mhz
Upload Speed:						115200?

### Description:
I am trying to set up Arduino for the first time for use with my ESP32 boards.  I had been implementing small projects using the Eclipse IDE with ESP-IDF.

When following various different tutorials I was never able to get a list of ESP32 boards to show up under the tools menu.  
I finally found a tutorial that spoke specifically about how to install boards using json under file/preferences.
It said to enter: https://git.oschina.net/dfrobot/FireBeetle-ESP32/raw/master/package_esp32_index.json  in the input window called: Additional Boards Manager URL's.  Which I did.
Next I went to tools/boards/Boards Manager:  There I found a listing named: FireBeetle-ESP32 Mainboard by DFRobot DFRDuino and also an option to install
But when I click on install it sits there for several minutes and finally stops and reports back with "bad CRC try later"
When I download and look at the  package_esp32_index.json file, on line 8 it has:
websiteURL: "https://github.com/espressif/esp32-arduino",
This seems to be a bad address.  It looks to be websiteURL: "https://github.com/espressif/arduino-esp32"
Is there some other way to get Arduino to see ESP32 boards and examples?
It looks like I have everything I need including a list of board in a file named: boards.txt in a folder at:
Documents\Arduino\hardware\espressif\esp32.
I also ran get.exe in the tools folder using 64bit bash tool which created the xtensa-esp32-elf
I am at a loss for a tutorial that is specific enough for me to understand what I am doing wrong.

All help is appreciated

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

pacav69, 2018-04-14:
Try these instructions
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md

copercini, 2018-07-28:
You can try with this repo .json file: https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

closed 2018-07-28 by copercini

evinracher, 2019-08-26:
Someone has information about FireBeetle-ESP32?? I am trying to probe it but I got: Error downloading https://git.oschina.net/dfrobot/FireBeetle-ESP32/raw/master/package_esp32_index.json

ASTEG012, 2019-08-28:
I have the same problem - just received the Firebeetle_ESP32 from DFRobot and followed the instructions to add the board to the IDE Boardmanager.

What I receive is: Error downloading https://gitee.com/dfrobot/FireBeetle-ESP32/raw/master/package_esp32_index.json

I tried: 
https://git.oschina.net/dfrobot/FireBeetle-ESP32/raw/master/package_esp32_index.json
and
https://gitee.com/dfrobot/FireBeetle-ESP32/raw/master/package_esp32_index.json

Always the same problem

Who knows a solution

pacav69, 2019-08-28:
try this 

    https://gitee.com/dfrobot/FireBeetle-ESP32/raw/master/package_esp32_index.json

pacav69, 2019-08-28:
follow these instructions

    https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

jrc, 2020-09-18:
Use https://raw.githubusercontent.com/DFRobot/FireBeetle-ESP32/master/package_esp32_index.json