espressif/arduino-esp32 #1015: WiFiClientSecure: self signed certificates

GermanSheepDog, 2018-01-20:
Hello,

I worked with WiFiClientSecure an the example WiFiClientSecure.ino. It worked fine.

I will used it to communicate with an Lightify Gateway and this gateway use a self signed certificate.

Now I got problems at two points of ssl_client.cpp in procedure start_ssl_client:

First point:
```
    log_v("Performing the SSL/TLS handshake...");
    while ((ret = mbedtls_ssl_handshake(&ssl_client->ssl_ctx)) != 0) {
        if (ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE) {
            Serial.println("ssl_client: handle error 1");
            return handle_error(ret);   // <--- Here I got an error!
        }
	vTaskDelay(10 / portTICK_PERIOD_MS);
    }
```

Second point:
```
    log_v("Verifying peer X.509 certificate...");
    Serial.println("ssl_client: Verifying peer X.509 certificate...");
    
    if ((flags = mbedtls_ssl_get_verify_result(&ssl_client->ssl_ctx)) != 0) {
        Serial.println("ssl_client: handle error 2");
        bzero(buf, sizeof(buf));
        mbedtls_x509_crt_verify_info(buf, sizeof(buf), "  ! ", flags);
        log_e("Failed to verify peer certificate! verification info: %s", buf);
        stop_ssl_socket(ssl_client, rootCABuff, cli_cert, cli_key);  //It's not safe continue.
        return handle_error(ret);    // <--- Here I got an error!
    } else {
        log_v("Certificate verified.");
    }

```

It could be a feature or an error.

Do anyone have a solution?

Best regards

----------------------------------------------------------------
### Hardware:
Board:						ESP32 Dev Module
Update date of WifiClientSecure:20/Jan/2018
IDE name:					Arduino IDE
Flash Frequency:				40Mhz
Upload Speed:				115200

copercini, 2018-01-22:
What errors?

GermanSheepDog, 2018-01-22:
I used this INO-File:

```
/* Wifi secure connection example for ESP32 Running on TLS 1.2 using mbedTLS
*/

#include <WiFiClientSecure.h>

const char* ssid     = "<mySSID>";     // your network SSID (name of wifi network)
const char* password = "<myPassword>"; // your network password

const char*  server1  = "www.howsmyssl.com";  // Server URL
const char*  page1   = "/a/check";
const int    sslPort1 = 443;

const char*  server  = "192.168.0.34";  // Server URL
const char*  page    = "/version";
const int    sslPort = 8443;

// www.howsmyssl.com root certificate authority, to verify the server
// change it to your server root CA
// SHA1 fingerprint is broken now!

const char* test_root_ca = \
"-----BEGIN CERTIFICATE-----\n" \
<MY CERTIFICATE> 
"-----END CERTIFICATE-----\n";

// You can use x.509 client certificates if you want
//const char* test_client_key = "";   //to verify the client
//const char* test_client_cert = "";  //to verify the client

char* bbb[80];

WiFiClientSecure client;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(100);

  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  // attempt to connect to Wifi network:
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    // wait 1 second for re-trying
    delay(1000);
  }

  Serial.print("\n\nConnected to ");
  Serial.println(ssid);

  client.setCACert(test_root_ca);
  //client.setCertificate(test_client_key); // for client verification
  //client.setPrivateKey(test_client_cert);	// for client verification
  
  Serial.println("Starting connection to "+ String(server) + ":" + String(sslPort));
  if (!client.connect(server, sslPort)) {
    Serial.println("Connection failed! Wrong certificat?\n");
    Serial.println(test_root_ca);
  }    
  else {
    Serial.println("Connected to " + String(server) + String(page));
    // Make a HTTP request:
    //client.println("GET https://www.howsmyssl.com/a/check HTTP/1.0");
    client.println("GET " + String(page) + " HTTP/1.0");
    client.println("Host: " + String(server) );
    client.println("Connection: close");
    client.println();

    Serial.println("Wait for answer");
    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
    }
    // if there are incoming bytes available
    // from the server, read them and print them:
    while (client.available()) {
      char c = client.read();
      Serial.write(c);
    }

    client.stop();
  }
}

void loop() {
  // do nothing
}
```

I got this error: "Connection failed! Wrong certificat?

Because I have expanded "ssl_client.cpp" I got at the first step this information:
"ssl_client: handle error 1"

Here are the lines:
```
   if (ret != MBEDTLS_ERR_SSL_WANT_READ && ret != MBEDTLS_ERR_SSL_WANT_WRITE) {
            Serial.println("ssl_client: handle error 1");
            return handle_error(ret); 
```
If I deleted the line "return handle_error(ret);" I got an error from this line:
```
    stop_ssl_socket(ssl_client, rootCABuff, cli_cert, cli_key);  //It's not safe continue.
    return handle_error(ret);
```

I tried to approach my problem(s).

copercini, 2018-01-22:
First, let's try to find what error it is, there are two ways to do it:
1- enable debug logs in Arduino IDE > tools > Core debug level > verbose
I think this menu is just shown to the board "ESP32 dev module" so maybe you will need to change your board...

2-or update to the [last Arduino core](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md#how-to-update-to-the-latest-code) and change your code to something like this:
```cpp
  Serial.println("Starting connection to "+ String(server) + ":" + String(sslPort));
  if (!client.connect(server, sslPort)) {
    Serial.println("Connection failed! Wrong certificat?\n");
    Serial.println(test_root_ca);  

    char err_buf[100];
    if (client.lastError(err_buf,100)<0) {
	Serial.print("Last WiFiClientSecure Error: ");
        Serial.println(err_buf);
    }
  }
```

GermanSheepDog, 2018-01-23:
First I load the regular library WiFiClientSecure from github.
Then I enabled logs and add my code.

The Monitor showed:
```
.[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:215] _eventCallback(): Event: 7 - STA_GOT_IP

Connected to Pg3
Starting connection to 192.168.0.34:8443
[V][ssl_client.cpp:48] start_ssl_client(): Free heap before TLS 137376
[V][ssl_client.cpp:50] start_ssl_client(): Starting socket
[V][ssl_client.cpp:86] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:95] start_ssl_client(): Setting up the SSL/TLS structure...
[V][ssl_client.cpp:108] start_ssl_client(): Loading CA cert
[V][ssl_client.cpp:143] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:158] start_ssl_client(): Performing the SSL/TLS handshake...
[E][ssl_client.cpp:28] handle_error(): X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[E][ssl_client.cpp:30] handle_error(): MbedTLS message code: -9984
[E][WiFiClientSecure.cpp:108] connect(): lwip_connect_r: 11
[V][ssl_client.cpp:209] stop_ssl_socket(): Cleaning SSL connection.
Connection failed! Wrong certificat?

-----BEGIN CERTIFICATE-----
<my certificate>
-----END CERTIFICATE-----

Last WiFiClientSecure Error: X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
```
Please remember, the certificate is a  self signed certificate. With Safari, Firefox, Chrom and a php-test-program it worked.

ritaru, 2018-05-02:
Have you checked the server certificate's CN?
It should match the IP of the server if you are connecting by its address.
Other methods, such as mDNS query, the CN of the server certificate should be the hostname of the server.

closed 2019-08-15 by stale[bot]