espressif/arduino-esp32 #1016: Own soldered BME680 not working properly

Skel0t, 2018-01-20:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			11/jul/2017
IDE name:							Arduino IDE
Flash Frequency:					?40Mhz?
Upload Speed:						115200

### Description:
Hello,
I have a bought BME680 that's working with the following Sketch. I'm using I2C as data bus.
Then I also have a BME680 that I soldered manually on a circuit board with 4.7k pull up resistors. This one is not working with the following sketch. It stops already when calling "bme680.begin()" in the setup() method and I'd like to know why.

Running an I2C scanner returns me 0x76 as address for the self soldered one while it gives me 0x77 for the bought one, so both are recognized as I2C devices. What could be the cause for this?

Thanks for any help!

PS: I have 3 self soldered ones that all have the same problem.

### Sketch:
https://pastebin.com/vT9BC4wf

#include <BME680_Library.h>
#include <Wire.h>
#include <BlynkSimpleEsp32.h>
#include <WiFi.h>;
#include <WiFiClient.h>;

BME680_Library bme680;

uint8_t led = 4;
WidgetLED led1(V6);
WidgetLCD lcd(V20);
int state = -1;

BLYNK_READ_DEFAULT(){
  bme680.configureForcedMode();
  bme680.read();
  Blynk.virtualWrite(2, bme680.getTemperature());
  float curr_hum =  bme680.getRelativeHumidity();
  Blynk.virtualWrite(3, curr_hum);
  Blynk.virtualWrite(4, bme680.getBarometricPressure());
  float gr = bme680.getGasResistance();
  Blynk.virtualWrite(5, gr);
  float aq = getAirQuality(gr, curr_hum);
  Blynk.virtualWrite(10, aq);
  if(aq < 50 && state != 0){
    lcd.clear();
    state = 0;
    lcd.print(0, 0, "Open Windows!");
    lcd.print(0, 1, "U evn breathin?");
  } else if (aq >= 50 && aq < 76 && state != 1){
    lcd.clear();
    state = 1;
    lcd.print(0, 0, "Everything Okay");
  } else if (aq >= 75 && state != 2){
    lcd.clear();
    state = 2;
    lcd.print(0,0, "Perfect Air!");  
  }
}

BLYNK_WRITE(V1){
  ledcWrite(1, param.asInt());
  led1.setValue(param.asFloat());
}

void setup(void){
  Serial.begin(115200);
  ledcAttachPin(led, 1);
  ledcSetup(1, 12000, 8);

  Blynk.begin(auth key, WLAN, WLAN_PASS, LocalServerIPAddress);
  
  Wire.begin();
  Serial.print("BME Initialization...");
  if(bme680.begin()){
     Serial.print("Succeeded!");
  } else {
    Serial.print("Failed!");
    for(;;); // spin forever
  }

  Serial.print("Configuring Forced Mode...");
  if(bme680.configureForcedMode()){
     Serial.print("Succeeded!");
  } else {
    Serial.print("Failed!");
    for(;;); // spin forever
  }
}

void loop(void){
  Blynk.run();
}

float getAirQuality(float curr_gr, float curr_hum){
  
  float hum_score, gas_score;
  float current_humidity = curr_hum;
  int hum_reference = 40;
  int gas_reference = 250000;
  if (current_humidity >= 38 && current_humidity <= 42)
    hum_score = 0.25*100; // Humidity +/-5% around optimum 
  else
  { //sub-optimal
    if (current_humidity < 38) 
      hum_score = 0.25/hum_reference*current_humidity*100;
    else
    {
      hum_score = ((-0.25/(100-hum_reference)*current_humidity)+0.416666)*100;
    }
  }
  
  //Calculate gas contribution to IAQ index
  int readings = 10;
  for (int i = 0; i <= readings; i++){ // read gas for 10 x 0.150mS = 1.5secs
    gas_reference += curr_gr;
  }
  gas_reference = gas_reference / readings;
  
  int gas_lower_limit = 50000;   // Bad air quality limit
  int gas_upper_limit = 500000;  // Good air quality limit 
  if (gas_reference > gas_upper_limit) gas_reference = gas_upper_limit; 
  if (gas_reference < gas_lower_limit) gas_reference = gas_lower_limit;
  gas_score = (0.75/(gas_upper_limit-gas_lower_limit)*gas_reference -(gas_lower_limit*(0.75/(gas_upper_limit-gas_lower_limit))))*100;
  return hum_score + gas_score;  
}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078ad0
BME Initialization...
```

stickbreaker, 2018-01-20:
@Skel0t The SDO line is use to select the address in I2C mode as per the datasheet:
[image]

If you want the Address to be 0x77 connect it to Vddio

Chuck.

Skel0t, 2018-01-20:
I connected it and now it works, thank you!

closed 2018-01-20 by Skel0t