espressif/arduino-esp32 #1018: First scan in WifiMulti runs synchronously and blocks device for few seconds

9a4gl, 2018-01-21:
### Hardware:
Board:							lolin32 lite
Core Installation/update date:			?
IDE name:						Platform.io
Flash Frequency:					?
Upload Speed:					115200

### Description:
In 9dae9cea373abac265f5ed28210bffaee78c92fd run method has been changed to call
WiFi.scanNetworks() instead WiFi.scanComplete()

As scanNetwork is called with any parameter it performs synchronous scanning causing program flow to block for few seconds.
As this is called when app starts it cause a lot of problem to other initialization flows, like it disrupts serial communication with my Nextion display, causing some bytes to miss on serial interface.

9a4gl, 2018-01-21:
I reverted that patch and everything is working fine. Looks initial problem is fixed and this patch should be reverted.

copercini, 2018-01-30:
My tests:
reverting `WiFi.scanNetworks();` to `WiFi.scanComplete();` it's working fine and asynchronously
but fully reverting the commit, it stuck =/
so keep `status != WL_CONNECTED` looks lika a good ideia

What do you think @9a4gl  ?

9a4gl, 2018-01-30:
Have you traced where and why it get stuck ? I do not remember I had such issues.
Currently in my project I am using more customized version of this libarary. I didn't like after Wifi.begin we have loop waiting to get connected. It bloked my app for few seconds each time it get reconnected. I may try to provide you this fully non-blocking version, but I am afraid it is not oneliner.

9a4gl, 2018-01-30:
Here is small update after some testing. So, my version which do scan (and connect) asynchronously works fine for first scan, it connects and works. But, if it loose network, new scan does not work, actually it finds something on each cca 20th scan, and then it fails to connect.
I would say there is a bug in asynchronous scanning, so having scanNetwork() (which actually do scan synchronously) instead scanComplete() (which just check if previously called scanNetwork(true) finished) is a hack to make it work. Looks we are not ready for this yet for this revert. And I am bit worried we have such bugs on this platform :(

closed 2019-08-15 by stale[bot]