espressif/arduino-esp32 #1020: I2C communication error.

BlackTiger44, 2018-01-21:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

Hello everyone.
I have a PCF8574A port that randomly gives me the following error:

[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 21

The PCF8574A is at address 21.

The code of the function that I use to perform the writing operations is the following:

void WriteDataPortPCF8574A(byte device,byte data)
  {
  Wire.beginTransmission((int)device);
  Wire.write(data);
  Wire.endTransmission(); 
  }

And the configuration of the I2C bus

void setup() 
{

Wire.begin(21,22,100000);
WriteDataPortPCF8574A(AdrPortOut0PCF8574T,B11111111);

 }

Can someone please tell me what is causing the failure in I2C communications with my PCF8574 port.

Thank you BlackTiger44.
```

everslick, 2018-01-22:
This is a known issue and top priority to the devs right now. Should hopefully be fixed in a few days.

BlackTiger44, 2018-01-23:
Thanks everslick I did not know about that fault in our ESP32, I hope it will be fixed soon.

Greetings BlackTiger44

JavierPascual, 2018-02-03:
Someone knows when it will be fixed? Thanks! :-)

stickbreaker, 2018-03-16:
@BlackTiger44 @JavierPascual I have a branch that may work for you.
* last stable v0.2.0 release [stickbreaker-arduino-esp32](https://github.com/stickbreaker/arduino-esp32)
* new branch that includes a Destructor function for `Wire()` so that you can use both i2c peripherals. [Wire-Destructor](https://github.com/stickbreaker/arduino-esp32/tree/Wire-Destructor)

Either one should work for you. After more testing, I plan on merging `Wire-Destructor` into my main.

Chuck.

BlackTiger44, 2018-03-18:
Thanks for contributing your work Stickbreaker.

The files to correct the failure of the ESP32 on the I2C bus are the following ??.

libraries\Wire\src
Wire.h
Wire.cpp
cores\esp32
esp32-hal-i2c.h
esp32-hal-i2c.h

A greeting
BlackTiger

stickbreaker, 2018-03-18:
@BlackTiger44 navigate to branch [Wire-Destructor](https://github.com/stickbreaker/arduino-esp32/tree/Wire-Destructor).  This is where I am uploading revisions that handle creating and destroying `TwoWire::` class objects.  I have some new changes that can handle multiple `Wire.begin()` statements without causing problems.

I found a problem when I was testing: If I specified the `sclpin` and 'sdapin` as something other than the default defined `SCL` `SDA`, i2c device libraries (liquidcrystal,, ds1307..) would reinit the pin assigments back to default,  Since the Default pins were not physically connected it failed.

I am still testing, I plan on uploading to [Wire-Destructor](https://github.com/stickbreaker/arduino-esp32/tree/Wire-Destructor) later today if my testing succeeds.

I am also testing multiple masters on the same bus.  I have a sketch that has one task using the second i2c hardware peripheral to access a 4x4 keypad via MCP23008, and a 24x1 LCD via a MCP23017.  While this is happening, the foreground task is using the first i2c peripheral to access a 20x4 lcd via MCP23008, and multiple 24LCxx EEPROMS,  all of these devices are connected to the same i2c bus, so the two peripherals are fighting for bus ownership. I am working through the Arbitration backout, retry and polite design rules.  I'm having to re-code my device libraries to logically recover from an arbitration loss.  This has stretched my brain. Deciding when to fail,(stop retrying), how long to wait for a clear bus, when to manually clear the bus.  It's kind of fun.

Chuck.

lbernstone, 2018-03-18:
You have a strange idea of fun, but thanks for all your work on this!

BlackTiger44, 2018-04-15:
Hi.

I have been testing in my program the files modified for the I2C bus by stickbreaker, and the truth is that I do not get stuck with the program, but even then the following error message comes out from time to time

[I][esp32-hal-i2c.c:971] i2cProcQueue(): Bus busy, reinit

But even when the aforementioned error message comes out, my program continues to work.

I have put "Wire.setTimeout(500);"  but I have not noticed any improvement.

Thanks stickbreaker for your work.

Greetings  
BlackTiger

stickbreaker, 2018-04-15:
@BlackTiger44 What i2c devices are you using?  Bus busy is reporting SDA or SCL are not in the Idle state (high).  Either you have a bus problem, or one of your devices is stretching SCL to pause the ESP during a master mode transmission. 

Chuck.

BlackTiger44, 2018-04-16:
Hello, stickbreaker.
To the I2C bus I only have one PCF8574AT connected, with which I control several relays, each one of the outputs is activated giving a pulse of about one second, and the outputs of the PCF8574AT I use to activate the interior lights of the garage, to activate the lights Outside of the garage, open and close garage doors, to know for example that the interior lights of the garage have been activated, use one of the two auxiliary contacts of the relay that controls the mentioned lights, taking it to an entrance of the ESP32. My program does not give any error message as long as I do not connect the auxiliary contact to the ESP32 input, but as soon as I connect the auxiliary contact it is when the problems start.

Greetings and thanks

BlackTiger

stickbreaker, 2018-04-17:
Post a schematic of your circuit, i think it is an- electrical problem

Chuck

BlackTiger44, 2018-04-19:
Hi stickbreaker.

I have made the electrical diagram, if something is not clear to you please tell me.
Thanks for your help.

a greeting
BlackTiger

[image]

stickbreaker, 2018-04-19:
@blacktiger44 Do you have any info on your relay module? I am questioning the current required to activate the relays. The pca 8574 is limited to 25mA per output 200mA max. 
Chuck

stickbreaker, 2018-04-19:
@blacktiger44 I just noticed you have the pca8574 powered by 3.3v but the relay module is powered by 5v.  The output stages of the '74 connect to 3.3v via pmosfet, you could be back feeding 5v into the 3.3v rail. Try measuring voltage on the output pins when the signal is off. if it is above 3.3, you are going to have to change the circuit.

I would change the circuit, power the '74 from 5v. Put a level shifter between esp32 and '74.

Chuck

BlackTiger44, 2018-04-19:
Thanks stickbreaker

Relay module information, as you can see the coils of the relays of the module work at 5V.

https://es.aliexpress.com/item/Brand-New-5V-4-Channel-Relay-Module-for-Arduino-PIC-ARM-DSP-AVR-Raspberry-Pi/32266538898.html?spm=a219c.10010108.1000016.1.3e485ae2Ylxiv4&isOrigTitle=true

Try to feed the PCF8574A to 5V.

Thanks again
BlackTiger.

BlackTiger44, 2018-04-22:
Hi stickbreaker.

I have fed the PCF8574 to 5V, but there has not been any improvement, even disconnecting the relay RL1 sometime I get a fault message on the I2C bus.

In the end I decided to use the digital outputs of the ESP32 itself, to activate the relays of the module, only one current of 2mA is necessary.

I hope and wish that espressif solve this problem of ESP32, since it is a great microcontroller.

Thanks for your help stickbreaker.

a greeting
BlackTige

copercini, 2018-07-01:
The I2C core was changed to @stickbreaker code officially after https://github.com/espressif/arduino-esp32/commit/13dcfe5c13bc882f8923d137ed604aa65b6db057 (thanks @stickbreaker), Let's close issues with old code and if have problems with the new code, open new issues =)

closed 2018-07-01 by copercini