espressif/arduino-esp32 #1023: HTTP requests from curl or wget succeed but browser requests hang or time out

tony-ozrelic, 2018-01-23:
### Hardware:
Board:							adafruit ESP32 Feather
Core Installation/update date:		IDK
IDE name:						Arduino 1.8.5
Flash Frequency:					80MHz
Upload Speed:					921600

### Description:
Using the webserver_tng class, I can request web pages using curl or wget with little or no problem. Using Chrome or Edge (haven't tried firefox yet) and WebServer::handleClient() gets a new client, but then it either times out or the state _currentStatus goes back to HC_NONE ( 'no client').

Going back to curl or wget, things work fine. Back to the browser, and it will occasionally fetch the page but more often than not it hangs, Sometimes after many seconds (10-20?) the server will return the page.

Does anybody else experience this? Are there any fixes or workarounds? Is some task starved for time?

### Sketch:

### Debug Messages:

lbernstone, 2018-01-24:
In Chrome, you can go int.o developer tools, and click on network.  Make the request, and see what is taking so long.

stickbreaker, 2018-01-24:
@tony-ozrelic are the browsers looking (sending an additional request) for the .icon file? 
What does your response block look like? 
Are you issuing a Connection: Close?
If you don't close the connection, the .icon request will reuse the previous connection.  If you have a loop waiting for the connection to close, you will wait for a while.

Chuck.

tony-ozrelic, 2018-01-24:
I don't have favicon.ico implemented, so the default is to return a 404 Not Found. 
As for using Chrome's dev tools, What I have found is that Chrome sends the request typically in <100usecs, then there is a long pause, typically 1 to 10 seconds, then the ESP32 responds, or Chrome decides there is no response and shows the 'This Site Can't be reached' (ERR_CONNECTION_RESET). 

Clicking on RELOAD sometimes causes the connection to complete, and the network timing tool shows the request issued in 32usecs or so, then a long pause waiting (TTFB) of several seconds, then a Content Download of 23msecs.

tony-ozrelic, 2018-01-24:
sorry, clicked wrong button...

tony-ozrelic, 2018-01-25:
when there is no response from the ESP32, Chrome shows it as 'Stalled' in the network timing tool. From the dev tools:

Stalled. The request could be stalled for any of the reasons described in Queueing.
Queueing. The browser queues requests when:
There are higher priority requests.
There are already six TCP connections open for this origin, which is the limit. Applies to HTTP/1.0 and HTTP/1.1 only.
The browser is briefly allocating space in the disk cache
Even with only one tab open, Chrome stalls.

AndreaBorgia-Abo, 2018-12-04:
@tony-ozrelic do you think the issue I described in #1787 could be related or similar?

tony-ozrelic, 2018-12-04:
no idea, it was almost a yr and several projects ago.

did you try something other than chrome?

AndreaBorgia-Abo, 2018-12-05:
> no idea, it was almost a yr and several projects ago. did you try something other than chrome?

I haven't used Chrome at all, but I've tried with Firefox, Konqueror and just plain telnet

tony-ozrelic, 2018-12-05:
hmm, if you can, try wget or curl.

Also try commenting out all code EXCEPT for the web servicing stuff and see
if that changes anything.

If I recall correctly, I had to reorganize everything around a round-robin
task manager that called my wifi task as much as possible in order to make
sure that a request did not fall thru the cracks.

The other thing I did was to create an inline my_yield() function to call
yield() if too much time was spent in some loop waiting for a button press
or some other activity:

inline void my_yield(void)
{
static uint32_t t;

if((micros()-t) > 10000)
{
yield();
t = micros();
}
}

This allowed code to run but then call yield() to let the ESP8266 do stuff
if it took too long, in this case 10 000 usecs or 10msecs.

tony-ozrelic, 2018-12-05:
> Also tried wget, surely. Not sure about curl. Thanks for the tips, now I
> see your issue is a bit different: in my case working code stopped working
> after a core change and going back to the earlier version makes it work
> again. Too bad I can't do that with the current branch.

Could you get clues from the bugs fixed in the core change?
Or is that too complicated?

closed 2019-08-15 by stale[bot]

mhilditch, 2024-01-15:
I stumbled on this issue while looking to solve another similar problem so thought I'd share my findings here for future travellers.
It turned out my issue was caused by a site using http:// rather than https:// as the site does not support secure connections. The feature of Chrome blocking it was HSTS which kept trying to automatically redirect to HTTPS:// and causing the page to not load.

Here's the instructions to resolve that as found at https://www.thesslstore.com/blog/clear-hsts-settings-chrome-firefox/  

**How to Delete HSTS Settings in Chrome:**
Navigate to chrome://net-internals/#hsts
This is Chrome’s UI for managing your browser’s local HSTS settings.

First, to confirm the domain’s HSTS settings are recorded by Chrome, type the hostname into the Query Domain section at the bottom of the page. Click Query.If the Query box returns Found with settings information below, the domain’s HSTS settings are saved in your browser.
HSTS Settings Chrome
Note that this is a very sensitive search. Only enter the hostname, such as www.example.com or example.com without a protocol or path.

Type the same hostname into the Delete domain section and click
Your browser will no longer force an HTTPS connection for that site! You can test if its working properly by refreshing or navigating to the page.

Note that depending on the HSTS settings provided by the site, you may need to specify the proper subdomain. For example, the HSTS settings for staging.yoursite.com may be separate from yoursite.com so you may need to repeat the steps as appropriate.