espressif/arduino-esp32 #1024: xtensa-esp32-elf-g++: exec format error

jrullan, 2018-01-24:
### Hardware:
Board:							MH-ET MiniKit ESP32
Core Installation/update date:			23/Jan/2018
IDE name:						Arduino IDE 1.8.5 (Running on Armbian on Pine64+)
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
Hitting the Verify button in the Arduino IDE immediatly throws the error:
fork/exec /home/jrullan/Arduino/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++: exec format error

### Sketch:
```cpp

uint64_t chipid;  

void setup() {
	Serial.begin(115200);
}

void loop() {
	chipid=ESP.getEfuseMac();//The chip ID is essentially its MAC address(length: 6 bytes).
	Serial.printf("ESP32 Chip ID = %04X",(uint16_t)(chipid>>32));//print High 2 bytes
	Serial.printf("%08X\n",(uint32_t)chipid);//print Low 4bytes.

	delay(3000);

}
```

### Debug Messages:
```

```

lbernstone, 2018-01-24:
There is no packaged version of the toolchain for ARM.  You will need to roll your own, or else use a supported platform.

lbernstone, 2018-01-24:
http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup-scratch.html

jrullan, 2018-01-25:
# Got it working!
I will post here my notes and results so others know what to expect!
I did this in a Pine64+ ARM (a 64bit ARM SBC).
Before running this I had installed Arduino IDE with this script: https://github.com/pfeerick/pine64-scripts/blob/master/install-arduino-ide.sh

To install ESP32 toolchain on Armbian:

See: http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup-scratch.html

1. Install Prerequisites:
`sudo apt-get install git wget make libncurses-dev flex bison gperf python python-serial
`

2. Install Dependencies:
`sudo apt-get install gawk grep gettext python-dev automake texinfo help2man libtool libtool-bin`

3. Download crosstool-NG 

```
cd ~/esp
git clone -b xtensa-1.22.x https://github.com/espressif/crosstool-NG.git
cd crosstool-NG
./bootstrap && ./configure --enable-local && make install

```

4. Build

```
./ct-ng xtensa-esp32-elf
./ct-ng build
chmod -R u+w builds/xtensa-esp32-elf

```

5. Toolchain will be built in ~/esp/crosstool-NG/builds/xtensa-esp32-elf. 
Follow instructions for standard setup to add the toolchain to your PATH.

Installation output (**So you know what to expect!**)

```
xxxxxxx@pine64:~/crosstool-NG$ ./ct-ng xtensa-esp32-elf
  MKDIR config.gen
  IN    config.gen/arch.in
  IN    config.gen/kernel.in
  IN    config.gen/cc.in
  IN    config.gen/binutils.in
  IN    config.gen/libc.in
  IN    config.gen/debug.in
  CONF  config/config.in
#
# configuration written to .config
#

***********************************************************

Initially reported by: Ivan Grokhotkov <ivan@espressif.com>
URL: 

Comment:
Based on xtensa-esp108-elf sample reported by Max Filippov <jcmvbkbc@gmail.com>

***********************************************************

WARNING! This sample may enable experimental features.
         Please be sure to review the configuration prior
         to building and using your toolchain!
Now, you have been warned!

***********************************************************

Now configured for "xtensa-esp32-elf"
xxxxxxx@pine64:~/crosstool-NG$ ./ct-ng build
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20180124.175557
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 1340.35s (at 22:25)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[INFO ]  Extracting and patching toolchain components: done in 302.74s (at 27:28)
[INFO ]  =================================================================
[INFO ]  Installing ncurses for build
[INFO ]  Installing ncurses for build: done in 157.91s (at 30:06)
[INFO ]  =================================================================
[INFO ]  Installing GMP for host
[INFO ]  Installing GMP for host: done in 181.29s (at 33:07)
[INFO ]  =================================================================
[INFO ]  Installing MPFR for host
[INFO ]  Installing MPFR for host: done in 138.04s (at 35:25)
[INFO ]  =================================================================
[INFO ]  Installing ISL for host
[INFO ]  Installing ISL for host: done in 149.44s (at 37:55)
[INFO ]  =================================================================
[INFO ]  Installing MPC for host
[INFO ]  Installing MPC for host: done in 63.21s (at 38:58)
[INFO ]  =================================================================
[INFO ]  Installing expat for host
[INFO ]  Installing expat for host: done in 32.46s (at 39:30)
[INFO ]  =================================================================
[INFO ]  Installing binutils for host
[INFO ]  Installing binutils for host: done in 376.66s (at 45:47)
[INFO ]  =================================================================
[INFO ]  Installing C library headers & start files
[INFO ]  Installing C library headers & start files: done in 0.19s (at 45:47)
[INFO ]  =================================================================
[INFO ]  Installing pass-2 core C gcc compiler
[INFO ]  Installing pass-2 core C gcc compiler: done in 1996.04s (at 79:03)
[INFO ]  =================================================================
[INFO ]  Installing C library
[INFO ]  Installing C library: done in 245.31s (at 83:09)
[INFO ]  =================================================================
[INFO ]  Installing final gcc compiler
[INFO ]  Installing final gcc compiler: done in 2581.76s (at 126:10)
[INFO ]  =================================================================
[INFO ]  Installing cross-gdb
[INFO ]  Installing cross-gdb: done in 744.19s (at 138:35)
[INFO ]  =================================================================
[INFO ]  Cleaning-up the toolchain's directory
[INFO ]    Stripping all toolchain executables
[INFO ]  Cleaning-up the toolchain's directory: done in 10.38s (at 138:45)
[INFO ]  Build completed at 20180124.201442
[INFO ]  (elapsed: 138:44.91)
[INFO ]  Finishing installation (may take a few seconds)...
[138:45] / xxxxxxx@pine64:~/crosstool-NG$ chmod -R u+w builds/xtensa-esp32-elf

```

closed 2018-01-25 by jrullan

BlrUsr, 2018-01-25:
Pine64 script to install arduino.  (PINE A64 is Single Board Computer powered by Quad-Core ARM Cortex A53 64-Bit Processor. )
But I use Orange PI PC2 (H5 High Performance Quad-core 64-bit Cortex-A53) and os Armbian .
It will be worked on Orange PI PC2?
Thanks.

BlrUsr, 2018-01-26:
Ok I compile it all.
But when I start ./ct-ng build
at
[INFO] Installing pass-2 core C gcc compiler 
[429:20] | 

I'm wait 12 hours , but it not response , and armbian not response . 
Help please to install it.

BlrUsr, 2018-01-30:
pi@orangepipc2:~/esp/crosstool-NG$ ./ct-ng build
[INFO ]  Performing some trivial sanity checks
[INFO ]  Build started 20180129.083432
[INFO ]  Building environment variables
[INFO ]  =================================================================
[INFO ]  Retrieving needed toolchain components' tarballs
[INFO ]  Retrieving needed toolchain components' tarballs: done in 91.28s (at 01:35)
[INFO ]  =================================================================
[INFO ]  Extracting and patching toolchain components
[INFO ]  Extracting and patching toolchain components: done in 348.94s (at 07:24)
[INFO ]  =================================================================
[INFO ]  Installing ncurses for build
[INFO ]  Installing ncurses for build: done in 154.18s (at 09:58)
[INFO ]  =================================================================
[INFO ]  Installing GMP for host
[INFO ]  Installing GMP for host: done in 174.38s (at 12:53)
[INFO ]  =================================================================
[INFO ]  Installing MPFR for host
[INFO ]  Installing MPFR for host: done in 126.54s (at 14:59)
[INFO ]  =================================================================
[INFO ]  Installing ISL for host
[INFO ]  Installing ISL for host: done in 139.24s (at 17:19)
[INFO ]  =================================================================
[INFO ]  Installing MPC for host
[INFO ]  Installing MPC for host: done in 58.68s (at 18:17)
[INFO ]  =================================================================
[INFO ]  Installing expat for host
[INFO ]  Installing expat for host: done in 32.68s (at 18:50)
[INFO ]  =================================================================
[INFO ]  Installing binutils for host
[INFO ]  Installing binutils for host: done in 322.32s (at 24:12)
[INFO ]  =================================================================
[INFO ]  Installing C library headers & start files
[INFO ]  Installing C library headers & start files: done in 0.25s (at 24:13)
[INFO ]  =================================================================
[INFO ]  Installing pass-2 core C gcc compiler
[ERROR]    g++: internal compiler error: Killed (program cc1plus)
[ERROR]    make[2]: *** [dwarf2out.o] Error 4
[ERROR]    make[1]: *** [all-gcc] Error 2
[ERROR]  - 
[ERROR]  >>
[ERROR]  >>  Build failed in step 'Installing pass-2 core C gcc compiler'
[ERROR]  >>        called in step '(top-level)'
[ERROR]  >>
[ERROR]  >>  Error happened in: CT_DoExecLog[scripts/functions@257]
[ERROR]  >>        called from: do_gcc_core_backend[scripts/build/cc/100-gcc.sh@545]
[ERROR]  >>        called from: do_gcc_core_pass_2[scripts/build/cc/100-gcc.sh@160]
[ERROR]  >>        called from: do_cc_core_pass_2[scripts/build/cc.sh@42]
[ERROR]  >>        called from: main[scripts/crosstool-NG.sh@646]
[ERROR]  >>
[ERROR]  >>  For more info on this error, look at the file: 'build.log'
[ERROR]  >>  There is a list of known issues, some with workarounds, in:
[ERROR]  >>      'docs/B - Known issues.txt'
[ERROR]  / 
[ERROR]  (elapsed: 974:53.56)
[974:54] / ct-ng:152: recipe for target 'build' failed
make: *** [build] Error 2
pi@orangepipc2:~/esp/crosstool-NG$ 

Error please help

jrullan, 2018-02-02:
@BlrUsr, I don't know what the error is about but regarding your question about the Pine64 script vs your particular architecture, if you look at the script in an editor you will see it's just a convenient collection of steps to set up the Arduino IDE in our platforms. 

Before using the script I got it working using these steps: https://forum.pine64.org/showthread.php?tid=1753&pid=28870#pid28870

skaldo, 2022-10-02:
Just in case someone else stumbles upon this issue (unable to execute toolchain-xtensa32 on aarch64 machine),

I just found that the 2.50200.80 works just fine, but 2.50200.97 does not.

```
root@5c53de3b-esphome:/# ./root/.platformio/packages/toolchain-xtensa32@2.50200.80/bin/xtensa-esp32-elf-g++
xtensa-esp32-elf-g++: fatal error: no input files
compilation terminated.
root@5c53de3b-esphome:/# ./data/cache/platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++
bash: ./data/cache/platformio/packages/toolchain-xtensa32/bin/xtensa-esp32-elf-g++: cannot execute binary file: Exec format error
root@5c53de3b-esphome:/# cat /data/cache/platformio/packages/toolchain-xtensa32/package.json 
{
  "name": "toolchain-xtensa32",
  "version": "2.50200.97",
  "description": "GCC Toolchain for Xtensa32 processor",
  "keywords": [
    "build tools",
    "compiler",
    "assembler",
    "linker",
    "preprocessor",
    "esp32"
  ],
  "license": "GPL-2.0-or-later",
  "repository": {
    "type": "git",
    "url": "https://github.com/espressif/esp-idf"
  },
  "system": [
    "linux_armv6l",
    "linux_armv7l",
    "linux_armv8l",
    "linux_aarch64"
  ]

```