espressif/arduino-esp32 #1025: Latest pull causing core panic and stackoverflows

DasVoelkel, 2018-01-24:
### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
commit 7fa8cafb27a1e28e3da74c5d44d4b0c50a252f05 
After pulling, my previously working code completly crashed out.
I get core panics and I get some StackOverflow errors which my system didn't have before.

Note to the traceback, I am not named Users/ficeto , i have no idea where that comes from.

One of the few errors:
***ERROR*** A stack overflow in task loopTask has been detected.

abort() was called at PC 0x40088034 on core 1

Backtrace:
 0x40087f1c:0x3ffcd3e0 0x4008801b:0x3ffcd400 0x40088034:0x3ffcd420 
0x40085823:0x3ffcd440 0x40086d78:0x3ffcd460 0x40086d2e:0x00000000

Decoded:
0x40088034:
 vApplicationStackOverflowHook at 
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c
 line 578
0x40087f1c: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x4008801b: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x40088034:
 vApplicationStackOverflowHook at 
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c
 line 578
0x40085823: vTaskSwitchContext at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4571
0x40086d78: _frxt_dispatch at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portasm.S line 406
0x40086d2e: _frxt_int_exit at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portasm.S line 206

```
//I think part of the issue is caused by creating a server or socket task, but I am not sure
WiFiServer server(TCP_PORT);
  WebSocketServer webSocketServer;

  server.begin();

state_Machine.debugprintln("loop to determin networkk ssid and password");
  while(!network._know_ssid_pw){
    /* listen for client */
    WiFiClient client = server.available();
    uint8_t data[DATA_TRANSFER_BUFFER];
    if (client) {
      Serial.println("\n new client");
      
      //timeout_start=millis();
      while (client.connected()) {
        if (client.available()) {
         // timeout_start=millis();
          int len = client.read(data, DATA_TRANSFER_BUFFER);
          if (len < TCP_IN_BUFFER_ESSENTIALS) {
            data[len] = '\0';
          } else {
            data[DATA_TRANSFER_BUFFER] = '\0';
          }
          bool x=handle_info(String((char*)data));
          Serial.println("handled received");
          if(x)
          break;
        }
      }
      Serial.println("Client disconnected\n");
      client.stop();
    }

  }
  server.end();

```
### Debug Messages:
```
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 1 - SCAN_DONE
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 7 - STA_GOT_IP
CORRUPT HEAP: multi_heap.c:370 detected at 0x3ffcd550
abort() was called at PC 0x400876c3 on core 1

Backtrace: 0x40087f1c:0x3ffcd480 0x4008801b:0x3ffcd4a0 0x400876c3:0x3ffcd4c0 0x40087c02:0x3ffcd4e0 0x40083c80:0x3ffcd500 0x40083cb1:0x3ffcd520 0x400842e9:0x3ffcd540 0x4000beaf:0x3ffcd560 0x40084d9a:0x3ffcd580 0x401014a9:0x3ffcd5a0 0x40102cf4:0x3ffcd5c0 0x40101b44:0x3ffcd5e0 0x400f0cf1:0x3ffcd620 0x400d39ae:0x3ffcd640 0x400d2832:0x3ffcd680 0x400d1a4f:0x3ffd11c0 0x400d1a6b:0x3ffd11e0 0x401183e4:0x3ffd1200

Rebooting...
⸮⸮
`
⸮HJJH⸮ !⸮!B
```

DasVoelkel, 2018-01-24:
After looking into it further, I found out after the connection worked (with  ```WiFi.begin(ssid,pw);``` and ```WiFi.status();``` responds ```WL_CONNECTED``` if i just put a delay after it ``` delay(5000); ``` it dies and retourns errors:
Stackoverflowerror:
```
***ERROR*** A stack overflow in task loopTask has been detected.
abort() was called at PC 0x40088034 on core 1

Backtrace: 0x40087f1c:0x3ffcd4d0 0x4008801b:0x3ffcd4f0 0x40088034:0x3ffcd510 0x40085823:0x3ffcd530 0x40086d78:0x3ffcd550 0x40086d2e:0x3ffcd560

Rebooting...
```

DECODED:
```

Decoding 7 results
0x40088034: vApplicationStackOverflowHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x40087f1c: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x4008801b: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x40088034: vApplicationStackOverflowHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 578
0x40085823: vTaskSwitchContext at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4571
0x40086d78: _frxt_dispatch at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portasm.S line 406
0x40086d2e: _frxt_int_exit at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./portasm.S line 206
```

DasVoelkel, 2018-01-24:
Another error that occurs:
```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU1)
Register dump:
PC      : 0x400de4ca  PS      : 0x00060d34  A0      : 0x8008610c  A1      : 0x3ffcd8c0  
A2      : 0x00000008  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x3ffcdf00  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc3e6c  A9      : 0x3ffc3e50  
A10     : 0x00000000  A11     : 0x00000001  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060021  A15     : 0x00060023  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400de4ca:0x3ffcd8c0 0x40086109:0x3ffcd8e0

Rebooting...
```
DECODED:
```

Decoding 3 results
0x400de4ca: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 85
0x400de4ca: esp_vApplicationIdleHook at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./freertos_hooks.c line 85
0x40086109: prvIdleTask at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./tasks.c line 4571

```

This is caused by my state machine ( that I built around the programme) switches out of the state which connects the wifi, into the next one. for some reason, this is only the case with a few states and not others, but there is no difference between them. Since there was no error here before I can't find any here either. 
Were any changes made that could effect these systems?

CelliesProjects, 2018-01-24:
Noticed that as well. 
I reverted back to 70d0d4648766cea047613062ed50a9ecfc9de31f and can work again.

Also, the current master 7fa8caf reported a `v3.0-dev-xxxx-xxxxxxxxx-dirty` version number in the  `ESP.getSdkVersion()` call.

DasVoelkel, 2018-01-24:
Thanks for the reply ! i will revert back to the library and try my code again to see if the bug persists.

DasVoelkel, 2018-01-26:
I think i've found the error.
 previously 
```
client.read(BUFFER,BUFFER_SIZE);
```
has worked, it pulled all available data on the client ( if one was connected) into the buffer.

I am getting all kinds of issues with all 
wifi.peek
wifi.available
wifi.read(buffer,length);

but mainly core panics and core crashes, all of them errors i dind't have before the latest update, but after reverting the changes they are still there ? i am not sure why, maybe i am viewing a different commit incorrectly.

CelliesProjects, 2018-01-26:
What does `ESP.getSdkVersion()' report?

probonopd, 2018-02-12:
I am affected, too: https://github.com/hitokuno/ESP32-SensorTag/issues/1

perll, 2018-03-31:
Me too, in a Wifi conn with GATT_SCAN.
Some news?
Regards!

CelliesProjects, 2018-03-31:
This week I completely uninstalled and deleted the complete Arduino IDE plus all ESP32 related stuff (including libs) and started fresh.

- Arduino IDE 1.8.5.
- ESP32 Arduino core @ commit 3a4ec66d41615cbb1c3e830cb6e761cdc4cca9d3
- All the latest versions of libraries I use.

Did it again on my laptop and it also is now compiling as expected.
A bit of work/time, but am I coding more confident and happy as the weird errors have disappeared.
YMMV.

tsathishkumar, 2018-07-04:
Bumping up this thread to see if there are any updates. Facing the same error sometime after connecting to WiFi.

closed 2019-08-15 by stale[bot]