espressif/arduino-esp32 #1026: Help to install ESP32 Arduino >IDE on ARMBIAN (UBUNTU)

BlrUsr, 2018-01-24:
I have minipc orange pi pc2 on H5 allwinner , when i try to add repository 

<<
pi@orangepipc2:/opt/arduino-1.8.5/hardware/espressif/esp32/tools$ python get.py
System: Linux, Info: Linux-4.14.14-sunxi64-aarch64-with-Ubuntu-16.04-xenial
Platform: x86_64-pc-linux-gnu
Downloading xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz
Done
Extracting xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz
Downloading mkspiffs-0.2.1-linux64.tar.gz
Done
Extracting mkspiffs-0.2.1-linux64.tar.gz
Renaming mkspiffs-0.2.1-linux64 to mkspiffs
Done
>>

ok?

and try to comile simple blynk project , and get error

<< 
fork/exec /opt/arduino-1.8.5/hardware/espressif/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++: exec format error
Error compiling for board ESP32 Dev Module.
>>

How resolved it?

lbernstone, 2018-01-24:
http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup-scratch.html

jrullan, 2018-01-25:
Coincidentally I was looking for the same answer  hours before you. I got it working see this post: https://github.com/espressif/arduino-esp32/issues/1024

everslick, 2018-04-19:
Seems to be solved -> closing.

closed 2018-04-19 by everslick