espressif/arduino-esp32 #1027: Getting ????? on serial monitor.

bkrajendra, 2018-01-25:
### Hardware:
Board:							ESP-WROOM-32
Core Installation/update date:			?25/01/2018?
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I got this new Module from one of the chinese  makers.
I've design a PCB for it and connected it to serial port.
Its continuously generating "?" symbols on serial monitor.
I've tried changing baud rate to understand it but no use.
It generated very fast.
I do get initial boot sequence messages.

### Sketch:
I'm trying to upload code but its not helping out. It does not respond to any boot mode conf.

### Debug Messages:
```
⸮⸮}jun⸮⸮>2⸮N⸮⸮:22:5

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read e22, ⸮⸮⸮A⸮⸮⸮⸮⸮⸮⸮&⸮⸮⸮1 
configsip: 0, SPIWP:0x00
clk_drv:0x00,qCdJ#⸮0⸮⸮⸮d6⸮⸮00,c3⸮(⸮⸮i0⸮⸮bB⸮}⸮⸮⸮⸮0x00,wp_drv:0x00
mode:DIO, clock div:0
load:⸮⸮fff0008,len:8
load:0x3fff001⸮⸮⸮n:⸮⸮u⸮⸮+m⸮⸮m⸮ݾ⸮⸮z⸮⸮ݺ⸮⸮j⸮⸮+⸮⸮⸮0080000,len:252
entry 0x40080034
[0;32mI (43) "c⸮⸮*⸮P⸮R⸮⸮⸮2⸮⸮2⸮⸮⸮+⸮e bootloader[0m
[0;32mI (44) boot: compile time ⸮Ғ⸮Ң⸮⸮[0m
[⸮⸮&⸮+
B⸮⸮J⸮⸮⸮⸮E⸮//⸮⸮⸮⸮[nw⸮u⸮⸮+⸮⸮k⸮⸮⸮⸮⸮⸮⸮⸮e⸮⸮i⸮⸮*5⸮⸮[<;3⸮[RB⸮⸮J⸮⸮⸮⸮ SPI Speed      : 40MHz[0m
[0;32mI (72) boot: SPI Mode       
 DI⸮⸮
jj
⸮+ʚ⸮j%⸮(⸮S	⸮⸮⸮⸮ SPI Flash Size : 4MB[0m
[0;⸮⸮+
BʺJ⸮⸮⸮⸮ Partition Table:[0m
[0⸮&⸮+⸮r⸮k⸮{oov⸮;۽⸮⸮el⸮⸮⸮⸮>><>>>>Usyae⸮O⸮y0⸮⸮⸮⸮⸮⸮ST Offset   Length[0m
[0;32mI (131) boot:  0 ⸮+}⸮⸮⸮с        RB ⸮()⸮         ⸮L⸮⸮⸮⸮⸮⸮2⸮00 00001000[0m
[0;32mI (154) boot8  1 otaeata          OT data  
B⸮⸮4) boot:  4 ota_0     
B⸮⸮⸮⸮J⸮⸮⸮: segmeFw⸮⸮'⸮⸮⸮⸮~?⸮/S⸮m⸮⸮⸮⸮f⸮O⸮⸮>⸮⸮>>>⸮⸮-⸮V⸮⸮⸮0400 (  1024) load[0m
[0;32mI (1547) boot: segment 4: paddr=0x0012e⸮S⸮⸮⸮j0⸮⸮⸮0⸮⸮⸮Jie⸮⸮28 (110632) load[0m
⸮⸮۫⸮⸮lm⸮⸮⸮ګK⸮ҽ⸮⸮韮뵕⸮⸮⸮:>~⸮⸮50x00149b7c vaddr=0x400c0000 size=0x00034 (    52) load[0m
[0;32mI (⸮⸮dJ⸮⸮Q 3ea⸮Y⸮⸮⸮
⸮⸮⸮u0x00149bb8 vaddr=0x00000000 size=0x0645" ( 2568⸮J⸮[0m
[0;32m⸮B⸮⸮⸮⸮J⸮⸮⸮⸮ segmenv⸮S⸮y⸮W⸮⸮⸮⸮⸮՟⸮⸮⸮⸮⸮wf⸮O⸮⸮>>⸮⸮⸮⸮>size=0x7a544 (501060) map[0m
[0;32mI (1670) heap_alloc_caps: Initializing& RAE a⸮H*V⸮⸮⸮⸮f#⸮"e⸮X[Z,
⸮⸮⸮⸮⸮ѥ⸮⸮⸮[0m
[0;32mI (1693) heap_al$oc_caps At 3FFBP⸮⸮ len 0002594⸮B⸮⸮⸮Z⸮	⸮: DR⸮Է⸮j⸮⸮⸮C⸮⸮⸮⸮⸮⸮⸮⸮⸮⸮[⸮s⸮y~_ynnos5⸮|s:<qt83FFE8000 len 00018000 (96 KiB): D/IRAM[0m
[0;32mI (1735) heax_alloc⸮⸮⸮3 A4 4⸮⸮B4⸮b⸮⸮⸮00004BD8 (18 KiB): IRAM[0m
[0;32mI (1756) cpu_sta.⸮⸮ɽ⸮⸮pu up.⸮j5
I (2103) wifi: Init rx⸮X[⸮Ձl⸮j⸮⸮c⸮-⸮j
I (⸮L⸮J⸮J⸮⸮) Ifi⸮b⸮⸮⸮͍⸮rx ampdu entry mblock:4
I (2112) wafi:0wif) po⸮Y.j⸮nag⸮.⸮⸮ͭ⸮ 0x3⸮օ⸮d4 prio:<⸮n⸮⸮⸮⸮⸮ݒ⸮⸮k⸮I⸮⸮⸮⸮⸮j⸮⸮⸮gZ'⸮⸮gi⸮⸮Z[Y&@⸮qk: 3ffcb254, prio:22, \W⸮⸮⸮3584
⸮⸮+ښ⸮j⸮a⸮L⸮⸮	⸮⸮y_init:`P⸮V"⸮鱂
⸮TZ)⸮2!dida⸮Yִ⸮5

????????????????????????????????????????????????????????????????????????
```
I tried to copy it ... but its not getting copied.
After this its all ???????????????????????????????????????
What could be the issue!

stickbreaker, 2018-01-25:
@bkrajendra how are you connecting to the Serial port.  The ESP32 is 3.3v.  What are you using to convert from the 3.3v ttl serial to your computer?

Chuck.

bobmc-rmm, 2018-01-26:
Since some strings print correctly, the baud rate must be ok. The ESP32 UART is usually connected to the USB such that the upload file and serial messages share the same path. Are you somehow using RS232 instead.

bkrajendra, 2018-01-26:
@stickbreaker and @bobmc-rmm I'm using general FTDI cable with proper 3.3 v jumper, also made it sure by checking voltages at all pins and its 3.3v approx. As I further checked and compared schematic with others, I found that I was not using IO15 and IO13. As soon I've connected IO15 CTS, this ??? generation stopped and ESP32 started responding as expected. Now When I press switch attached to EN, it restarts and shows some meaningful dump. 

But the main issue of programming is not resolved. As you can see, I've autoreset circuit, which should automatically put ESP into boot mode and start programming but its not working. I also tried EN/IO0 button sequence to put it into boot mode, but it could not help. After pressing IO0 attached switch and then reset (EN) it shows some message which looks like its entered into programming mode. But when I start uploading, nothing happens. It just says fail to upload.

lbernstone, 2018-01-28:
Keep it simple.  All you need connected for programming is TX, RX, GND, 3V3 and GPIO0 (and enable of course).  Try programming without any additional UART lines and see if you can get it to upload.  GPIO0 may need to be held low until programming starts.  The boot message should be "Waiting for download" when pin 0 is grounded.

lbernstone, 2018-06-06:
Did you resolve this problem?

bkrajendra, 2018-06-07:
Yes, it did resolve the issue. I just removed transistor pai im using for autoreset.
This is the component im using for it:
[image]

Programming working fine. Normal Operation working fine. Just pulling down IO0 serves the purpose of programming mode.

But still i need to get solution for autoreset/programming mode to work. ANy suggestions for it.

lbernstone, 2018-06-07:
There are a number of schematics available that demonstrate how to set up a pair of transistors on gpio0 & EN (https://dl.espressif.com/dl/schematics/ESP32-Core-Board-V2_sch.pdf).  That clearly has nothing to do with arduino-esp32, and should be asked on https://www.esp32.com.  If your issue is solved, please close it.

closed 2019-08-02 by bkrajendra

Arnold208, 2024-02-07:
> There are a number of schematics available that demonstrate how to set up a pair of transistors on gpio0 & EN (https://dl.espressif.com/dl/schematics/ESP32-Core-Board-V2_sch.pdf). That clearly has nothing to do with arduino-esp32, and should be asked on https://www.esp32.com. If your issue is solved, please close it. @lbernstone 

in the schematics, does this 10uF/25V(10%)(NC) mean I do not connect that line?