espressif/arduino-esp32 #1028: Configuring WiFi Connection through Bluetooth

egadgetjnr, 2018-01-25:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:		25/01/2018
IDE name:						Platform.io
Flash Frequency:					40Mhz
Upload Speed:					115200

### Description:
I'm wanting to configure a Wireless AP connection through the Bluetooth (essentially, using a customised app to perform the configuration).

I've found a way to do it through BLE, but ofcourse, this isn't the most secure.

Am I maybe going about this the wrong way? Does anyone have any suggestions on the best way to do this?

beegee-tokyo, 2018-01-25:
Did you try ESPTouch?  Also known as SmartConfig  
[Espressif ESPTouch info](https://espressif.com/en/products/software/esp-touch/overview)    
[Android source code for ESPTouch](EsptouchForAndroid)    
[ESP32 example](https://github.com/espressif/esp-idf/tree/master/examples/wifi/smart_config) for esp-idf    
[ESP32 example](https://github.com/espressif/arduino-esp32/tree/57dbc9ab774c4730f52eef2069fbfeedf7e690e5/libraries/WiFi/examples/WiFiSmartConfig) for Arduino framework

egadgetjnr, 2018-01-25:
Wow... Now I feel quite embarrassed. lol. I actually didn't even know that's what that was...
I'm only fairly new to the ESP32, but a long time ESP8266 user so I didn't even think about that.

I think that will work perfectly and shall investigate. Thanks heaps @beegee-tokyo!

closed 2018-01-25 by egadgetjnr