espressif/arduino-esp32 #1031: ESPNOW Setup Question

m0nki, 2018-01-25:
I have 3 x ESP32s and want to send sensor data from two of them to one connected to PC via serial (so I can suck data into PC for manipulation). 
Most of the code examples I have seen are for multislave environment but I think I am looking for a multimaster setup? Just trying to get my head around how ESPNOW works but feel I might be completely missing something.  Can anyone point me to some code with two ESP's sending data to another one via ESPNOW? Thanks any helpful advice welcome.

beegee-tokyo, 2018-01-25:
[ESPNow examples](https://github.com/espressif/arduino-esp32/tree/4c36e033812a4142ba417565926ff8f919a32646/libraries/ESP32/examples/ESPNow)

dpharris, 2018-01-25:
Thanks.
Is ESPNow incompatible with regular Wifi use? -- ie can an ESP32 be used as
a bridge between several ESPNow 'slaves' and the wider net?

David

beegee-tokyo, 2018-01-26:
I got ESPNow and connection to AP working, but AP must be on WiFi channel 0. [Example](http://wiki.giesecke.tk/doku.php?id=tips_tricks:espnow_not_working)

dpharris, 2018-01-26:
@beegee-yokyo ... thanks for that, and good news.

closed 2018-04-19 by everslick