espressif/arduino-esp32 #1032: problem including the library

robademar, 2018-01-25:
I've a problem to get the net addrs using webbino so I tried to download again yìthe esp32 library, but when I include it from zip IDE says no valid library. Any idea? thanks

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE? ?Platform.io? ?IDF component?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2018-01-25:
The arduino-esp32 directory should be in <arduino_path>/hardware/espressif.  The hardware directory should already be there after installing the IDE.  When you open the IDE, you should then see ESP32 variants in the boards list. If you then get compile errors, you have a problem with your code.

robademar, 2018-01-25:
Thanks but the problem is different and it is the following:
Arduino:1.8.3 (Windows 10), Scheda:"ESP32 Dev Module, QIO, 80MHz, 4MB (32Mb), 921600, None"

La cartella/file ZIP specificata/o non contiene una libreria valida

Questo report potrebbe essere più ricco di informazioni abilitando l'opzione
"Mostra un output dettagliato durante la compilazione"
in "File -> Impostazioni"

The library arduino-esp32-master .zip,I downloaded from GitHub, causes that error when I include it with the include from zip option of IDE.            Invalid library

closed 2018-01-25 by robademar

lbernstone, 2018-01-25:
esp32 is a board, not a single library.  Please follow the installation instructions: https://github.com/espressif/arduino-esp32#installation-instructions

robademar, 2018-01-25:
ok right suggestion. I forgot how I did it before. Many thanks