espressif/arduino-esp32 #1035: Wifi / WPA2 connection problem: Guru Meditation Error of type LoadStoreError

nubiator123, 2018-01-26:
### Hardware:
Board:							ESP32 WROOM
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
After reconnecting to the Wifi I get an Exception. Sometimes it works, sometimes the cient stops working. The Exception is thrown randomly.

### Sketch:
```cpp

//Change the code below by your sketch
void setup() {
[...]
  WiFi.mode(WIFI_STA);
  WiFi.onEvent(WiFiEvent);//WiFi.setHostname(clientName.c_str());
  WiFi.begin(ssid, password);
[...]
}

void WiFiEvent(WiFiEvent_t event){
   Serial.printf("[Wifi-event] event: %d\n", event);
    switch(event) {
        case SYSTEM_EVENT_STA_START:
              WiFi.setHostname(clientName.c_str());
              esp_wifi_sta_wpa2_ent_set_identity(identity, sizeof(identity)-1);
              esp_wifi_sta_wpa2_ent_set_cert_key(testuser_cert_pem, sizeof(testuser_cert_pem)-1, testuser_key_pem, sizeof(testuser_key_pem)-1, NULL, 0);
              esp_wifi_sta_wpa2_ent_enable();
            break;
        case SYSTEM_EVENT_STA_CONNECTED:
            break;
        case SYSTEM_EVENT_STA_GOT_IP:
           disconnectLoop = 0;
            static bool sta_was_connected = true;
            wifi_connected = true;
            break;
        case SYSTEM_EVENT_STA_DISCONNECTED:
            wifi_connected = false;
            wifiOnDisconnect();
            break;
        default:
            break;
    }
}

### Debug Messages:
Decoding 14 results
0x4008250c: _realloc_r at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/newlib/./syscalls.c line 48
0x4012d4e4: wpabuf_resize at ?? line ?
0x4011c585: eap_peer_peap_register at ?? line ?
0x4011cb10: eap_peer_tls_process_helper at ?? line ?
0x4011d0f8: eap_peer_tls_phase2_nak at ?? line ?
0x4011ac7c: eap_sm_process_request at ?? line ?
0x4011af90: wpa2Task at ?? line ?

[D][WiFiGeneric.cpp:182] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:187] _eventCallback(): Reason: 204 - WIFI_READY
[Wifi-event] event: 5
[Wifi] Disconnected
Guru Meditation Error of type LoadStoreError occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4000c320  PS      : 0x00060a30  A0      : 0x8008250c  A1      : 0x3fffa9c0  
A2      : 0x3fffd5c8  A3      : 0x400000c4  A4      : 0x000012a7  A5      : 0x3fffe86c  
A6      : 0x1009c900  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3fffa990  
A10     : 0x3fffd5c8  A11     : 0x00000004  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00000000  A15     : 0x3fffd2a0  SAR     : 0x0000001f  EXCCAUSE: 0x00000003  
EXCVADDR: 0x400000c4  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4000c320:0x3fffa9c0 0x4008250c:0x3fffa9d0 0x4000bee0:0x3fffa9f0 0x4012d4e4:0x3fffaa10 0x4011c585:0x3fffaa30 0x4011cb10:0x3fffaa70 0x4011d0f8:0x3fffaab0 0x4011ac7c:0x3fffab00 0x4011af90:0x3fffab40

CPU halted.

nubiator123, 2018-02-01:
Any ideas, what this means?

everslick, 2018-04-19:
You did not provide any code for people to look at. If you feel like having hit a bug in the Arduino core, please show some code that demonstrates the issue. for general user support the gitter channel is a better place to ask for help.

If this issue is not of relevance anymore, please close it.

tito, 2019-02-26:
I recently hit this issue while doing OTA via the ESP32 WifiUpdate.
```
< Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
< Core 1 register dump:
< PC      : 0x400d6d18  PS      : 0x00060630  A0      : 0x800d6dbc  A1      : 0x3ffd9d00  
< A2      : 0x3ffedeac  A3      : 0x3ffd9d4f  A4      : 0x00000001  A5      : 0x40000000  
< A6      : 0x0000a1e8  A7      : 0x3ffd82a0  A8      : 0x00000000  A9      : 0xffffffff  
< A10     : 0x00000000  A11     : 0x3fff5e18  A12     : 0x00000050  A13     : 0x00000020  
< A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000a  EXCCAUSE: 0x00000003  
< EXCVADDR: 0x40000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
< 
< Backtrace: 0x400d6d18:0x3ffd9d00 0x400d6db9:0x3ffd9d20 0x401bfc07:0x3ffd9d40 0x400d8bb4:0x3ffd9d70 0x400d8c0d:0x3ffd9d90 0x400d434d:0x3ffd9db0 0x400d5511:0x3ffd9f90 0x400d67a6:0x3ffda1a0 0x400d35be:0x3ffda1f0 0x4019a508:0x3ffda210
< 
< Rebooting...
```

closed 2019-08-15 by stale[bot]