espressif/arduino-esp32 #1036: little differences from normal Arduino in SD library

MhageGH, 2018-01-27:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			13/jun/2017
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
1. You need root symbol "/" when you specify a file in arduino-esp32. 
   You don't need it in normal Arduino.
2. "FILE_WRITE" means "overwrite"  in arduino-esp32. 
   It means "adding" in normal Arduino.

### Sketch:

```
#include <FS.h>
#include <SD.h>

void setup() {
   SD.begin();
   //File file = SD.open("abc.txt", FILE_WRITE);  // can't open
   File file = SD.open("/abc.txt", FILE_WRITE);  // file content is "abcde".
   file.print("xxx");  // file content is not "abcdexxx" but "xxx".
   file.close();
}

void loop() {
}

```

embedded-creations, 2018-04-16:
There seem to be more than a few "little differences" in the libraries that are making it very difficult to write an Arduino sketch that is compatible with both libraries.  Compare the output from [this example sketch](https://www.arduino.cc/en/Reference/FileIsDirectory) on both ESP32 and Teensy:

ESP32

```
/gifs/
    /gifs/bigbuck1-64x32.gif        148635
    /gifs/._bigbuck1-64x32.gif      4096
    /gifs/ezgif-259182266.gif       199781
    /gifs/._ezgif-259182266.gif     4096
```

Teensy
```
GIFS/
    BIGBUC~1.GIF        148635
    _BIGBU~1.GIF        4096
    EZGIF-~1.GIF        199781
    _EZGIF~1.GIF        4096
```

raumaankidwai, 2018-11-27:
This is actually causing problems for me due to File.name() returning the entire file *path* instead of the *name*.

closed 2019-08-01 by MhageGH