espressif/arduino-esp32 #1037: captiveportal example not working on some platforms

tablatronix, 2018-01-27:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							doit v1, ESP32 Dev Module
Core Installation/update date:			today
IDE name:							platformio
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
using captiveportal example
No captive portal launches on windows, linux, android.
surprisingly it works on osx and ios.

Testato, 2018-01-27:
Bug confirmed here on Arduino IDE also

tablatronix, 2018-01-29:
I see lots of malformed dns packets in wireshark.

```
iturd-Pro:tools shawn$ dig google.com
;; Warning: Message parser reports malformed message packet.

; <<>> DiG 9.9.7-P3 <<>> google.com
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 42797
;; flags: qr rd ad; QUERY: 0, ANSWER: 0, AUTHORITY: 0, ADDITIONAL: 1
;; WARNING: recursion requested but not available

;; Query time: 8 msec
;; SERVER: 192.168.4.1#53(192.168.4.1)
;; WHEN: Sun Jan 28 18:09:06 CST 2018
;; MSG SIZE  rcvd: 12
```

`[Expert Info (Error/Malformed): Malformed Packet (Exception occurred)]`

tablatronix, 2018-01-29:
`Source mac must not be a group address` and lg and ig bits present, does that sound right?

tablatronix, 2018-01-29:
Looks like dig always fails, since it sends additional OPT every time, I noticed OPT and PTRs fail.

Ill look into this again tomorrow and new machines and see what I see.

Testato, 2018-01-29:
Probably i found the problem:
The DNSserver redirect all to 192.168.1.1, instead of the captive portal IP (192.168.4.1)

[image]

[image]

Log made on Win10

tablatronix, 2018-01-29:
Are you using the captiveportal example, it uses that address instead.

tablatronix, 2018-01-29:
This fixes it, the dns repsones are not including the question along with the answer.
This is from esp8266, not sure why @me-no-dev left it out.

DNSServer.cpp
```
@@ -135,15 +119,35 @@
   if (_buffer == NULL) return;
   _dnsHeader->QR = DNS_QR_RESPONSE;
   _dnsHeader->ANCount = _dnsHeader->QDCount;
-  _dnsHeader->QDCount = 0;
+  _dnsHeader->QDCount = _dnsHeader->QDCount; 
+  //_dnsHeader->RA = 1;  
 
   _udp.beginPacket(_udp.remoteIP(), _udp.remotePort());
   _udp.write(_buffer, _currentPacketSize);
+
+  _udp.write((uint8_t)192); //  answer name is a pointer
+  _udp.write((uint8_t)12);  // pointer to offset at 0x00c
+
+  _udp.write((uint8_t)0);   // 0x0001  answer is type A query (host address)
+  _udp.write((uint8_t)1);
+
+  _udp.write((uint8_t)0);   //0x0001 answer is class IN (internet address)
+  _udp.write((uint8_t)1);
+ 
   _udp.write((unsigned char*)&_ttl, 4);
```

tablatronix, 2018-01-29:
heres the blame
https://github.com/esp8266/Arduino/blame/20b7e480b5a3e9aeffc056020d816a2fad4cbe12/libraries/DNSServer/src/DNSServer.cpp

and forum post
http://www.esp8266.com/viewtopic.php?f=32&t=3618&start=36

nicogon, 2018-03-26:
Have you found a solution for android 7? Still not showing up popup

Testato, 2018-03-26:
It is important understand if the portal do not work or only do not automatically show the connection webpage.

After the connetction try to manually open Chrome browser on the phone and write the address www.test.it 
If the captive portal work you will be recirected to the Configuration page.

(It is important that you use an address not https, so google.com is not a good address)

tablatronix, 2018-03-26:
Yes of course, but that doesn't help with the issue.
@nicogon this is how I debugged my issue.

Here is my PR to add debugging to DNS server
https://github.com/espressif/arduino-esp32/pull/1046

define DEBUG_ESP_PORT in your build flags, or un-comment in the file.
You can also enable webserver debugging in a similar fashion.

If you enable both those and still see no logging, then your device is not even requesting a captive portal. In that case, either it disabled somehow, or it does not have satisfactory dns information to request a site. Which could indicate a bug.

Can you get a captive portal to work somewhere else? say a store or coffeeshop or something ?

treii28, 2018-03-30:
This seems to be a problem with most modern iphone, android and other devices. None of the code or examples I've seen of handling captive portal from ESP8266 work on anything newer than somewhere about IceCream I believe on the android side. You can see the DNS requests being made, but the phone doesn't appear to trust the responses as it never makes an http or https request to the 'spoofed' ip address. (at least not in my tests)

treii28, 2018-03-30:
As for my experiences, I have gotten captive portals before on my android phone where the pop-up appears. Both at public places like McDonald's or xFinity wifi as well as at home on an openWRT router. I've been trying to have someone more knowledgeable in C/Arduino/ESP/tcp/ip look into this for well over a year now. I was about to start boning up on lwip myself and hacking apart both the Android OS source code and openWRT source code to try to figure it out myself.

tablatronix, 2018-03-30:
So are you saying captive portal example works on neither esp32 nor esp8266 for your device/s?

treii28, 2018-04-01:
I have been trying to find a working captive portal for over a year. I even tried sniffing the packets at one point but had only a limited idea what to look for. The best I have ever achieved on an android device is a pop-up telling me the device is not connected to the internet. (no redirect)
If you need any help researching this issue or doing testing, don't hesitate to ask. I have half a dozen wifi adapters, Raspberry pis, laptops (all dual booted with windows and linux) and more than one of just about every type of ESP. I would love to see this working and will do whatever I can to help. But I'm not as familiar with using things like wireshark so if you need packets sniffed, you'll have to help me figure out the configuration/filters to get what you need.

tablatronix, 2018-04-01:
Can you run android on a raspi or virtuabox?
I have a zero, I gotta buy a android phone on ebay or craigslist, or see if someone I know has one

tablatronix, 2018-04-01:
I have android nougat running on virtualbox , but the mouse barely moves, and I cannot get the wireless bridge mode to work, not sure if this requires an extension pack or is just not possible.

I might need a better vm.

nicogon, 2018-04-02:
On android only works if you try to open a http page (not https). And is
not automatic as in ios or a pc.
I've been researching about this, and it seems the problem is that android
dont like that captive portal ip is in the same network. i've tested with a
functional captive portal that dns resolves correctly (for android captive
https://www.gstatic.com/generate_204 as an external ip) but it seems there
is some kind of http redirection of the request to the captive portal.
dns service should give an external ip and then do some kind of http
redirection (that i havent been able to do it with esp8266). Do you know if
there is a way of intercepting http packet to an external ip and deliver a
302 redirection to the captive portal?

treii28, 2018-04-02:
Android can be run on a pi, but I don't see the download on their standard downloads page anymore. Nor do I know how closely it would resemble what goes on in a phone. I may have to try it with my pi zero w (the one with wifi/bluetooth - they cost about $10-15 btw, Adafruit appears to be out-of-stock at the moment but amazon prime has an 'essentials' kit for $24. piMoroni, piHut and CanaKit don't appear to show them out of stock)
[http://androidpi.wikia.com/wiki/Android_Pi_Wiki](url)

But again, if there's anything you need tested, let me know. As mentioned, I could even set up an account on the pi-zero-w or the pi-3. (both have built in wifi. They aren't the most robust platforms for atmel/esp development, but I've done it on them. I generally use platformio when on the pis)

Testato, 2018-04-02:
Also on Android there is an automatic connection to the captive portal, but it depends on the Android version. On an LG G5 with Android 7, it work correctly. On  my second phone, Android 4.4 Samsung, i need to manually go to an http address

tablatronix, 2018-04-02:
Yeah I read that there are some bugs or changes in behavior after a certain android version, I also read about 204 expected url, and redirects expecting 307 temporary redirects, we send location header and a 302 redirect.

I also read about, captive portals failing if you leave cellular on, since it can still get out, and some notes about walled gardens.. no idea what that means.

treii28, 2018-04-02:
Hmmm. I had the opposite experience. On my older androids it still works and on anything newer than 4.4 it stopped working.

treii28, 2018-04-02:
re: cellular on
Hmmm, interesting. I may need to test this later as that makes sense. But doesn't help me with my project as I'm trying to get something to work that requires minimal instruction. If that's the case, I may need to go with an android app after all to handle the connection/data-exchange transparently.

tablatronix, 2018-04-02:
Yes I think I read that too, around version 5 something changed etc,

tablatronix, 2018-04-02:
To summarize things I want to test.

on device
* no security restrictions enabled , not sure specifics
* no custom dns 8.8.8.8 etc
* cellular off

in code
* on generate_204 requests send 204
* on all other send 307 redirect

treii28, 2018-04-02:
I suspected (but it's only my blind guessing) that they may have put some 'known trusted' ips somewhere in the code or some way of checking the ip so it doesn't follow dns spoofing attempts to localnet or other addresses. The android code itself doesn't show anything I could pick out as a test on it, but when I ran some of the DNS debug mods in the ESP, I could see the requests for sites like google.com coming in and the DNS responding with the local net address, I had the debugging in my web server code maxxed out so literally any request would show in some way, but I never see any of the requests coming after the DNS query from android.
That was when I tried setting up wireshark to try to see if 'any' TCP/IP/http connections went out after the DNS query, but I didn't capture any. In fact, many of the backgrounded apps still keep trying to go to their respective heartbeat URLs.

tablatronix, 2018-04-02:
It could also be some minor dns packet issue that it does not like, that would be the hope.

treii28, 2018-04-02:
I thought the gist on the 204 requests was to 'not' send a 204. i.e. Google requests the document and 'expects' a 204 if it's connected to the internet but you want the device to know it's "not connected" to the internet and thus redirect to the content.

tablatronix, 2018-04-02:
oh yeah that is to close the captive portal

treii28, 2018-04-02:
I've used the default coding in most of the examples as well as modified them every way from Sunday trying to see if something would work, but - at least as per my sniffing attempt - it doesn't seem to be even making the request after the DNS lookup.  It's been a while so I might need to try it again. (I was fighting with setting up a new development laptop this weekend)

tablatronix, 2018-04-02:
I gotta get me a device, I have every codebase and platform to test with and an enterprise captive portal at my disposal to sniff. Just no android lol. Maybe I will spin up some different vms today and see if any of them have wifi hardware bridge support, I bet vmware works, gonna try that next

lbernstone, 2018-04-02:
It works properly on my Pixel, using WiFiManager.  It asks me to log into the website, and clicking on the notification takes me to the WiFiManager homepage. I use genymotion for android vms on linux.  It is not fast, but gives reasonably accurate behavior.

treii28, 2018-04-10:
I got my esp32dev board the other day and tried the captive portal example on it and got the same result in android. It pops up that the thing may not be connected but doesn't do anything more or (give any options to) pop open a browser to any landing page.  Again if there's anything needs doing, don't hesitate to ask.

tablatronix, 2018-04-10:
Maybe post your android versions? Shrug

tablatronix, 2018-04-12:
So this is most likely not an security or process change in android since that goes back quite a ways.

treii28, 2018-04-12:
I hadn't really tested with the 4.4 in a while. At the time I first started trying this, my 5.0 device was on 4.4 and the 4.4 device was on 4.3.  The 4.3 still got the pop-up (at that time) and I haven't gotten a redirect on a ESP based captive portal on >4.4 ever

tablatronix, 2018-04-12:
you can manually pull up a page and be redirected, can you manually get to the ip address, do dns and webserver are both working ?

Testato, 2018-04-12:
@treii28
remember that the manual way work only for http, not https

treii28, 2018-04-17:
@tablatronix - yeah the webserver works just fine even with a given hostname spoofed by the dns server. but the application I'm trying to implement will be devices left out in public places out of site. (that's kinda the whole point of virtual geocaching -  you have to get close enough to connect to the device)  Thus I can't necessarily give detailed instructions on how to connect to it and even if I could, the kind of people likely to use it might have a hard time with it.
I'm talking to some local android guys to try to get them to help me to build an app to connect to them, but it would be nice if I could also allow wifi-captive redirection for people who don't have the app.

@Testato yes, I'm aware of that. I'm not trying to do anything fancy with ssl. I just need the device capable of being signed into so a geo-cacher can add a lot entry that they found the device. (or at least it's location - the devices will be hidden out of site so people don't tamper with them)

tablatronix, 2018-04-17:
ok so its not a dns library issue then

nicogon, 2018-04-20:
Did some test with a raspberry and I got it working by 
-Setting the DNS to resolve all queries  to 172.217.28.206 (public ip of http://clients3.google.com/)
-Then have a tcp rule to forward all port 80 traffic to my host 
-> captive portal working in all platforms. 
The problem with esp libraries is that I haven't been able to forward outgoing 80 tcp traffic to esp host. 
Is there any way  of doing this?

tablatronix, 2018-04-21:
how did you test this?

Why does it need or care what the ip is?

What are you forwarding ? On the client?

nicogon, 2018-04-21:
It looks like captive  portal on android won't show if client3.google.com
is resolved with a local ip (tested this, popup won't show). If i fake the
dns query with an external ip, when the http query (
http://captive.apple.com/generate_204 ) is made the raspberry would answer
as if it was an external host.

Example
1.Client make a client3.google... dns query to raspberry.
2. Raspberry answer  dns query with an external ip (i used the real
client3.googe hardcoded ip)
3. The client will make a http query to that external ip to the ap
(raspberry)
4. Instead of forwarding this packet to the wan (because there is no wan)
rasberry will answer to this packet as if it was an external ip with the
captiveportal html

Sorry for my bad English

tablatronix, 2018-04-21:
So does setting the ip manually for the esp to a non class c resolve this ? it should if that is indeed what is going on.

nicogon, 2018-04-21:
hell yeah! you were right, no need of http reroute. 
**Just have the captive portal ip as an external ip**. I used 
172.217.28.1 as esp ip with captive portal example (network mask 255.255.255.0) and I got the popup woking on android 7 (Samsung S7), Iphone 7 and OSX. 

I would suggest change captive portal example  with this configuration. 

[image]
[image]

treii28, 2018-04-21:
talking with nic on facebook atm - I'll give that a shot on my end with the IP for clients3.google.com - what I'm thinking is perhaps set up the primary points such as the google and apple main addresses in combination so it can answer either locally without just a spoofing.
(updating a banana pi zero OS at the moment or I would have tried it already)

tablatronix, 2018-04-21:
Well we need to see where this is documented first, and find out why this behavior is such, si ce you really arent supposed to be doing stuff like that.

tablatronix, 2018-04-21:
Have we confirmed that android is actually using the dhcp addresses assigned to it by esp?

beegee-tokyo, 2018-04-22:
@nicogon just for correctness, 172.x.x.x is not an 'external IP' but according to [RFC1918 chapter 3](https://tools.ietf.org/html/rfc1918) one of the address ranges for "Private Address Space". It is however a "class B" address range and it seems that solves the problem. An AP address of 10.x.x.x should work as well as it is a "class A" address range.

tablatronix, 2018-04-22:
Thanks for the clarification, makes sense, I have seen 172 and 10 used alot more and was wondering how that worked

tablatronix, 2018-04-22:
I have no problem with ios or windows however when using 192. space

tablatronix, 2018-04-22:
I would expect all https to be blocked by browsers with cors compliance

Testato, 2018-04-22:
Also for me there is no problem on Android7 and 192 space
A complete test should be at least on major OS used
Win10
Ubuntu
MacOs
Android7
Ios

treii28, 2018-05-01:
FYI, tried Nic's suggestion yet again with a 172.16.1.1 address specifically. Still nada

tablatronix, 2018-05-02:
The problem is not dns, dns already does that with *, the problem is there seems to be a ip restriction requirement.

Chinoforest, 2018-07-03:
Any update on this? Did someone found any official documentation that explains  this problem?

Chinoforest, 2018-07-03:
Maybe this does not work because we are not working EXACTLY like a captive portal should. Check this:
[https://serverfault.com/questions/368644/how-do-captive-portal-network-connections-work](url)
[https://community.arubanetworks.com/t5/Security/How-does-captive-portal-authentica-tion-really-work/m-p/165596#M12502](url)

This is what we do: 
1) Android connects to the AP and sends DNS request for clients3.google.com/
2) We responde with the AP IP address.
3) Android sends HTTP GET for generate_204 to AP webserver.
4) We responde, "it moved .... Location: http:// AP Ip address"

The DNS request for clients3.google.com returns the same IP Address as the request for generate_204. We don't hijack the generate_204 http as the linked solutions, we just reply because we pretend to be clients3.google.com. Probably there's a issue with that. I'm thinking out loud here, trying to brain storm this problem.

RobbesU, 2018-08-03:
my setup: Android 7.0 on S6 with Firefox browser.   (after manually going to 192.168.4.1)
Gents, I'm by no means an experts in this topic, but one thing that strikes me as not being great is that the result from the wifiManager configuration page, i.e. the "192.168.4.1/wifisafe?s=[ssid]&p=[password]" is send in the open. The issue is that this is now stored in the users web history (unless done in Private Mode on browser) and thus open to vulnerability from that end on that phone/pc/tablet. That doesn't sound cosher.

tablatronix, 2018-08-04:
This has nothing to do with wifimanager this is captive portal testing using default example.

Also that is fixed in development version.

Use minimal captiveportal code

codebeat-nl, 2019-02-08:
Had the same problem.

Found this: https://www.esp8266.com/viewtopic.php?f=6&t=15993

This post of eduperez:

----------------

*"Android devices have the Google DNSs hard-coded, they will **always use 8.8.8.8** and 8.8.4.4, despite what the DHPC server might tell them to use. You need to configure your gateway to redirect all outgoing traffic to port 53 to your DNS."*

---------------

That gives the idea, use **8.8.8.8 as IP-address and as DNS-server address**. 
Tada! Works like a charm!

So this demo sketch provided/included at ESP library will work:

    #include <ESP8266WiFi.h>
    #include <WiFiClient.h>
    #include <ESP8266WebServer.h>
    #include <DNSServer.h>
    #include <ESP8266mDNS.h>
    #include <EEPROM.h>
    
    /*
       This example serves a "hello world" on a WLAN and a SoftAP at the same time.
       The SoftAP allow you to configure WLAN parameters at run time. They are not setup in the sketch but saved on EEPROM.
       Connect your computer or cell phone to wifi network ESP_ap with password 12345678. A popup may appear and it allow you to go to WLAN config. If it does not then navigate to http://192.168.4.1/wifi and config it there.
       Then wait for the module to connect to your wifi and take note of the WLAN IP it got. Then you can disconnect from ESP_ap and return to your regular WLAN.
       Now the ESP8266 is in your network. You can reach it through http://192.168.x.x/ (the IP you took note of) or maybe at http://esp8266.local too.
       This is a captive portal because through the softAP it will redirect any http request to http://192.168.4.1/
    */
    
    /* Set these to your desired softAP credentials. They are not configurable at runtime */
    #ifndef APSSID
    #define APSSID "TheGeekMan"
    #define APPSK  "12345678"
    #endif
    
    const char *softAP_ssid = APSSID;
    const char *softAP_password = APPSK;
    
    /* hostname for mDNS. Should work at least on windows. Try http://esp8266.local */
    const char *myHostname = "thegeekman";
    
    /* Don't set this wifi credentials. They are configurated at runtime and stored on EEPROM */
    char ssid[32] = "";
    char password[32] = "";
    
    // DNS server
    const byte DNS_PORT = 53;
    DNSServer dnsServer;
    
    // Web server
    ESP8266WebServer server(80);
    
    /* Soft AP network parameters */
    //IPAddress apIP(192, 168, 4, 1);
    IPAddress apIP(8, 8, 8, 8);
    IPAddress netMsk(255, 255, 255, 0);
    
    
    /** Should I connect to WLAN asap? */
    boolean connect;
    
    /** Last time I tried to connect to WLAN */
    unsigned long lastConnectTry = 0;
    
    /** Current WLAN status */
    unsigned int status = WL_IDLE_STATUS;
[... 244 lines cut ...]
            Serial.print("IP address: ");
            Serial.println(WiFi.localIP());
    
            // Setup MDNS responder
            if (!MDNS.begin(myHostname)) {
              Serial.println("Error setting up MDNS responder!");
            } else {
              Serial.println("mDNS responder started");
              // Add service to MDNS-SD
              MDNS.addService("http", "tcp", 80);
            }
          } else if (s == WL_NO_SSID_AVAIL) {
            WiFi.disconnect();
          }
        }
        if (s == WL_CONNECTED) {
          MDNS.update();
        }
      }
      // Do work:
      //DNS
      dnsServer.processNextRequest();
      //HTTP
      server.handleClient();
    }

[10 more replies]

closed 2019-09-13 by stale[bot]