espressif/arduino-esp32 #1040: SoftAP interface erratic when STA connection is not established

anteph, 2018-01-28:
### Hardware:
Board:							ESP32 Generic NodeMCU module
Core Installation/update date:	       30/dez/2017
IDE name:						Arduino IDE

### Description:
Hi! I've been working with the WIFI_MODE_APSTA and ran into some problems. 

Basically, what I'm trying to achieve is setting the ESP32 as soft AP and running a HTTP server to receive a request with the credentials of a WiFi network. Upon receiving the credentials, I want to check if the ESP32 can connect to it, so I can confirm the password and SSID are correct.

Nonetheless, I want to test it synchronously, so I can return an answer to the HTTP Client still on the soft AP interface. The testing client I've used was Postman, but in the future it will be a smartphone or tablet app, although this shouldn't make much difference for the issue.

So far, the only way I've found to test the credentials correctness is by trying to connect to the supplied WiFi network and check if it succeeds.

The problem is when the client sends the wrong credentials, the soft AP starts behaving unpredictably. For instance, most of the times it doesn't return an answer to the client and if I repeat the request, much of the times it no longer reaches the server.

Furthermore, when it does, even if in the next request I send the correct credentials of the WiFi network, it can't connect to it.

Another point is that most of the times I cannot reach the server again without resetting the ESP32. Sometimes, disconnecting and reconnecting my computer to the soft AP network works, but it's always unpredictable.

When I send the correct credentials of the WiFi network in the first time, everything seems to work fine and I receive the answer to the HTTP request still on the soft AP interface.

I've searched over the internet and it seems the ESP8266 had a similar problem due to the fact that both the AP and station interfaces shared the same radio and that would cause troubles for the situation where the credentials of the network are wrong because it would need to scan in all the channels, thus disconnecting the soft AP.

Is this the same situation of the ESP32? Is there any workaround?

I really don't need to connect the ESP32 to the WiFi network as soon as I receive the credentials, I can do it later after a software restart. Nonetheless, I really need to confirm they are correct, so I can answer back to the client. So, if there is an alternative way to confirm the credentials (in particular the password, the network name is easy with a scan) that would also work for me.

Nonetheless, having the possibility to confirm this seems to be critical for real application use cases, since for an IoT product we most likely will not have access to the WiFi network credentials beforehand and will probably have to collect them over a soft AP connection. Thus, being able to deal with the user inputting an incorrect password would facilitate things tremendously.

Please let me know if you need me to test any particular sequence of steps with the debugger enabled, or some other procedure that may help.

Thanks in advance,
Nuno Santos

### Sketch:
```
#include <WiFi.h>
#include <FS.h>
#include <AsyncTCP.h>
#include <ArduinoJson.h>
#include <Preferences.h>
#include <ESPmDNS.h>
#include <ESPAsyncWebServer.h>

const char* softssid = "esp";
const char* softpassword =  "esp";

AsyncWebServer server(80);
char * connectWiFiRoute = "/connect";

void setup() {

  Serial.begin(115200);
  
  WiFi.mode(WIFI_MODE_APSTA);
  WiFi.softAP(softssid, softpassword);
  Serial.println(WiFi.softAPIP());

  delay(2000);
  
  server.onRequestBody([](AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total){
      
  // Handle wifi connect body
  if(request->url() == connectWiFiRoute){
  
      StaticJsonBuffer<100> JSONBuffer; //Memory pool
      JsonObject& parsed = JSONBuffer.parseObject((const char*)data); //Parse JSON
      
      const char * password = parsed["password"];
      const char * ssid = parsed["ssid"];

      WiFi.begin(ssid, password);
      
      if (WiFi.waitForConnectResult() != WL_CONNECTED) {
        request->send(200, "text/plain", "error");
      }else{
        request->send(200, "text/plain", "Noice");
      }
    }
    
  });
  
  server.begin();
}

void loop() {}
```

everslick, 2018-01-28:
> I've searched over the internet and it seems the ESP8266 had a similar problem due to the fact that both the AP and station interfaces shared the same radio and that would cause troubles for the [...]

It seems to be the same for ESP32. The solution for me is to not worry about what the user enters as SSID/PASSWORD for the local WiFi and just try to connect after a reboot. If it is wrong, the ESP32 will not connect anyway and the user can again get access the via SoftAP and enter correct credentials.

The important thing is to disconnect the STA as soon as some client connects to the SoftAP, because otherwise things will get messed up as you describe, due to the ESP32 only having one radio. see pseudo code below:

```
static void default_event_handler(WiFiEvent_t event) {
    if (event == SYSTEM_EVENT_AP_STACONNECTED) {
      log_print(F("WIFI: client connected to SoftAP"));

      if (wifi_is_enabled && !net_connected()) {
        log_print(F("WIFI: disconnecting from AP"));

        WiFi.enableSTA(false);
      }
    } else if (event == SYSTEM_EVENT_AP_STADISCONNECTED) {
      log_print(F("WIFI: client disconnected from SoftAP"));

      if (wifi_is_enabled && !net_connected() && (net_ap_clients() == 0)) {
        log_print(F("WIFI: reconnecting to AP"));

        WiFi.enableSTA(true);
      }
    }
```

tablatronix, 2018-01-28:
This is well documented and all over the issues, you have to disable sta if not connected or use set channels, otherwise sta reconnect scans destabilize softap.

anteph, 2018-01-29:
@everslick thanks for the suggestion :) If there's no workaround then we will have to use a similar solution. Although it won't be the smooth user experience we wanted, we can live with that.

@tablatronix thanks for the feedback, I've not been able to find any documentation on this, which is why I've opened the issue, but probably didn't look in the correct place. Could you please share the link for the documentation where this is explained? It will be useful for me to understand what are the limitations and hopefully also for others who are experiencing the same problem.

Best regards,
Nuno Santos

closed 2018-04-19 by everslick