espressif/arduino-esp32 #1042: Installation instructions for Debian / Ubuntu OS fails due python script

birgersp, 2018-01-29:
Ubutu 17.10, 64-bit

Following [these instructions](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md) fails:

```
birger@computer:~/Arduino/hardware/espressif/esp32/tools$ python get.pyTraceback (most recent call last):
  File "get.py", line 27, in <module>
    current_dir = os.path.dirname(os.path.realpath(unicode(__file__)))
NameError: name 'unicode' is not defined
```

And this answer on stack overflow explains the probable cause: https://stackoverflow.com/a/19877309/2583765

birgersp, 2018-01-29:
Changing the command from `python get.py` to `python2 get.py` fixes the issue

closed 2018-03-05 by birgersp

jefarnham, 2018-03-11:
When I enter "python2 get.py" in the Terminal, I get:  
File "/usr/local/lib/python2.7/urllib.py", line 222, in open_unknown raise IOError, ('url error', 'unknown url type', type) 
IOError: [Errno url error] unknown url type: 'https'

The script fails to create xtensa-esp32-elf folder and can't populate it when the required files to compile with the ESP32 board.
Any ideas about how to fix this?

JamesGKent, 2018-03-12:
looking at the answer here:
https://stackoverflow.com/questions/28376506/urllib-https-request-urlopen-error-unknown-url-type-https

you need ssl support in your python installation.