espressif/arduino-esp32 #1044: Ubuntu 17.10: ESP boards does not show in list of boards in Arduino IDE

birgersp, 2018-01-29:
Followed installation instructions for Debian/Ubuntu, restarted Arduino IDE but the ESP boards does not show in the list of boards in the Arduino IDE.

Ubuntu 17.10, 64-bit

Arduino installed to `/usr/share/arduino`

lbernstone, 2018-01-29:
Works for me on same OS, installed last week.  Check where you cloned the git repo.  It should be in <arduino>/hardware/espressif/.  Check your permissions there.

birgersp, 2018-01-30:
Removed arduino with `sudo apt remove --purge arduino`, and re-installed it (installed to `/usr/share/arduino`). changed ownership of `usr/share/arduino` so I own the folder. Created the directory `/usr/share/arduino/hardware/espressif`, `cd`ed into it and executed the following commands:

```
sudo usermod -a -G dialout $USER && \
sudo apt-get install git && \
wget https://bootstrap.pypa.io/get-pip.py && \
sudo python get-pip.py && \
sudo pip install pyserial && \

git clone https://github.com/espressif/arduino-esp32.git esp32 && \
cd esp32 && \
git submodule update --init --recursive && \
cd tools && \
python2 get.py
```

Still, ESP boards does not show in the list...

@lbernstone did you install Arduino to `/usr/share/arduino`?

birgersp, 2018-01-30:
Removed `arduino`, installed arduino from arduino.cc. rather than from ubuntu apt repository. Installed arduino to `~/Apps/arduino-1.8.5`, installed the esp32-thing at `~/Apps/arduino-1.8.5/hardware/espressif/esp32`. It seems to work now.

But instructions give the impression that the `esp32` stuff should reside in `~/Arduino/hardware/espressif/esp32`, but isn't that **entirely** dependant on where you have arduino installed?

closed 2018-01-30 by birgersp

lbernstone, 2018-01-30:
Instructions certainly aren't as clear as they could be, and should probably be changed to something more standard (<arduino_path>).  I installed to the default location in home, but tested moving it into /usr/bin and giving ownership to root when I read your message. That worked fine.  The arduino in apt is 1.0.5, which is pretty ancient and pre-dates esp32, maybe even esp8266.