espressif/arduino-esp32 #1047: bug in reporting of bytes used when flashing

Dannsei, 2018-01-29:
(Using Arduino IDE 1.8.5)

Bytes used by variables in the sketch are not being correctly reported.  The same value is reported after adding or deleting variables from the sketch or making an array larger in the sketch. e.g. a sketch with this line:

`const int trialSeq[] = {3702,1714,3702,17435,3702,1714};`

is reported as using same number of bytes of program and dynamic memory as when the line is changed to:

`const int trialSeq[] = {3702,1714,3702,17435,3702,1714,8702,1354,3752,14335,702,4714};`

I tested by compiling the same sketch for ESP8266; these changes are correctly reflected with that board.

igrr, 2018-01-30:
The binaries flashed into ESP32 are composed from several sections (code, read-only data, initial values for read-write data, etc). Some of these sections must start at certain offset in flash memory (multiple of 64kB) due to limitations of the Flash MMU. So the tool which generates the binaries will add some amount of padding between different sections. If the section has padding after it, increasing the size of the section will not immediately increase the total binary size, but will reduce the amount of padding.

Dannsei, 2018-01-30:
Thank you for this very clear explanation. Issue resolved.

closed 2018-01-30 by Dannsei