espressif/arduino-esp32 #1048: BLE Arduino library - library.properties refers to non-existent include file BLE.h

timr49, 2018-01-29:
### Hardware:
Board:							Sparkfun ESP32 Thing
Core Installation/update date:			30/1/2018
IDE name:							Arduino IDE 1.8.5
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
If you select from the Arduino IDE menu:

Sketch
Include Library>
ESP32 BLE Arduino

the following includes are inserted into your sketch:

#include <BLE.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

However when you compile the sketch it fails with

... fatal error: BLE.h: No such file or directory

The files included by Sketch/Include/ESP32 BLE Arduino are determined by the includes= line in the library's library.properties files. (See https://github.com/arduino/arduino/wiki/arduino-ide-1.5:-library-specification). This line is:

includes=BLE.h, BLEUtils.h, BLEScan.h, BLEAdvertisedDevice.h

but there is no file BLE.h in the src directory.

### Sketch:
```cpp

#include <BLE.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

### Debug Messages:
```
/var/folders/5k/v4z291_x3sd08tt8ldfb34km0000gn/T/arduino_modified_sketch_27873/sketch_jan30a.ino:1:17: fatal error: BLE.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board SparkFun ESP32 Thing.
```

lrmoreno007, 2018-01-29:
You must install submodules. Read how to install in the documentation.

timr49, 2018-01-30:
Thanks, but I  have installed theBLE submodule installed from
https://github.com/nkolban/ESP32_BLE_Arduino.git
and, alternatively, I have run
git submodule update --init —recursive
in the esp32 arduino directory per issue #766, but the file BLE.h simply does not exist in libraries/BLE/src.

-Tim.

closed 2019-08-15 by stale[bot]