espressif/arduino-esp32 #1050: DS18B20 temp probe keeps on showing 85

JacoFourie, 2018-01-30:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module?
Core Installation/update date:			?11/jul/2017?
IDE name:							?Arduino IDE 
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
DS18B20 temp probe keeps on showing 85

### Sketch:
```

#include <DallasTemperature.h>
#include <OneWire.h>

OneWire oneWire_1(4);
DallasTemperature sensor_t1(&oneWire_1);

void setup() {
   
  // initialize serial port
  Serial.begin(115200);
  while (!Serial) {
    ; // wait for serial port to connect
  }

  Serial.flush();

}

void loop() {

  Serial.print("Temp is : ");
  Serial.println(sensor_t1.getTempCByIndex(0));
  

}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

DeltaCore, 2018-01-30:
I was unable to get a ds18b20 to run reliably as well, not sure why till this day.

stickbreaker, 2018-01-30:
@JacoFourie @DeltaCore What version of the `OneWire.h` library are you using? The OneWire protocol is very timing sensitive. I modified the current version maintained by Paul Stoffregen [PaulStoffregen/OneWire](https://github.com/PaulStoffregen/OneWire) to work with the ESP32 it is in my Github repo at [stickbreaker/OneWire](https://github.com/stickbreaker/OneWire).  Try it. It has reliably worked for many people.

Chuck.

DeltaCore, 2018-01-30:
@stickbreaker Oh wow, i was not aware of a lib that has the correct timing for the ESP32. 
Thanks for that, i will give it a try when i have the time for that. 
The DS18B20 was on my todo list aswell.

JacoFourie, 2018-01-31:
Hi Check. I did download it. But then I saw the compiler was not using your lib.

```
Multiple libraries were found for "OneWire.h"
 Used: C:\Users\Jaco\Documents\Arduino\libraries\OneWire
 Not used: C:\Program Files (x86)\Arduino\libraries\OneWire
```
I now removed the older one but now it will not compile.

```
Compiling library "OneWire"
"C:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/spiffs" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/console" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/soc" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -MMD -c -DF_CPU=240000000L -DARDUINO=10804 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="ESP32_DEV" -DARDUINO_VARIANT="esp32"  -DESP32 -DCORE_DEBUG_LEVEL=0 "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Program Files (x86)\Arduino\hardware\espressif\esp32\variants\esp32" "-IC:\Program Files (x86)\Arduino\libraries\ArduinoTemperatureControlLibrarymaster" "-IC:\Program Files (x86)\Arduino\libraries\OneWire" "C:\Program Files (x86)\Arduino\libraries\OneWire\OneWire.cpp" -o "C:\Windows\Temp\arduino_build_744339\libraries\OneWire\OneWire.cpp.o"
In file included from C:\Program Files (x86)\Arduino\libraries\OneWire\OneWire.cpp:143:0:

C:\Program Files (x86)\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h: In function 'uint32_t directRead(uint32_t)':

C:\Program Files (x86)\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:111:13: error: redefinition of 'uint32_t directRead(uint32_t)'

 IO_REG_TYPE directRead(IO_REG_TYPE pin)

             ^

In file included from C:\Program Files (x86)\Arduino\libraries\OneWire\OneWire.h:60:0,

                 from C:\Program Files (x86)\Arduino\libraries\OneWire\OneWire.cpp:142:

C:\Program Files (x86)\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:111:13: note: 'uint32_t directRead(uint32_t)' previously defined here

[... cut ...]
C:\Program Files (x86)\Arduino\libraries\OneWire\util/OneWire_direct_gpio.h:168:6: note: 'void directModeOutput(uint32_t)' previously defined here

 void directModeOutput(IO_REG_TYPE pin)

      ^

Using library ArduinoTemperatureControlLibrarymaster at version 3.7.6 in folder: C:\Program Files (x86)\Arduino\libraries\ArduinoTemperatureControlLibrarymaster 
Using library OneWire at version 2.3.3 in folder: C:\Program Files (x86)\Arduino\libraries\OneWire 
exit status 1
Error compiling for board ESP32 Dev Module.
```

JacoFourie, 2018-01-31:
OK so I downloaded your one again and now it compiles. But I still get 85 all the time.

stickbreaker, 2018-01-31:
@JacoFourie

I think I figured out your problem.  Your posted sketch is missing a couple of commands.
If you are using `DallasTemperature.h` by [milesburton/Arduino-Temperatue-Control-Library](https://github.com/milesburton/Arduino-Temperature-Control-Library), the following is the `Simple.pde`

```c++
// Include the libraries we need
#include <OneWire.h>
#include <DallasTemperature.h>

// Data wire is plugged into port 2 on the Arduino
#define ONE_WIRE_BUS 2

// Setup a oneWire instance to communicate with any OneWire devices (not just Maxim/Dallas temperature ICs)
OneWire oneWire(ONE_WIRE_BUS);

// Pass our oneWire reference to Dallas Temperature. 
DallasTemperature sensors(&oneWire);

/*
 * The setup function. We only start the sensors here
 */
void setup(void)
{
  // start serial port
  Serial.begin(9600);
  Serial.println("Dallas Temperature IC Control Library Demo");

  // Start up the library
  sensors.begin();
}

/*
 * Main function, get and show the temperature
 */
void loop(void)
{ 
  // call sensors.requestTemperatures() to issue a global temperature 
  // request to all devices on the bus
  Serial.print("Requesting temperatures...");
  sensors.requestTemperatures(); // Send the command to get temperatures
  Serial.println("DONE");
  // After we got the temperatures, we can print them here.
  // We use the function ByIndex, and as an example get the temperature from the first sensor only.
  Serial.print("Temperature for the device 1 (index 0) is: ");
  Serial.println(sensors.getTempCByIndex(0));  
}
```
You need to to add `sensor_t1.begin()` to your `setup()` and you have tell the sensors to do a conversion with `sensor_t1.requestTemperatures()`.  Before you can read the temperature with `sensor_t1.getTempCByIndex(0)`.

Chuck.

JacoFourie, 2018-01-31:
Aaaaa Chuck I feel like an idiot. Sorry for wasting your time. i have been using this code for years on Arduino. I missed the conversion part. That is what you get doing a copy paste at 2 in the morning. All is working now. Thanks for fixing the lib.

stickbreaker, 2018-01-31:
@JacoFourie Join the club 😃 
If your problem is resolve, close this issue.

Chuck.

closed 2018-01-31 by JacoFourie

iwas108, 2021-11-06:
Don't forget to call sensor.requestTemperatures()